ev3dev/ev3dev #968: PS3 controller connectivity issues

dwalton76, 2017-09-26:
### What are you trying to do?
Pair a PS3 controller via bluetooth by following this guide
http://www.ev3dev.org/docs/tutorials/using-ps3-sixaxis/

### What did you expect to happen?
For it to pair :)

### What actually happened?
Everything matches up with the guide up until the final step

> 7. The brick now asks “Authorize service HID?” Press “Accept”

I never get the "Authorize service HID" message.  When I did
> 3. Connect the gamepad via a mini USB cable to the EV3

dmesg showed
```
[  884.854259] usb 1-1.2: new full-speed USB device number 5 using ohci
[  884.995348] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[  884.995433] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  884.995486] usb 1-1.2: Product: PS3 GamePad
[  884.995533] usb 1-1.2: Manufacturer: SHANWAN
[  885.363522] input: SHANWAN PS3 GamePad as /devices/platform/ohci.0/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0001/input/input2
[  885.379245] sony 0003:054C:0268.0001: input,hidraw0: USB HID v1.10 Joystick [SHANWAN PS3 GamePad] on usb-ohci.0-1.2/input0
```
I stepped through this tutorial
https://wiki.gentoo.org/wiki/Sony_DualShock

and this piece seems key

**Enabling Sony DualShock support**
```
    Device Drivers --->
          Input device support --->
            <*>   Joystick interface
            <*>   Event interface
          HID support --->
            [*]   Battery level reporting for HID devices
            [*]   /dev/hidraw raw HID device support
            <*>   Generic HID driver
                  Special HID drivers --->
                    <*> Sony PS2/3/4 accessories
                    [*]   Sony PS2/3/4 accessories force feedback support
                USB HID support --->
                  <*> USB HID transport layer
      [*] LED Support --->
            <*>   LED Class Support
```
How do I check to see if HID support is enabled in our kernel?

### What hardware and software are you using (including version numbers)?
#### Hardware
Hardware is an EV3.  My controller isn't an official SONY controller, it is this one by MarioRetro:
http://addictedgamer.com/playstation-3-games/wireless-gaming-controller-ps3-bluetooth-gamepad-game-controller-with-dual-vibration-joysticks-for-playstation-3-generic-by-marioretro/

I'm not sure if that is part of the issue or not.  If anyone has an official PS3 controller can you see if it pairs?

#### Software
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-09-26:
Official PS3 controller is working for me on fresh install of ev3dev-jessie (same kernel).

```
robot@ev3dev:~$ ev3dev-sysinfo 
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-09-26:
What is output of `lsusb`? I suspect this device is not in the mainline kernel.

dwalton76, 2017-09-26:
It shows up as a PS3 controller
```
robot@ev3[~]# lsusb
Bus 001 Device 005: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 006: ID 054c:0268 Sony Corp. Batoh Device / PlayStation 3 Controller
Bus 001 Device 004: ID 1a40:0101 Terminus Technology Inc. 4-Port HUB
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
robot@ev3[~]# 
```

dlech, 2017-09-26:
You might try ev3dev-stretch since it has a newer kernel. There are a few hack for 3rd party controllers that aren't 100% compatible, but you might be out of luck unless you want to build your own kernel and debug it.

JorgePe, 2017-09-26:
What do you get if you print all devices?
```
devices = [evdev.InputDevice(fn) for fn in evdev.list_devices()]
for device in devices:
    print(device)
```
Also on my Ubuntu I need to run with sudo or give 777 permissions to /dev/input/event20 AFTER pairing my generic BT gamepad - evdev needs membership of group plugdev / input but doesn't work on my Ubuntu laptop and it doesn't see any device.

dwalton76, 2017-09-27:
Same problem on stretch....I'll chalk this up to the $10 controller I bought.

dwalton76, 2017-10-05:
I got an official PS3 controller and get the `Authorize service HID?` prompt now but the controller does not show up via `evdev.list_devices()`.  David does yours show up as an evdev device?

dlech, 2017-10-24:
As a matter of fact, it doesn't:

```

[597881.224820] usb 1-1.4.4: new full-speed USB device number 5 using ohci
[597881.432668] usb 1-1.4.4: New USB device found, idVendor=054c, idProduct=0268
[597881.474157] usb 1-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[597881.534254] usb 1-1.4.4: Product: PLAYSTATION(R)3 Controller
[597881.567024] usb 1-1.4.4: Manufacturer: Sony
[597882.273288] input: Sony PLAYSTATION(R)3 Controller as /devices/platform/soc@1c00000/ohci-da8xx/usb1/1-1/1-1.4/1-1.4
.4/1-1.4.4:1.0/0003:054C:0268.0001/input/input2
[597882.399249] sony 0003:054C:0268.0001: input,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on us
b-ohci-da8xx-1.4.4/input0
[597889.001317] usb 1-1.4.4: USB disconnect, device number 5
[597911.881476] usb 1-1.4.4: new full-speed USB device number 6 using ohci
[597912.079365] usb 1-1.4.4: New USB device found, idVendor=054c, idProduct=0268
[597912.098164] usb 1-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[597912.122616] usb 1-1.4.4: Product: PLAYSTATION(R)3 Controller
[597912.146993] usb 1-1.4.4: Manufacturer: Sony
[597912.270579] input: Sony PLAYSTATION(R)3 Controller as /devices/platform/soc@1c00000/ohci-da8xx/usb1/1-1/1-1.4/1-1.4
.4/1-1.4.4:1.0/0003:054C:0268.0002/input/input3
[597912.320390] sony 0003:054C:0268.0002: input,hidraw0: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on us
b-ohci-da8xx-1.4.4/input0
[597913.384453] Bluetooth: HIDP (Human Interface Emulation) ver 1.2
[597913.464282] Bluetooth: HIDP socket layer initialized
[597913.694183] input: PLAYSTATION(R)3 Controller as /devices/platform/soc@1c00000/serial8250.2/tty/ttyS2/hci0/hci0:1/0
005:054C:0268.0003/input/input4
[597913.819641] sony 0005:054C:0268.0003: input,hidraw1: BLUETOOTH HID v1.00 Joystick [PLAYSTATION(R)3 Controller] on 0
0:17:e7:b4:71:28
[597914.089542] sony 0005:054C:0268.0003: controller with MAC address 04:98:f3:a3:1c:eb already connected
[597914.308959] sony: probe of 0005:054C:0268.0003 failed with error -17
[597919.207997] usb 1-1.4.4: USB disconnect, device number 6
```

dlech, 2017-10-24:
After disconnecting and re-connecting via Bluetooth it is working. Apparently, during the pairing process when it is connected via USB and Bluetooth at the same time, it tries to register the Bluetooth connection with the same name as the USB connection.

dlech, 2017-10-25:
Might need to pick up this patch: https://github.com/torvalds/linux/commit/4f967f6d73746f66514528cc1191025f0b5d69b3#diff-ea8a58a0802dc8df05d5caf91e2c7934

dlech, 2018-04-20:
@dwalton76, can you try again with the 4.14.x kernel?

dwalton76, 2019-03-25:
Sorry for the slow (almost 1 year) reply :(  Will give it a try.

dwalton76, 2019-03-26:
My PS3 controller is dead :(  Does anyone have one they can use to test this?

dlech, 2019-03-28:
The workaround still works for me. I had to press and hold the PS button for a while to get the authorization box to pop up.

yosefl20, 2019-10-21:
Hey guys. I got a SHANWAN clone PS3 gamepad which as same as the post mentioned. And I'm using the kernel 4.14.96. It still doesn't work.
the logs of dmesg I got like this:
[ 3899.490319] usb 1-1.2: new full-speed USB device number 16 using ohci-da8xx
[ 3899.651162] usb 1-1.2: New USB device found, idVendor=054c, idProduct=0268
[ 3899.651235] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 3899.651273] usb 1-1.2: Product: PS3 GamePad
[ 3899.651307] usb 1-1.2: Manufacturer: SHANWAN
[ 3899.762446] input: SHANWAN PS3 GamePad Motion Sensors as /devices/platform/soc@1c00000/ohci-da8xx/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0009/input/input19
[ 3899.851386] input: SHANWAN PS3 GamePad as /devices/platform/soc@1c00000/ohci-da8xx/usb1/1-1/1-1.2/1-1.2:1.0/0003:054C:0268.0009/input/input18
[ 3899.853847] sony 0003:054C:0268.0009: input,hidraw0: USB HID v81.10 Joystick [SHANWAN PS3 GamePad] on usb-ohci-da8xx-1.2/input0
[ 3901.853082] usb 1-1.2: USB disconnect, device number 16

It's a clone of PS3, not manufactured by SONY.

denorlov, 2019-11-16:
Tryied to pair SHANWAN PS3 Gamepad with latest ev3dev (ev3dev-stretch-ev3-generic-2019-10-23). Without any luck. When connected over USB - PS3 vibrates. Also wasn't able to come to situation when ev3dev ask about authorisation