ev3dev/ev3dev #969: Installing custom kernel on EV3 using Debian packaging

oknystad, 2017-09-26:
### What are you trying to do?
Installing a custom-built kernel on EV3 using Debian packaging.

### What did you expect to happen?
After following instructions on https://github.com/ev3dev/ev3dev-buildscripts/tree/ev3dev-jessie,  ending up with a .deb package file that I can run on the EV3 to install the new kernel.

### What actually happened?
Received various error messages during packaging (see below), no .deb file is created.

### What hardware and software are you using (including version numbers)?
Windows 10 Subsystem Linux w/ Ubuntu 16.04.

As described in https://github.com/ev3dev/ev3dev-buildscripts/tree/ev3dev-jessie - First kernel build, I have cloned ev3-kernel repo (kernel version 4.4.87-22) and made a simple change in one file (in `board-legoev3.c`, changed `.rotate = 270` to `.rotate = 90`, in order to change orientation of an Adafruit 1,8" display). Building the kernel works fine, and `/build_area` is created.

For the Debian packaging, I've set local version to `-oknystad-ev3` via `./menuconfig`. Command `./build-kernel deb-pkg KDEB_PKGVERSION=1` gives the output:
```
-------------------------------------------------------------------------------
BUILDING KERNEL
-------------------------------------------------------------------------------

CROSS COMPILING KERNEL
make: Entering directory '/mnt/c/okn-work/ev3-kernel'
make[1]: Entering directory '/mnt/c/okn-work/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
  CHK     include/config/kernel.release
make clean
  TAR     linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+.tar.gz
Building source tarball is not possible outside the
kernel source tree. Don't set KBUILD_OUTPUT, or use the
binrpm-pkg or bindeb-pkg target instead.
tar: linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+/.config: Cannot stat: No such file or directory
tar: linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+/.scmversion: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
make KBUILD_SRC=
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  Using /mnt/c/okn-work/ev3-kernel as source for kernel
  CHK     include/generated/utsrelease.h
make[6]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    /mnt/c/okn-work/ev3-kernel/scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  SKIPPED include/generated/compile.h
  CHK     kernel/config_data.h
  Kernel: arch/arm/boot/Image is ready
  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
  MODPOST 205 modules
...
<lots of CC and LD [M] output>
...
  BUILDDEB
fakeroot, while creating message channels: Function not implemented
This may be due to a lack of SYSV IPC support.
fakeroot: error while starting the `faked' daemon.
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
/mnt/c/okn-work/ev3-kernel/scripts/package/Makefile:91: recipe for target 'deb-pkg' failed
make[2]: *** [deb-pkg] Error 1
/mnt/c/okn-work/ev3-kernel/Makefile:1252: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 2
make[1]: Leaving directory '/mnt/c/okn-work/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
Makefile:150: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
make: Leaving directory '/mnt/c/okn-work/ev3-kernel'
```
Suggestions on how to eliminate the errors? Trying to clean up with `git clean -dfX` does not make a difference.

WasabiFan, 2017-09-26:
Ooh, I think this is the first post I've seen since we added that note about WSL to the instructions! I'm going to try out exactly the same steps and see if the same issue occurs for me. This WSL suggestion is pretty much just an experiment at this point :laughing: Of course, @dlech is the one who knows about what's actually going on here.

WasabiFan, 2017-09-26:
I'm going to bet it's this: https://github.com/Microsoft/BashOnWindows/issues/2465

Not sure if @dlech is aware of a workaround.

dlech, 2017-09-26:
Try setting the `DEB_BUILD_OPTIONS` environment variable.

    DEB_BUILD_OPTIONS="-rsudo" ./build-kernel deb-pkg KDEB_PKGVERSION=1

WasabiFan, 2017-09-26:
```
$ DEB_BUILD_OPTIONS="-rsudo" ./build-kernel deb-pkg KDEB_PKGVERSION=1

-------------------------------------------------------------------------------
BUILDING KERNEL
-------------------------------------------------------------------------------

CROSS COMPILING KERNEL
make: Entering directory `/home/wasabifan/ev3-kernel'
make[1]: Entering directory `/home/wasabifan/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
  CHK     include/config/kernel.release
make clean
  TAR     linux-4.4.87-22-ev3dev-ev3+.tar.gz
Building source tarball is not possible outside the
kernel source tree. Don't set KBUILD_OUTPUT, or use the
binrpm-pkg or bindeb-pkg target instead.
tar: linux-4.4.87-22-ev3dev-ev3+/.config: Cannot stat: No such file or directory
tar: linux-4.4.87-22-ev3dev-ev3+/.scmversion: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
make KBUILD_SRC=
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  Using /home/wasabifan/ev3-kernel as source for kernel
  CHK     include/generated/utsrelease.h
make[6]: `include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    /home/wasabifan/ev3-kernel/scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  Kernel: arch/arm/boot/Image is ready
  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
  MODPOST 205 modules
  BUILDDEB
fakeroot, while creating message channels: Function not implemented
This may be due to a lack of SYSV IPC support.
fakeroot: error while starting the `faked' daemon.
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
make[2]: *** [deb-pkg] Error 1
make[1]: *** [deb-pkg] Error 2
make[1]: Leaving directory `/home/wasabifan/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
make: *** [sub-make] Error 2
make: Leaving directory `/home/wasabifan/ev3-kernel'
```

dlech, 2017-09-26:
Hmm... there doesn't seem to be an easy way to work around it. I suppose you could run the whole build as root.

    sudo ./build-kernel deb-pkg KDEB_PKGVERSION=1

WasabiFan, 2017-09-27:
Yeah, that seems to do it.

oknystad, 2017-09-27:
Somehow it seems that your sudo is more powerful than mine @WasabiFan :wink:. It still fails to create the .deb file when I run `sudo ./build-kernel deb-pkg KDEB_PKGVERSION=1`. Printout:

```
-------------------------------------------------------------------------------
BUILDING KERNEL
-------------------------------------------------------------------------------

CROSS COMPILING KERNEL
make: Entering directory '/mnt/c/okn-work/ev3-kernel'
make[1]: Entering directory '/mnt/c/okn-work/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
  CHK     include/config/kernel.release
make clean
  TAR     linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+.tar.gz
Building source tarball is not possible outside the
kernel source tree. Don't set KBUILD_OUTPUT, or use the
binrpm-pkg or bindeb-pkg target instead.
tar: linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+/.config: Cannot stat: No such file or directory
tar: linux-4.4.87-22-ev3dev-ev3-oknystad-ev3+/.scmversion: Cannot stat: No such file or directory
tar: Exiting with failure status due to previous errors
make KBUILD_SRC=
  CHK     include/config/kernel.release
  GEN     ./Makefile
  CHK     include/generated/uapi/linux/version.h
  Using /mnt/c/okn-work/ev3-kernel as source for kernel
  CHK     include/generated/utsrelease.h
make[6]: 'include/generated/mach-types.h' is up to date.
  CHK     include/generated/timeconst.h
  CHK     include/generated/bounds.h
  CHK     include/generated/asm-offsets.h
  CALL    /mnt/c/okn-work/ev3-kernel/scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CHK     kernel/config_data.h
  Kernel: arch/arm/boot/Image is ready
  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
  Kernel: arch/arm/boot/zImage is ready
  MODPOST 205 modules
  BUILDDEB
...
lots of INSTALL output
...
dpkg-deb: error: control directory has bad permissions 777 (must be >=0755 and <=0775)
/mnt/c/okn-work/ev3-kernel/scripts/package/Makefile:91: recipe for target 'deb-pkg' failed
make[2]: *** [deb-pkg] Error 2
/mnt/c/okn-work/ev3-kernel/Makefile:1252: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 2
make[1]: Leaving directory '/mnt/c/okn-work/ev3dev-buildscripts/build-area/linux-ev3dev-ev3-obj'
Makefile:150: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
make: Leaving directory '/mnt/c/okn-work/ev3-kernel'
```

oknystad, 2017-09-27:
Running `DEB_BUILD_OPTIONS="-rsudo" ./build-kernel deb-pkg KDEB_PKGVERSION=1` produced the exact same result for me as for @WasabiFan, well, except for an additional line `SKIPPED include/generated/compile.h` in my case.

oknystad, 2017-09-27:
Yes, I did. I re-did cloning of the repo and compilation (without any changes made to code) just now, with same result as before when building Debian package. No luck with the commands:
```
/mnt/c/okn-work/ev3-kernel$ git checkout debian/
error: pathspec 'debian/' did not match any file(s) known to git.

/mnt/c/okn-work/ev3-kernel$ chmod 755 debian/control
chmod: cannot access 'debian/control': No such file or directory
```

WasabiFan, 2017-09-27:
Looks like my guess for where the package directory was is incorrect... I guess it's generated into the `build-area`. I have no idea how that works, so this seems to be a @dlech question 😆

dlech, 2017-09-27:
The debian directory is created as part of the build process. I think was @WasabiFan is getting at is that you are building in `/mnt/c/okn-work/ev3-kernel`, which is presumably using NTFS, which is not case sensitive and does not support POSIX file permissions. If you clone using `git` in WSL to a path in WSL not under `/mnt/`, like `/home/$USER/work/`, then it should work better.

oknystad, 2017-09-28:
Ah, I see. 
That does seem like a plausible explanation - I will test it within a couple of days and report back :relaxed:.

oknystad, 2017-10-01:
Running `sudo ./build-kernel deb-pkg KDEB_PKGVERSION=1` from a local Linux directory under `$HOME `worked just fine, and all is in order (i.e. the Adafruit display har correct orientation) after installing the .deb image on the EV3.
Thanks for all help!

[image]

closed 2017-10-01 by oknystad

antonvh, 2018-01-10:
sudo worked for me too. I will do a PR for the readme.
- you might want to mention that you can't use capitals in the local version name.
- just running the buildscript without the debian packaging worked fine.
- AFAK there is - alas - no way to access USB directly from WSL. So having a preformatted linux SD card with a boot and file partition is no go.
- I will try to find out how to use docker and a docker sd image to get around this.

oknystad, 2018-01-27:
> AFAK there is - alas - no way to access USB directly from WSL.

@antonvh: USB devices work for me in Ubuntu 16.04 on Windows WSL (tested with USB stick and USB SD card adapter). I.e., if Windows drive for USB device is `D:`, then
```
sudo mkdir /mnt/d
sudo mount -t drvfs D: /mnt/d
ls /mnt/d
```
... lists out the contents correctly.

It seems Windows OS build 16176 or newer is required. [Link](https://stackoverflow.com/a/44001783)

antonvh, 2018-02-03:
Wel yes, that works for simply accessing files. But if you need to make special Linux partitions it doesn't.