ev3dev/ev3dev #970: Stock Images and Sounds

dlech, 2017-09-30:
### What are you trying to do?

I've started a new [ev3dev-media](https://github.com/ev3dev/ev3dev-media) repository that will serve as a collection of stock images and sounds that will be distributed with ev3dev-stretch.

So far, I have a bunch of sounds that approximate most the sounds that come with EV3-G. There are a few missing in the mechanical, movements and system categories, so if you are an aspiring sound effects artist, this could be a good opportunity for you.

Also I don't have any images yet, so if you are an aspiring graphics artist, this could be a good job for you. I think imitating the images from EV3-G would be a good start, but I don't want to copy them too exactly so we don't get into any copyright trouble. For example, the [ev3go project uses the Go Gopher](https://github.com/ev3go/ev3dev/tree/master/examples/demo#demo) for the "Eyes" in EV3-G. I would like to see something creative like that.

dlech, 2017-09-30:
List of progress can be found [here](https://github.com/ev3dev/ev3dev-media/blob/ev3dev-stretch/ev3g-equivalents.md).

WasabiFan, 2017-09-30:
Would I be correct to assume that there is a legal reason that we can't use LEGO's official media?

dlech, 2017-09-30:
They are distributed with the EV3 desktop software and not part of the open-source firmware, so this license applies: https://education.lego.com/en-us/support/mindstorms-ev3/software-license

I has a long list of things you can't do.

That is not to say that someone could write to LEGO and ask permission.

dlech, 2017-09-30:
There is a visual reference for the images in the EV3 Users Guide available on the LEGO MINDSTORMS download page: https://www.lego.com/en-us/mindstorms/downloads

dlech, 2017-09-30:
Color versions of the images would be cool too for EVB/PiStorms.

ndward, 2017-12-31:
On page 3 of the EV3 End User Licence Agreement at https://education.lego.com/en-us/support/mindstorms-ev3/software-license I read that '**The text, building instructions and images within the Software are the property of LEGO System and/or its suppliers and licensors. However you may use these for your own non-commercial purposes.' Doesn't that mean we can use the BMP images included in the software for non-commercial purposes?**
For the sound files, it's different since they are not mentioned in the same way and since the .RSF (robot sound file) format of the sound files in the software is not compatible with ev3dev so either ev3dev would need to be made compatible with that format or the files would need to be converted to .WAV. (I understand ev3dev is not compatible with .MP3?)
How do other groups such as LeJos deal with this problem of needing to be able to use standard Lego EV3 images and sounds?

dlech, 2017-12-31:
> Doesn't that mean we can use the BMP images included in the software for non-commercial purposes?

Perhaps.

> I understand ev3dev is not compatible with .MP3?

ev3dev can play virtually any sound format out there. We don't use MP3 normally because it takes a massive amount of CPU to decode them. It is also easy enough to play RSF files. But they are low quality (8kHz), so we can do better.

> How do other groups such as LeJos deal with this problem of needing to be able to use standard Lego EV3 images and sounds?

It think they just don't include it.

dlech, 2017-12-31:
By the way, I'll send a couple of ev3dev laptop stickers to anyone who completes one of the remaining categories. As of now, there are 6 image categories and 1 sound category left.