ev3dev/ev3dev #971: apt update hanging due to zero sized file

kortschak, 2017-09-30:
### What are you trying to do?
Update the distribution using `sudo apt update`

### What did you expect to happen?
Successful completion of command.

### What actually happened?
Hang with 

```
$ sudo apt update
[sudo] password for robot: 
Ign http://ftp.debian.org jessie InRelease                                  
Hit http://ftp.debian.org jessie Release.gpg                        
Hit http://ftp.debian.org jessie Release
Hit http://ev3dev.org jessie InRelease
Get:1 http://ftp.debian.org jessie/main armel Packages [6,645 kB]           
Hit http://ev3dev.org jessie/main armel Packages         
Ign http://ev3dev.org jessie/main Translation-en_US                            
Ign http://ev3dev.org jessie/main Translation-en                               
Get:2 http://ftp.debian.org jessie/contrib armel Packages [36.7 kB]            
Get:3 http://ftp.debian.org jessie/non-free armel Packages [59.7 kB]           
Hit http://ftp.debian.org jessie/contrib Translation-en                        
Get:4 http://ftp.debian.org jessie/main Translation-en [4,582 kB]              
Get:5 http://ftp.debian.org jessie/non-free Translation-en [72.1 kB]           
100% [4 Translation-en bzip2 0 B] [1 Packages xz 0 B]                          
```

This appears to be due to zero sized files on the repository. This may be debian's fault.

### What hardware and software are you using (including version numbers)?

EV3 rev 6
```
$ uname -a
Linux ev3dev 4.4.47-19-ev3dev-ev3 #1 PREEMPT Wed Feb 8 14:15:28 CST 2017 armv5tejl GNU/Linux
```

dlech, 2017-09-30:
Are you sure it is hung up? `apt update` can take up to 10 minutes depending on how fast your SD card is and what else might be running in the background.

dlech, 2017-09-30:
You can start a second SSH terminal an run `top` to see if there is actually something running or not by looking at the CPU usage.

kortschak, 2017-09-30:
I think you might be right, xz and bzip2 are very busy. I looked like an old debian bug where zero byte files cause this behaviour. I'll leave it for ~ minutes and close this if it finishes.

kortschak, 2017-09-30:
Sorry for the noise. It has now completed. I forget how slow these devices can be living most of my life on HPC and compute farms.

closed 2017-09-30 by kortschak