ev3dev/ev3dev #973: EV3 restart cannot open Wireless and Networks after enable and start NFS.

superPershing, 2017-10-01:
I follow the <http://www.ev3dev.org/docs/tutorials/setting-up-an-nfs-file-share/> to enable and start NFS to share files between my computer and the EV3. But after I restart my ev3 robot, I find I cannot open `Wireless and Networks`, it keep show `loading....`. I write my micro SD card once and once again. I find that once I enable and start NFS on my ev3, I will cannot open `Wireless and Networks` after reboot. If I do not do anything about NFS or disable NFS, it will work normally. I have open and closed an issue before:[image], at that time, I didn't know that the NFS service cause this problem. But now, after more than 20 times tests(rewrite my micro SD card), I found the NFS was the source of this bug!
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```
I use the EW-7811Un wifi dongle.

dlech, 2017-10-01:
Does the problem also happen on the ev3dev-stretch snapshot images?

dlech, 2017-10-01:
What if you create a systemd mount unit file instead of following the tutorial on the ev3dev website?

https://cloudnull.io/2017/05/nfs-mount-via-systemd/

JesApp, 2017-11-17:
In all likelihood, this is the same problem as discussed in #905 .

dlech, 2017-11-19:
Please follow the updated tutorial using a systemd unit file instead of fstab.

http://www.ev3dev.org/docs/tutorials/setting-up-an-nfs-file-share/

closed 2017-11-19 by dlech