ev3dev/ev3dev #974: Issue auto-detecting ms-absolute-imu

laurensvalk, 2017-10-01:
### What are you trying to do?

Connect the [Absolute IMU](http://www.mindsensors.com/ev3-and-nxt/15-gyro-multisensitivity-accelerometer-and-compass-for-nxt-or-ev3)  and read values from it.

Ultimately using the Python library, but it looks like I need to take a step back first.

### What did you expect to happen?

Plugging in the sensor and have it auto-detected.

### What actually happened?

It registers, then unregisters. This is dmesg after plugging the sensor some time after boot, doing nothing else.

```
[  129.914707] lego-port port2: Added new device 'ev3-ports:in3:nxt-i2c-host'
[  130.805261] lego-sensor sensor0: Registered 'ms-absolute-imu' on 'i2c-legoev35:i2c17'.
[  136.299013] lego-port port2: Removed device 'ev3-ports:in3:nxt-i2c-host'
[  136.312948] lego-sensor sensor0: Unregistered 'ms-absolute-imu' on 'i2c-legoev35:i2c17'.
[  139.983720] lego-port port2: Added new device 'ev3-ports:in3:nxt-i2c-host'
```

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-stretch-ev3-generic-2017-09-14
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653420A17
BOARD_INFO_TYPE=main
```

dlech, 2017-10-01:
Mine seems to be working, so let's go with the theory that you have changed the address or that you have a broken firmware.

Step 1 (this assumes the sensor is still connected to input port 3 - you may also need to put the port into other-i2c mode):

    sudo i2cdetect -y 5

you should see something like this

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```

If it looks different than this, then we have confirmed the changed address theory.

Step 2: 

    sudo i2cdump -y -f -r 0x00-0x17 5 0x11

This will tell us the firmware version. It should look like this:

```
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 31 2e 32 30 00 00 00 6d 6e 64 73 6e 73 72 73    V1.20...mndsnsrs
10: 41 62 73 49 4d 55 00 00                            AbsIMU.. 
```

laurensvalk, 2017-10-01:
``` 
robot@ev3dev:~$ sudo i2cdetect -y 5
    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:    -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
robot@ev3dev:~$ sudo i2cdump -y -f -r 0x00-0x17 5 0x11
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 31 2e XX XX XX 00 00 6d 6e 64 7e XX 00 72 73    V1.XXX..mnd~X.rs
10: 41 62 73 XX XX XX XX XX                            AbsXXXXX
```

Thanks for the help. I'm not sure what to make of this. It doesn't seem to be at 0x11, or does it?

laurensvalk, 2017-10-01:
Well, it gets more interesting if I repeat the last step

```
robot@ev3dev:~$ sudo i2cdump -y -f -r 0x00-0x17 5 0x11
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: 56 XX XX XX 34 33 00 00 XX XX XX XX XX 73 72 73    VXXX43..XXXXXsrs
10: 41 62 XX XX XX XX XX XX                            AbXXXXXX        
robot@ev3dev:~$ sudo i2cdump -y -f -r 0x00-0x17 5 0x11
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX XX 2e 30 34 33 00 00 6d XX XX XX XX XX XX 00    XX.043..mXXXXXX.
10: 41 62 73 49 XX XX XX XX                            AbsIXXXX        
robot@ev3dev:~$ sudo i2cdump -y -f -r 0x00-0x17 5 0x11
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX 31 2e 30 34 33 ff XX XX XX XX 49 6e 73 72 73    X1.043.XXXXInsrs
10: XX XX XX 20 4d 55 00 00                            XXX MU..        
robot@ev3dev:~$ sudo i2cdump -y -f -r 0x00-0x17 5 0x11
No size specified (using byte-data access)
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f    0123456789abcdef
00: XX 31 2e XX 34 33 00 00 6d XX XX XX XX 73 72 73    X1.X43..mXXXXsrs
10: 41 62 73 48 XX XX XX XX                            AbsHXXXX        
```
I guess I have FW 1.043, but there might be something else causing the incoherent output. 

I'm going to try and see if the cable might be broken.

laurensvalk, 2017-10-01:
The output changes in between repetition of the command even when the cable and sensor are stationary. 

Have you ever attempted to open up a Mindsensors sensor?

dlech, 2017-10-01:
I haven't opened one, but I think the covers are just glued on.

Also, you could try using an NXT to upgrade the firmware on the sensor and then try again. You might have to ask mindsensors for the firmware file.

laurensvalk, 2017-10-01:
> I think the covers are just glued on.

Confirmed!

laurensvalk, 2017-10-01:
> Also, you could try using an NXT to upgrade the firmware on the sensor and then try again.

I'll have to try this later. My NXTs are currently in a storage.

laurensvalk, 2017-10-01:
As this seems to be specific to my sensor, I'll close this issue.

laurensvalk, 2017-10-01:
But it is working fine (auto-detecting and reporting values) on this EV3 after I put in the very same micro SD card.

```
Image file:         ev3dev-stretch-ev3-generic-2017-09-14
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016533DDB29
BOARD_INFO_TYPE=main
```

... and now on that first EV3, too.

dlech, 2017-10-01:
So maybe you fixed that loose/broken wire?

laurensvalk, 2017-10-07:
I'm not entirely sure. I'll close this issue until I find out more.

closed 2017-10-07 by laurensvalk