ev3dev/ev3dev #975: TimeoutError when setting mode for Gyro sensor

dwalton76, 2017-10-01:
### What are you trying to do?
When trying to set the mode on the Gryo sensor via:
```
robot@evb[bin]# cat gyro.py 
#!/usr/bin/env python3

from ev3dev.auto import GyroSensor
from pprint import pformat
import logging
import sys
import time

# logging
logging.basicConfig(level=logging.DEBUG,
                    format="%(asctime)s %(levelname)5s: %(message)s")
log = logging.getLogger(__name__)

gy = GyroSensor()
#gy.reset()

while True:
    print("angle %s" % gy.angle)
    print("rate %s" % gy.rate)
    (angle, rate) = gy.rate_and_angle
    print("rate %s, angle %s" % (rate, angle))
    time.sleep(1)
robot@evb[bin]# 
```

### What did you expect to happen?
I expect it to set the mode :)

### What actually happened?
It barfs due to
```
robot@evb[bin]# ./gyro.py 
Traceback (most recent call last):
  File "./gyro.py", line 20, in <module>
    print("angle %s" % gy.angle)
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 2622, in angle
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 2091, in mode
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 291, in set_attr_string
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 251, in _set_attribute
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 275, in _raise_friendly_access_error
  File "/usr/local/lib/python3.5/dist-packages/python_ev3dev-0.8.0.post72-py3.5.egg/ev3dev/core.py", line 247, in _set_attribute
TimeoutError: [Errno 110] Connection timed out
robot@evb[bin]# 
```
Some more details over in https://github.com/rhempel/ev3dev-lang-python/pull/387

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-bone-generic-2017-09-01
Kernel version:     4.9.50-ev3dev-1.5.1-bb.org
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_MODEL=FatcatLab EVB
BOARD_INFO_TYPE=aux
Board:              board1
BOARD_INFO_HW_REV=000C
BOARD_INFO_MODEL=A335BNLT
BOARD_INFO_SERIAL_NUM=4214BBBK2552
BOARD_INFO_TYPE=main
```

My gryo is version `O3N3`

dlech, 2017-10-01:
Thanks. This appears to only affect older gyro sensors (identified by the manufacturing stamp ending in 3)

also, from my initial test, it seem that it could be related to the GYRO-G&A mode, e.g. once you switch to that mode, then you are stuck in that mode.

the problem is also reproducible from brickman by using the device browser.

dlech, 2017-10-02:
I'm marking this `wontfix` because these sensors are defective.

Once the sensor is in `GYRO-G&A` mode, there is no getting it out of that mode. In the official EV3-G software the sensor resets because the software stops sending keepalives.

I'm thinking we leave this as-is so that people know that their sensor is defective and intentionally work around it.

If a program only uses `GYRO-G&A` mode, then set the mode once and then don't worry about it.

If a program really needs to use `GYRO-G&A` and then switch to another mode, use the `lego-port` device to reset the sensor by changing the mode to `other-uart` and then back to `auto`.

closed 2017-10-02 by dlech