ev3dev/ev3dev #976: PIXY CAMERA TAKE PHOTO AND SAVE

Negociation, 2017-10-02:
Hi , my name is marcos and i'm a student of computer science on UTFPR ( Universidade Tecnológica Federal do Paraná ). 

I will make my graduation project using LEGO + EV3DEV Python , i wanna build a Self Driving Car using Lego + Neural Network . I Will use a Socket TCP conection with my Windows Tablet to processing data and send commands to robot execute. I will use the Netgear N150 + ADHOC Network ... OK

1) Starting the Data-set 
- I will get the data in 1 second of the Ultrassom Sensor , Color Sensor , Gyro *Probably , a Image of the Front of Robot using PiXY Camera

- My first issue is about the PIXY Camera , i wanna use it to take a photo and save on PNG/TIFF... and sent to my tablet that data. 

- I saw some examples to work with the camera data to follow objects , draw in lego display... , but i need help to take a picture with camera and save on a file , someone can help-me with that ?

dlech, 2017-10-02:
Have a look at http://cmucam.org/boards/9/topics/5106?page=1&r=7910

You will have to attach the camera via USB. Ev3dev ships with a python version of libpixyusb if you want to use python instead of C/C++.

Negociation, 2017-10-03:
Farts , i will use the USB port to Wireless Dongle... , u have another ideia ?

dwalton76, 2017-10-03:
I use these with my ev3

https://www.amazon.com/Sabrent-4-Port-Individual-Switches-HB-UMLS/dp/B00BWF5U0M/ref=sr_1_5?s=pc&ie=UTF8&qid=1506993526&sr=1-5&keywords=usb+hub

gmollard, 2018-10-03:
Hi @Negociation ! Did you finally manage to use the usb connection of the ev3 and the library libpixyusb to send an image to your tablet ? I want to do more or less the same thing but do not really know if it is possible...

Negociation, 2018-10-03:
Man , i give up a long time ago... , Now i'm move from a EV3 to Raspberry Pi with BrickPI , and to mount my dataset i recorder the video using my smartphone , and extracting the frames from a timestamps .txt generated by my robot example 1:30:13 --> PROXIMITY 25 so i extract the frame using FFMPEG library...

Now for running the network on realtime , (Probably i want to do that) i'm using the same device as a IP Camera , and i get the frames using opencv on my robot. :D

closed 2020-08-17 by dlech