ev3dev/ev3dev #978: pistorm LEDs

dwalton76, 2017-10-04:
### What are you trying to do?
Trying to get LEDs working on pistorms via ev3dev-lang-python

### What actually happened?
Two issues

#### First issue
The `s` in `status` is missing for both `green` LEDs
```
robot@pistorms[ev3dev-lang-python]# ls /sys/class/leds/
led0  pistorms:BA:blue:brick-status  pistorms:BA:red:brick-status   pistorms:BB:green:brick-statu  rt2800usb-phy0::assoc    rt2800usb-phy0::radio
led1  pistorms:BA:green:brick-statu  pistorms:BB:blue:brick-status  pistorms:BB:red:brick-status   rt2800usb-phy0::quality
robot@pistorms[ev3dev-lang-python]# 
```

#### Second issue
The group is `root` instead of `ev3dev` for several files and the `brightness` file is not writeable except for root
```
robot@pistorms[ev3dev-lang-python]# ls -l /sys/class/leds/pistorms:BB:green:brick-statu/
total 0
-rw-r--r-- 1 root root 4096 Oct  4 02:50 brightness
lrwxrwxrwx 1 root root    0 Oct  4 02:49 device -> ../../../1-001b
-r--r--r-- 1 root root 4096 Oct  4 02:49 max_brightness
drwxr-xr-x 2 root root    0 Oct  4 02:49 power
lrwxrwxrwx 1 root root    0 Oct  4 02:49 subsystem -> ../../../../../../../../class/leds
-rw-r--r-- 1 root root 4096 Oct  4 02:49 trigger
-rw-r--r-- 1 root root 4096 Oct  4 02:49 uevent
robot@pistorms[ev3dev-lang-python]# 
robot@pistorms[ev3dev-lang-python]# 
```

on an EV3 we see
```
robot@ev3dev[ev3dev-lang-python]# ls -l /sys/class/leds/led0:green:brick-status/
total 0
-rw-rw-r-- 1 root ev3dev 4096 Sep 30 19:45 brightness
lrwxrwxrwx 1 root root      0 Oct  4 02:47 device -> ../../../leds
-r--r--r-- 1 root ev3dev 4096 Sep 30 19:42 max_brightness
drwxr-xr-x 2 root root      0 Oct  4 02:37 power
lrwxrwxrwx 1 root root      0 Oct  4 02:47 subsystem -> ../../../../../class/leds
-rw-rw-r-- 1 root ev3dev 4096 Sep 30 19:45 trigger
-rw-rw-r-- 1 root ev3dev 4096 Sep 30 19:42 uevent
robot@ev3dev[ev3dev-lang-python]# 
```

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-09-30
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A21041
BOARD_INFO_MODEL=Raspberry Pi 2 Model B Rev 1.1
BOARD_INFO_SERIAL_NUM=00000000dd20fe9a
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=V2.06
BOARD_INFO_MODEL=PiStorms
BOARD_INFO_TYPE=aux
```

dlech, 2017-10-04:
FYI, second issue is a consequence of the first issue. Only LEDs that end in `brick-status` trigger the udev rule that sets permissions.

dlech, 2017-11-01:
Fix released in `ev3dev-1.6.0` kernel.

closed 2017-11-01 by dlech