ev3dev/ev3dev #979: Unified Java Library for ev3dev

mob41, 2017-10-04:
Hi,

I am sorry for hanging this library for a long time. I have finally completed the library implementation and wiki. May I ask that can I open a pull request to update the official ev3dev page to add about this library?

By the way, my source code seemed to be stolen and used in another library without proper permission. I am disappointed about this. How can I solve this problem?

P.S. https://github.com/mob41/ev3dev-lang-java

Regards,
mob41

dlech, 2017-10-04:
> May I ask that can I open a pull request to update the official ev3dev page to add about this library?

You can ask. But it is better if you just to it. :wink:

> By the way, my source code seemed to be stolen and used in another library without proper permission. I am disappointed about this. How can I solve this problem?

I would suggest that you contact the author of that library and politely point out that they are not complying with the license and offer a suggestion on how they can fix the problem.

markyi370, 2017-10-05:
@mob41 May I ask what library infringed on your license?

dlech, 2017-10-24:
https://github.com/ev3dev/ev3dev.github.io/commit/f5f69329f526c6cc75bb9325a5697cba31f81be5

closed 2017-10-24 by dlech