ev3dev/ev3dev #980: Barometric sensor in TEMP mode returns 0 as temperature

instance01, 2017-10-06:
### What are you trying to do?
Using sysfs I'm trying to read the current temperature on the barometric sensor. After setting the mode to TEMP, value0 continuously returns 0. All other values are undefined (as expected).
PRESS mode does return a value, it's around 91000, but it's not as defined on the wiki (http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#hitechnic-nxt-barometric-sensor).

### What did you expect to happen?
Expected to see room temperature.

### What hardware and software are you using (including version numbers)?

I'm using this barometric sensor: http://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NBR1036

On the normal Lego mindstorms OS the sensor works fine.

Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

instance01, 2017-10-11:
**Workaround**

I followed the documentation and I can read i2c values like this:

Given your sensor is plugged into port in1.

Disable autodetection of sensors:
echo -n N > /sys/module/nxt_i2c_sensor/parameters/allow_autodetect

For temperature:
i2cget 3 0x01 0x42
i2cget 3 0x01 0x43
For pressure:
i2cget 3 0x01 0x44
i2cget 3 0x01 0x45

If you have more sensors, you'll have to load them all manually with this workaround (because auto detection is off). E.g. acceleration sensor on port in2:

echo ht-nxt-accel 0x01 > /sys/bus/i2c/devices/i2c-4/new_device

dlech, 2017-10-14:
FYI, you can disable autodetection for a single port by putting it in `other-i2c` mode instead of using the module parameter that affects all ports.

It looks like the driver definition is incorrect. Currently both values use the 0x42 address and are 8-bit values, but pressure needs to be changed to 0x44 and both values need to be changed to 16-bit. 

https://github.com/ev3dev/lego-linux-drivers/blob/ev3dev-stretch/sensors/nxt_i2c_sensor_defs.c#L690

There is currently not enough data to tell if the values are big-endian or little-endian.

dlech, 2017-11-01:
Fix released in the `ev3dev-1.6.0` kernel. I don't have one of these sensor to test, so it may or may not actually be fixed correctly. Please grab the latest snapshot for ev3dev-stretch to test. 

https://oss.jfrog.org/artifactory/list/oss-snapshot-local/org/ev3dev/brickstrap/

closed 2017-11-01 by dlech