ev3dev/ev3dev #982: Crash due to hot plugging sensors

dedady157, 2017-10-09:
### What are you trying to do?
looking how sensors are connected to the device
ie the name thay are given

### What did you expect to happen?
it to go back to sensor0

### What actually happened?
it crashed

### What hardware and software are you using (including version numbers)?
ev3dev lego mindstorm

         brickman v0.8.1
 The ev3dev Brick Manager
 (C) 2014-2015 ev3dev.org
             SystemInfo
----------------------------
                Kernal:
4.4.87-22-ev3dev-ev3
                model:
LEGO MINDSTORMS EV3
  programmable Brick
               Reversion:
                  0006
            Serial Number:
***************************

dlech, 2017-10-09:
It's not really clear what is crashing. Can you give more detail on what exactly you are trying do to?

dlech, 2017-10-10:
> every time i unpluged it and pluged it back in it got a different ID

This is normal

> i keept doing this untill i reached 4 or 5 then it crashed

Was there an error message?

> i wanted to see if it would loop back to 0 or keep going

I'm sure it will if you do this over 4 billion times, but not before then.

dedady157, 2017-10-10:
There was none it just froze up and didn't start working agen

dedady157, 2017-10-10:
Miss clicked I'm on phone cuz internet's down

dlech, 2018-04-15:
Does this crash happen when you use ev3dev-stretch?