ev3dev/ev3dev #984: BALANC3R / Segway code updates

laurensvalk, 2017-10-12:
I've recently made some updates to the [ev3dev Python code for the BALANC3R Segway](https://github.com/laurensvalk/segway/) based on user feedback.

In the past, I believe some of you have tried it out with mixed results. With the recent changes, performance should be much more consistent across (y)our robots.

Biggest changes:
- Scaling of duty cycle with battery level
- Using the wall clock instead of the process clock to time the loop
- Use ev3dev-lang-python instead of a workaround to find the hardware files (thanks to @dwalton76)
- You can now tune the parameters without restarting the program. Just update parameters.py. Very easy with the VS Code IDE. (However, with the above changes, you may not need to tune much, if at all.)
- Added IR control example
- Made acceleration less abrupt, for smoother movement. 

If you want to try it out, the code and links to the building instructions are [here](https://github.com/laurensvalk/segway/). It's a pretty quick build.

If you try it out, I'd love to hear how it goes.

ddemidov, 2017-10-13:
Will this work with [AbsoluteIMU](http://www.mindsensors.com/ev3-and-nxt/15-gyro-multisensitivity-accelerometer-and-compass-for-nxt-or-ev3) sensor?

laurensvalk, 2017-10-13:
If you can read one gyro rate axis in under 30 ms, it might work straight away.

I think I can adapt the settings and stretch the loop to 50 ms if the sensor needs more time.

They're using that sensor [with an old version of my code on the NXT](https://www.youtube.com/watch?time_continue=19&v=CTMntaMlgi0), so the sensor should be suitable for balancing.

I think you'll want to disable the sensor's internal low pass filter (or set it to its lowest value) because the code does its own filtering. Filtering on the sensor only results in more delay.

laurensvalk, 2017-10-13:
This was two years ago. I'd be interested to see what's possible now.

[image]

laurensvalk, 2017-10-29:
I saw @gregcowell and @dwalton76 worked on implementing the updates in the [GyroBalancer](https://github.com/ev3dev/ev3dev-lang-python/blob/develop/ev3dev/control/GyroBalancer.py) module, nice!

I was wondering if you've tried the new code and whether you had any issues running it? 

In particular, did you see it do OK at first [and then suddenly crash after some time](https://github.com/laurensvalk/segway/issues/10)?

gregcowell, 2017-10-29:
Yep, I mostly only ran the code for a minute or two while testing but the robot did suddenly fall over on occasion. I’ll do some more testing and get some better statistics.

BertLindeman, 2017-10-29:
Had just rebuilt balanc3r. Elas the support for the hi-technic gyro is not built in in GyroBalancer, as it is in the EV3-project. 
I have no EV3-gyro available. 
On ev3dev (using stretch) the gyro is not automatically recognized, so I set it by hand:
```
echo ht-nxt-gyro > /sys/class/lego-port/port1/set_device
```

laurensvalk, 2017-10-29:
>  Yep, I mostly only ran the code for a minute or two while testing but the robot did suddenly fall over on occasion. I’ll do some more testing and get some better statistics.

I guess it's a good thing we're seeing the same problem :) Would be great to have your input [here](https://github.com/laurensvalk/segway/issues/10) if you run into this problem again.

laurensvalk, 2017-10-29:
@BertLindeman does the default deg/s output value for this sensor give you decimal numbers or does it round to the nearest integer? If it rounds, I can give you some tips in order to use the full resolution with this code.

BertLindeman, 2017-10-29:
@laurensvalk What I see is integer value and the file decimals is zero:
```
Process 1 /sys/class/lego-sensor/sensor2
 sensor 1 driver_name: ht-nxt-gyro on port ev3-ports:in2
address:              ev3-ports:in2
mode:                 GYRO
modes:                GYRO
commands:             not implemented
decimals:             0
num_values:           1
value0:               31
units:                d/s
uevent:
LEGO_DRIVER_NAME=ht-nxt-gyro
LEGO_ADDRESS=ev3-ports:in2
```
But I do not know (yet) how to adapt the python code to accept the ht-gyro sensor. :-(

laurensvalk, 2017-10-29:
@BertLindeman open an issue over [here](https://github.com/laurensvalk/segway/) and I can give you some pointers to add it later this week; it's a pretty small change. In the mean time, see if you can configure it as an nxt-analog sensor (this may even be the default) and read a value (in between 0 and 4096) which varies as you rotate the robot. 

No pressure, [but let's aim for this](https://www.youtube.com/watch?v=lqF8FfUcZA4) :smile:

BertLindeman, 2017-10-29:
Thank you  @laurensvalk
Will open an issue (tomorrow) 
Confirmed: nxt-analog *is* the default driver if mode is `auto` and then the sensor reports:
```
Process 1 /sys/class/lego-sensor/sensor3
 sensor 1 driver_name: nxt-analog on port ev3-ports:in2
address:              ev3-ports:in2
mode:                 ANALOG-0
modes:                ANALOG-0 ANALOG-1
commands:             not implemented
decimals:             3
num_values:           1
value0:               2965
units:                V
```

BertLindeman, 2017-10-30:
Segway [issue](https://github.com/laurensvalk/segway/issues/11) opened.

laurensvalk, 2017-11-05:
@ddemidov: I updated the code to make it easier to work with other sensors. If you can get the IMU sensor to poll reasonably quickly, there's hardly anything you'll need to change in the code.

The HiTechnic Sensor is now supported thanks to @BertLindeman. 

Two parts in the code are affected: the sensor setup, and a constant in the parameter file. Given the HiTechnic example, I think you'll find it easy to adapt it to your sensor.

closed 2018-01-16 by laurensvalk