ev3dev/ev3dev #985: OSX Bluetooth connection not working

tprupsys, 2017-10-14:
### What are you trying to do?
ssh to the ev3dev device. It's telling connection refused. What I've done already:
- I've set (in my mac) bluetooh PAN to have IP of 192.168.2.1 and subnet mask to 255.255.255.0. 
- I've set (in my ev3dev) ip 192.168.2.22, subnet mask 255.255.255.0, gateway 192.168.2.1
- The robot and the computer is paired via bluetooth
- In robot, it say's it's network status is connected
- I can even ping the device from my mac:
```
tpr:lego-robot tpr$ ping 192.168.2.22
PING 192.168.2.22 (192.168.2.22): 56 data bytes
64 bytes from 192.168.2.22: icmp_seq=0 ttl=64 time=33.573 ms
64 bytes from 192.168.2.22: icmp_seq=1 ttl=64 time=10.788 ms
```
- But there is no port open on it (to ssh, 22 should be open) and the sshing says connection refused
```
tpr:lego-robot tpr$ sudo nmap -sS 192.168.2.22
All 1000 scanned ports on 192.168.2.22 are closed
```

```
tpr:lego-robot tpr$ ssh robot@192.168.2.22
ssh: connect to host 192.168.2.22 port 22: Connection refused
```

What am I doing wrong?

[image]

### What did you expect to happen?
To ssh successfully

### What actually happened?
```
tpr:lego-robot tpr$ ssh robot@ev3dev.local
ssh: connect to host ev3dev.local port 22: Connection refused
```

### What hardware and software are you using (including version numbers)?
macOS Sierra 10.12.6
ev3dev 4.4.32-17 kernel

dlech, 2017-10-15:
Don't manually configure IP addresses. Be sure to enable Internet Connection Sharing on the Mac for Bluetooth PAN. Then connect the network connection on the EV3. It will get an automatic IP address and be connected to the Internet.

tprupsys, 2017-10-15:
@dlech it doesn't help either - same situation, I can ping the robot but I cannot ssh and there's no port open on it:

1) I make bluetooth pan to have automatic dhcp: 
[image]
2) In robot, I set 'automatic osx settings' to be set, so it shows ipv4 settings:
```
Method: DHCP
Address: 192.168.2.2
Mask: 255.255.255.0
Gateway: 192.168.2.1
```

3) The internet sharing is enabled:
[image]

4) Bluetooth PAN icon is yellow (it's not connected I assume):
[image]

5) Again, I can ping:
```
tpr:lego-robot tpr$ ping ev3dev.local
PING ev3dev.local (192.168.2.2): 56 data bytes
64 bytes from 192.168.2.2: icmp_seq=0 ttl=64 time=10.667 ms
64 bytes from 192.168.2.2: icmp_seq=1 ttl=64 time=48.514 ms
64 bytes from 192.168.2.2: icmp_seq=2 ttl=64 time=9.680 ms
```

6) again, I cannot ssh:
```
tpr:lego-robot tpr$ ssh robot@ev3dev.local
ssh: connect to host ev3dev.local port 22: Connection refused
```

7) And again, I can see there's no port open on robot's ip:
```
tpr:lego-robot tpr$ sudo nmap -sS 192.168.2.2
All 1000 scanned ports on 192.168.2.2 are closed
```

dlech, 2017-10-15:
Sometimes if you have a bad SD card, it can cause SSH to not start. Do you have another SD card to try? Or try re-flashing your SD card.

tprupsys, 2017-10-15:
@dlech thanks, I've re-flashed the SD card and it works! :-)

closed 2017-10-15 by tprupsys

dlech, 2017-10-16:
Great!