ev3dev/ev3dev #986: Read from sensor commands attribute returns ENOTSUP

kortschak, 2017-10-14:
The recent changes in ev3go/ev3dev to eagerly get a variety of constant values causes sensor intialisation to fail when there are no commands. This is confirmed when you `cat commands` in a sensor without commands (in this case mindsensors 9dof):

```
robot@ev3dev:/sys/class/lego-sensor/sensor0$ cat commands 
cat: commands: Operation not supported
```

This is surprising. I would expect an empty list from commands, but a successful read.

The fix in ev3go is to either ignore errors from the caching of the `commands` attribute or to conditionally pass when the underlying cause is `ENOTSUP`, which is a little brittle:

```
index 45a8e6b..8ddb0e5 100644
--- a/sensor.go
+++ b/sensor.go
@@ -9,6 +9,7 @@ import (
        "os"
        "path/filepath"
        "strconv"
+       "syscall"
        "time"
 )
 
@@ -60,7 +61,9 @@ func (s *Sensor) setID(id int) error {
        }
        t.commands, err = stringSliceFrom(attributeOf(&t, commands))
        if err != nil {
-               goto fail
+               if pe, ok := cause(err).(*os.PathError); !ok || pe.Err != syscall.ENOTSUP {
+                       goto fail
+               }
        }
        t.modes, err = stringSliceFrom(attributeOf(&t, modes))
        if err != nil {
```
See https://github.com/ev3go/ev3dev/issues/69

Also see https://github.com/ev3go/ev3dev/issues/53#issuecomment-333283141.

dlech, 2017-10-15:
This is as designed. But, I would take a PR that returns an empty list instead.

kortschak, 2017-10-15:
What is the reason? It's at odds with the behaviour of motors.

dlech, 2017-10-15:
It's because I really have no idea what I am doing an I just make this stuff up as I go along. :wink:

kortschak, 2017-10-15:
You too!

kortschak, 2017-10-15:
jessie is closed I take it, so I should send against stretch?

dlech, 2017-10-15:
Yes. (You should find that stretch is the default branch on all of the GitHub repos)

closed 2017-10-19 by dlech

dlech, 2017-11-01:
Fix released in `ev3dev-1.6.0` kernel.