ev3dev/ev3dev #987: serial console seems broken with snapshot-ev3dev-stretch-ev3-generic-2017-09-30.img

perkinbr, 2017-10-16:
### What are you trying to do?
Access the serial console in stretch to see boot messages.  I tried activating it with both the ev3dev-config setting and manually editing /etc/flash-kernel/bootscript/bootscr.ev3 to set the console to ttyS1

### What did you expect to happen?
I expected to see kernel messages in the serial console.  

### What actually happened?
I checked with a logic analyzer and I did not see any transitions.  When I flash the latest release version (ev3dev-jessie-ev3-generic-2017-09-14) and used  ev3dev-config, I _do_ see  kernel messages on the serial console.
Note that this test was done on a completely clean image in both cases.

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=001653462FE6
BOARD_INFO_TYPE=main

dlech, 2017-10-16:
`/etc/flash-kernel/bootscript/bootscr.ev3` should not exist and I doubt that creating it will have any effect.

In ev3dev-stretch, the input port drivers are different, so I suspect that is what causing the problem. I suppose we need a way to set the port to `other-uart` mode during boot.

perkinbr, 2017-10-16:
Thanks for the quick response.  I'll try to poke around the kernel source to see if I can learn anything.

Re: /etc/flash-kernel/bootscript/bootscr.ev3
This file _does_ exists in the image referenced above and seem to affect the output of /boot/flash/boot.scr in a way that I expected.  Just to clarify, I only hacked bootscr.ev3 when ev3dev-config didn't work in order to see if it didn't like having two console settings.

You _are_ correct that it didn't make any difference.

From older threads I see this same file used to be in /usr/share instead of /etc

dlech, 2017-10-16:
Thanks for the clarification. I guess flash-kernel changed the location of the script files between jessie and stretch.

What really needs to be done is for input port 1 to be disabled in device tree. The "right" thing to do would be to tell u-boot to disable this node, but u-boot resides in the flash memory of the EV3 and is part of the official LEGO firmware. It's a really old version (2009) that doesn't even know about device tree.

So, I think our best option will be to compile a new device tree file and use flash-kernel to write it to the FAT partition of the SD card. 

Here is the device tree for EV3: https://github.com/ev3dev/ev3-kernel/blob/ev3dev-stretch/arch/arm/boot/dts/da850-lego-ev3.dts#L192

Basically, we just need to add a `status = "disabled";` property to the `in1` node (and maybe fix the driver if it does not already handle this property). Then figure out some way to automate the process.

perkinbr, 2017-10-16:
Sounds like a plan. I'll try to fool around with it if I can find some time; shouldn't be hard to test since I already got kernel compiles working.

perkinbr, 2017-10-17:
Confirmed that setting status = "disabled" in the in1 section causes kernel messages to be emitted from the in1 serial port.

dlech, 2018-06-14:
This is fixed in kernel `4.14.49-ev3dev-2.0.2-ev3`.

The driver will now automatically detect when the serial port is being used for console and disable the lego-port driver for that port.

closed 2018-06-14 by dlech