ev3dev/ev3dev #988: Question about LED and frequency

JorgePe, 2017-10-22:
I'm resurrecting old CodePilot and MicroScout with VLL control (light pulses)

I got VLL working with an USB FTDI adapter and a LED. It works with python in my Ubuntu laptop, my Raspberry Pi and also EV3 ev3dev. I'm using almost a straight copy of NCQ code available on the web, using python time.sleep for timings - base unit is 20 ms. A similar code, in Arduino, also works fine.

But when I try with WeDo 1.0 (USB) and Power Functions LED it doesn't work. First guess was not enough light because of USB 5.0V instead of PF 9.0V, but a custom brighter LED also fails.

So I try with EV3 using NXT-to-RCX adapter and setting OUT A as dcmotor (can't find a way to use the led in python if I set the port as led). It also doesn't work.

It seems a timing problem but not related to my use of time.sleep because it works with the FTDI adapter. So I'm suspecting that the output ports cannot couple with the timings needed and the problem might be on the transitions from ON to OFF and OFF to ON.

So two questions:

1. Do you know what's the response time of the output ports or the maximum frequency we can expect if we want to send just ON/OFF commands?

2. Any suggestion of a better way to flash a LED with EV3? This can also be useful for making a strobo
 :)

This is a sample of my code:

```
def vll1():
    led.duty_cycle_sp = 100
    sleep(0.02)
    led.duty_cycle_sp = 0
    sleep(0.04)

def vll0():
    led.duty_cycle_sp = 100
    sleep(0.04)
    led.duty_cycle_sp = 0
    sleep(0.02)

def vllinit():
    led.duty_cycle_sp = 100
    sleep(0.4)    

def vllstart():
    led.duty_cycle_sp = 0
    sleep(0.02)

def vllstop():
    led.duty_cycle_sp = 100
    sleep(0.02)
    led.duty_cycle_sp = 0
    sleep(0.06)
    led.duty_cycle_sp = 100
    sleep(0.12)
```

WasabiFan, 2017-10-22:
My gut reaction is that Python's sleep methods are going to be quite unreliable when you need this level of accuracy. Other intermediate processes on the brick are going to be vying for the tight pool of CPU cycles and your code won't be consistent. At a minimum, either 1) writing in C++ or another lower-level language and/or 2) setting your process to very high priority might help. A more extreme approach would be to implement it as a kernel module. I am not an expert though.

JorgePe, 2017-10-22:
Thanks.

I know that sleep method is not precise, already tested it against other method (don't remember which) and my conclusion was that for less than 5 or 6 ms sleep is useless. But for 20 ms is OK and VLL is also  tolerant enough - that's why same timings work with my USB FTDI LED adapter

Meanwhile I changed to internal LEDs instead of OUT port, like this:

```
def vll1():
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.YELLOW)
    sleep(0.02)
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.BLACK)
    sleep(0.04)
```

and it works!

So time.sleep is precise enough for what I want and the EV3 leds react fast enough for what I need. Maybe the Output port diodes/capacitors are slowing down my led transitions from ON to OFF and OFF to ON

The only problem with using the EV3 leds is that I cannot attach a LEGO brick to them in order to use LEGO fiberglass or even transclear round bricks as guidelines for the light.

dlech, 2017-10-22:
You can put the output ports in `led` mode and you should get a LED device in `/sys/class/leds/`. Also, when using motor control on the output port, you have to send a `run-direct` command before setting `duty_cycle_sp` will have any effect.

dlech, 2017-10-22:
> Maybe the Output port diodes/capacitors are slowing down my led transitions from ON to OFF and OFF to ON

Not a chance. Normally, they use ~13KHz PWM so 50Hz should be no problem. :wink:

JorgePe, 2017-10-22:
I get

```
leds ev3-ports:outA::brick-status: Bound to device 'ev3-ports:outA:rcx-led'
lego-port port4: Added new device 'ev3-ports:outA:rcx-led
```
but whatever I put in python generates an error.

That's why I opted for dcmotor (and yes, using run-direct)

13 KHz look nice for a strobe. So output ports is not the problem... I guess I need to re-check everything.

dlech, 2017-10-22:
You should be able to use `ev3-ports:outA::brick-status` as the LED name in python.

dlech, 2017-10-22:
You could also try turning the motor on and off instead of using the duty cycle (e.g. set `duty_cycle_sp` to 100 one time and use `run-direct` and `stop` commands to turn it on and off).

JorgePe, 2017-10-23:
Thanks for the suggestions.

For the LED usage I'll open an issue.

For the control with run_direct/stop instead of duty_cycle_sp that's interesting, changed for that.  But same behavior.

I decided to use the 3 methods in same script:
- FTDI bitbanging and a blue hi-bright LED
- EV3 LED
- EV3 outA and a Power Functions Light (white LED)

This is the relevant part for generating '1' and '0':

```
def vll1():
    bb.port = 1
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.YELLOW)
    led.run_direct()
    sleep(0.02)
    bb.port = 0
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.BLACK)
    led.stop()
    sleep(0.04)

def vll0():
    bb.port = 1
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.YELLOW)
    led.run_direct()
    sleep(0.04)
    bb.port = 0
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.BLACK)
    led.stop()
    sleep(0.02)
```

I turn all devices ON or OFF in the same sequence so the timing is the same for all LEDs.
The FTDI and the EV3 LED work.
The OutA   doesn't work.

I'm starting to think that PF Light has some strange wavelenght - it has much more bright than EV3 LED.

JorgePe, 2017-10-23:
I'm going to give up the NXT-RCX adapter method and stay with the USB FTDI adapter.
But to my surprise, the EV3 LED has enough brightness to control a MicroScout through a 1.20m (4 feet) TOSLink cable:
https://youtu.be/n1hfLL_yW-4

JorgePe, 2017-10-23:
Well, well, well...!

If i set port4 (outA) as dc-motor, set duty_cycle_sp at 100 and use run_direct/stop (or change duty_cycle_sp between 100 and 0 instead) nothing happens.

But if I set port4 as led, and use brightness values 0 or 100 it works !?!?

```
def vll1():
    bb.port = 1
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.RED)
    #led.run_direct()
    led.brightness = 100
    sleep(0.02)
    bb.port = 0
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.BLACK)
    #led.stop()
    led.brightness = 0
    sleep(0.04)

def vll0():
    bb.port = 1
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.RED)
    #led.run_direct()
    led.brightness = 100
    sleep(0.04)
    bb.port = 0
    ev3.Leds.set_color(ev3.Leds.LEFT, ev3.Leds.BLACK)
    #led.stop()
    led.brightness = 0
    sleep(0.02)
```

I thought 'dc-motor' and 'led' modes were basically the same but (at least for the timings I'm using) they are not.

I have VLL working as I wanted, I'm happy. But got curious... why the different behavior?

dlech, 2017-10-23:
You should check `max_brightness` of the LEDs. It could be that 100 brightness on the LEDs is not 100% duty cycle.

But really, the best way to see the difference would be to use an oscilloscope.

JorgePe, 2017-10-23:
cat /sys/class/leds/ev3-ports\:outA\:\:brick-status/max_brightness 
100

visually the bright level seems the same (great science! :D )
but the patterns seem different (great science again!)

I keep promising myself a logic analyzer for this xmas...

closed 2020-08-17 by dlech