ev3dev/ev3dev #989: bone and rpi2 development snapshots 2017-09-30 shutdown shortly after brickman starts

kortschak, 2017-10-22:
### What are you trying to do?
Boot stretch development snapshot of 2017-09-30 on BrickPi+, BrickPi3 and FatCatLab beagle bone black.

### What did you expect to happen?
Successful boot.

### What actually happened?
Successful boot into brickmon (via HDMI for PrickPi+/BrickPi3, via built-in screen on FatCatLab), then a couple of seconds later a clean power down.

### What hardware and software are you using (including version numbers)?

Information on for hardware and image - other infomormation not available due the issue.

BrickPi+:
```
Image file:         snapshot-ev3dev-stretch-rpi2-generic-2017-09-30
Board:              Raspberry Pi 2 Model B Rev 1.1
Revision:           a21041
```

BrickPi3
```
Image file:         snapshot-ev3dev-stretch-rpi2-generic-2017-09-30
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
```

FatCat:
```
Image file:         snapshot-ev3dev-stretch-bone-generic-2017-09-30
Board:              TI AM335x BeagleBone Black
Revision:           00c0
```

In all cases appropriate boot configuration text was modified:

BrickPi+: uncomment `dtoverlay=brickpi`, `init_uart_clock=32000000` and `dtparam=brickpi_battery=okay`
BrickPi3: uncomment `dtoverlay=brickpi3`
FatCat: uncomment `cape=evb`

dlech, 2017-10-22:
How are you powering these devices?

dlech, 2017-10-22:
Also see: https://github.com/ev3dev/ev3dev/issues/861#issuecomment-329334778

kortschak, 2017-10-22:
That makes sense, though leads to some surprising behaviour. I'm powering the device by USB through the main board just for testing kernel behaviour (with the full understanding that the capes won't drive anything). Is there some way to distinguish 5V through the main board from a low power through the cape?

dlech, 2017-10-22:
The "right way" to fix it would be to write power supply drivers for the USB ports that power the device.

This is complicated by the fact that all existing ev3dev programs assume that there is only one "System" power supply.

kortschak, 2017-10-22:
OK. So in the interim, where is `brickd` started so I can remove that service on the card at least temporarily (I'm spelunking at the moment).

dlech, 2017-10-22:
I don't have time to look at the moment, but I think you can mask systemd services by creating a symlink from `brickd.service` to `/dev/null` in `/etc/systemd/system/`.

kortschak, 2017-10-23:
Thanks, that works.

Does this close or become a feature request for richer power handling?

dlech, 2017-10-23:
> Does this close or become a feature request for richer power handling?

You tell me :wink:

If someone wants the bad enough to do the work, the sure, I don't see a reason why we could not make this happen.

kortschak, 2017-10-23:
In the short term I've just uninstalled `brickd`, so I'm fine to leave the situation as it is, but the plan for `brickd` looks like it's going to be an integral part of ev3dev. If that happens, then it may be difficult to have an ev3dev install without `brickd` and this would be needed. So either I change the name here for a long term issue, or we close it and make a new issue later. I'm easy for either of these.

dlech, 2018-04-15:
Fixed in `brickd_1.1.2`.

zombiezen, 2018-04-21:
This is not fixed for me. Booting up the BrickPi3 with just USB power to the main Raspberry Pi board will shut off.  I can confirm I am running brickd 1.1.2.

zombiezen, 2018-04-21:
```
robot@ev3dev:~$ for f in /sys/bus/iio/devices/iio\:device0/in_voltage_*; do echo -n "$f: "; cat $f; done
/sys/bus/iio/devices/iio:device0/in_voltage_3V3_input: 3355
/sys/bus/iio/devices/iio:device0/in_voltage_5V_input: 5028
/sys/bus/iio/devices/iio:device0/in_voltage_9V_input: 9095
/sys/bus/iio/devices/iio:device0/in_voltage_VCC_input: 4347
```

dlech, 2018-04-21:
This is with RPi 3B+ correct?

zombiezen, 2018-04-21:
Yes.

dlech, 2018-04-21:
Should be fixed now in `brickd_1.1.3`. I increased the minimum voltage from 4.2V to 4.5V.

zombiezen, 2018-04-22:
I can confirm this fixes the issue on my RPi 3B+. Thanks!

dlech, 2018-05-31:
Bumped the voltage up to 4.75V in `brickd_1.1.4`.

closed 2018-05-31 by dlech