ev3dev/ev3dev #990: How to use non-system LED in python?

JorgePe, 2017-10-23:
I'm connection a Power Functions Light to EV3 output port A through a NXT-RCX cable adapter.

I configure port4 (OUTA) to LED mode:

`echo led > /sys/class/lego-port/port4/mode`

dmesg:

```
[ 2983.128420] leds ev3-ports:outA::brick-status: Bound to device 'ev3-ports:outA:rcx-led'
[ 2983.149530] lego-port port4: Added new device 'ev3-ports:outA:rcx-led'
```

ls /sys/class/leds/ : 
```
ev3-ports:outA::brick-status  led0:red:brick-status    led1:red:brick-status
led0:green:brick-status       led1:green:brick-status
```

In python I don't know how to use this LED:

```
import ev3dev.ev3 as ev3
led = ev3.Led('ev3-ports:outA::brick-status')
```

`led.connected` returns 'False' and cannot set brightness

```
uname -a
Linux ev3dev 4.9.49-ev3dev-1.5.1-ev3 #1 PREEMPT Thu Sep 14 10:40:42 CDT 2017 armv5tejl GNU/Linux
cat /etc/debian_version 
9.1
```

dlech, 2017-10-23:
cc: @ev3dev/python

WasabiFan, 2017-10-23:
I can't recall exactly, but you might need to specify the name as a kwarg like here: https://github.com/ev3dev/ev3dev-lang-python/blob/jessie/ev3dev/ev3.py#L50

Also, are you on Jessie or Stretch?

JorgePe, 2017-10-23:
Thanks, it works:

```
import ev3dev.ev3 as ev3
led =  ev3.Led(name_pattern='ev3-ports:outA::brick-status')
led.brightness = 100
```

stretch:

```
cat /etc/os-release
PRETTY_NAME="ev3dev-stretch"
NAME="ev3dev-stretch"
ID=ev3dev
ID_LIKE=debian
HOME_URL="http://www.ev3dev.org"
SUPPORT_URL="http://www.ev3dev.org/support"
BUG_REPORT_URL="https://github.com/ev3dev/ev3dev/issues"
```

closed 2017-10-23 by JorgePe

WasabiFan, 2017-10-23:
Great!

You should be aware that we released a version of the Python library that properly supported Stretch only a couple days ago, so if you aren't running that you should upgrade to avoid odd issues down the road.

JorgePe, 2017-10-23:
Thanks.
I updated yesterday and I think I checked for updates today, not sure. But it is working and it allowed me to use VLL with 100% LEGO parts (not that I care much for purism but others might).