ev3dev/ev3dev #994: Motor will not turn with RaspberryPi/BrickPi3 but is shaking

CaptnMo, 2017-10-24:
I am trying to use BrickPi3 and after installing everything I am trying to test a Motor. I can only get a connection with m = Motor(), then m.run_timed(time_sp = 1000, speed_sp = -750) Motor will only make a noise but not turn. 

I do not understand on how to connect to the Motor. m = MediumMotor('spi0.1:MA') will not work. 
Using the Dexter OS the motor behaved correctly but I would like to re-use much of the code I was writing for my EV3 and I like the ev3 python.

### What hardware and software are you using (including version numbers)?
[    2.780500] brickpi3 spi0.1: Address: 1
[    2.780510] brickpi3 spi0.1: Mfg: Dexter Industries
[    2.782768] brickpi3 spi0.1: Board: BrickPi3
[    2.785318] brickpi3 spi0.1: HW: 3.2.1
[    2.787728] brickpi3 spi0.1: FW: 1.4.4
[    2.790091] brickpi3 spi0.1: ID: 75D43F06514D32384E202020FF112B3B
[    2.792336] lego-port port0: Registered 'spi0.1:S1' on 'spi0.1'.
[    2.792504] lego-port port1: Registered 'spi0.1:S2' on 'spi0.1'.
[    2.792655] lego-port port2: Registered 'spi0.1:S3' on 'spi0.1'.
[    2.792803] lego-port port3: Registered 'spi0.1:S4' on 'spi0.1'.
[    2.792960] lego-port port4: Registered 'spi0.1:MA' on 'spi0.1'.
[    2.793084] lego-port port4: Added new device 'spi0.1:MA:lego-nxt-motor'
[    2.793240] lego-port port5: Registered 'spi0.1:MB' on 'spi0.1'.
[    2.793364] lego-port port5: Added new device 'spi0.1:MB:lego-nxt-motor'
[    2.793522] lego-port port6: Registered 'spi0.1:MC' on 'spi0.1'.
[    2.793640] lego-port port6: Added new device 'spi0.1:MC:lego-nxt-motor'
[    2.793799] lego-port port7: Registered 'spi0.1:MD' on 'spi0.1'.
[    2.793925] lego-port port7: Added new device 'spi0.1:MD:lego-nxt-motor'
[    2.822294] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 83) (baudrate 100000)
[    3.084676] tacho-motor motor0: Registered 'lego-nxt-motor' on 'spi0.1:MA'.
[    3.087505] tacho-motor motor1: Registered 'lego-nxt-motor' on 'spi0.1:MB'.
[    3.090412] tacho-motor motor2: Registered 'lego-nxt-motor' on 'spi0.1:MC'.
[    3.093001] tacho-motor motor3: Registered 'lego-nxt-motor' on 'spi0.1:MD'.

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-rpi2-generic-2017-09-14
Kernel version:     4.4.78-21-ev3dev-rpi2
Board:              Raspberry Pi 3 Model B Rev 1.2
Revision:           a02082
Brickman:           0.8.1
ev3devKit:          0.4.3
```

dlech, 2017-10-24:
BrickPi3 FW: 1.4.4 has a bug that causes motors to do this if the battery voltage is < 9V. This should be fixed in FW: 1.4.6, but I have not had a chance to verify it yet.

If that does not fix the problem, please try ev3dev-stretch. http://docs.ev3dev.org/en/ev3dev-stretch/

CaptnMo, 2017-10-25:
Hi,
updating the firmware fixed it. Thank you very much!
But could explain how to connect to the motor on other ports?
>>> m = MediumMotor('outA')
>>> m.connected
False
>>> m = MediumMotor('MA')
>>> m.connected
False
>>> m = MediumMotor('spi0.1:MA')
>>> m.connected
none of this will work. How would it be done correctly?

CaptnMo, 2017-10-25:
m = Motor('spi0.1:MA') works

closed 2017-10-25 by CaptnMo