ev3dev/ev3dev #996: Connecting EV3 brick with Bluetooth using ev3dev and Eclipse

edouardo17, 2017-10-25:
Hi all,

I've been trying for a few days to connect my EV3 brick revision 0006 to my computer running on Ubuntu using a Bluetooth connection. I think I'm not far from success, but I still have a couple of issues I can't solve...
I'm using the ev3dev Brick Manager on my EV3 brick, running on brickman v0.8.1, and on the computer, I'm on the last release of Ubuntu 16.04 LTS.
As for my IDE, I'm using Eclipse Oxygen (latest release as well), and coding trying to execute some Java code on my brick. I also have installed the leJOS plugin in Eclipse.

I have created a Network Access Point with my brick, and am able to pair in Bluetooth in Ubuntu, which is already a good start. When I ping the brick using the IP address displayed at the top of my EV3 LCD screen, the device responds correctly to the packets sent. I don't have any issue either when I connect to my brick in the command line using [code]$ ssh robot@ev3dev.local[/code].
In Eclipse, I have configured the leJOS plugin in Window->Preferences->leJOS as shown bellow:

[url=https://www.hostingpics.net/viewer.php?id=970641Screenshotfrom20171025131805.png][img]https://img4.hostingpics.net/pics/970641Screenshotfrom20171025131805.png[/img][/url]

I purposely ticked the "Use SSH" box because... well, just because it seemed to work better that way  :? 
Now, the issue I'm having comes when I want to run a program from Eclipse on the brick, I have the following message showing after hitting Run: 

[url=https://www.hostingpics.net/viewer.php?id=528132Screenshotfrom20171025132221.png][img]https://img4.hostingpics.net/pics/528132Screenshotfrom20171025132221.png[/img][/url]

When I choose to untick the SSH box, well... same kind of issue, the connection is refused...

[url=https://www.hostingpics.net/viewer.php?id=707209Screenshotfrom20171025132332.png][img]https://img4.hostingpics.net/pics/707209Screenshotfrom20171025132332.png[/img][/url]

Does someone has an idea of where the problem is coming from? Isn't the brick refusing the connection even though it is properly connected to the laptop in Bluetooth? 

Many thanks in advance,
Edouard

WasabiFan, 2017-10-25:
I am a bit doubtful that the LeJOS plugin will work with ev3dev. I've never used it, so I don't know what it does exactly, hut my guess is that it is attempting to use protocols and credentials that are specific to LeJOS.

dlech, 2017-10-25:
If you want to use the leJOS plugin for Eclipse, then you also need to run leJOS on the EV3, not ev3dev.

If you want an IDE for ev3dev, then I would suggest trying ev3dev-stretch and Visual Studio Code. http://docs.ev3dev.org/en/ev3dev-stretch/

closed 2020-08-17 by dlech