ev3dev/ev3dev #997: ev3 Sound.Speak does not work on newest stretch

CaptnMo, 2017-10-25:
### What are you trying to do?
I need my robot to talk. I am using Brickpi3 and Raspberry Pi 3 with newest stretch

trying to run:
ev3.Sound.speak('I am a beer drinking robot!').wait()

results in:

### What actually happened?
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
aplay: main:788: audio open error: No such file or directory

### What hardware and software are you using (including version numbers)?
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-rpi2-generic-2017-10-24
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000e184528d
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=75D43F06514D32384E202020FF112B3B
BOARD_INFO_TYPE=aux

dlech, 2017-10-25:
cc: @ev3dev/python

dlech, 2017-10-25:
Did you enable the RaspberrPi sound card in `config.txt`?

dlech, 2017-10-25:
```
# uncomment to enable sound (loads snd_bcm2835)
dtparam=audio=on
```

CaptnMo, 2017-10-25:
Sorry, that is correct. Now it works

closed 2017-10-25 by CaptnMo