ev3dev/ev3dev #998: Using Lego Energy Display with BrickPi3 and RaspberryPi 3

CaptnMo, 2017-10-25:
### What are you trying to do?
I would like to use the Lego Energy Display but it is not yet implemented for BrickPi3. Is there a feasible workaround? A quick solution?

### What hardware and software are you using (including version numbers)?
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-rpi2-generic-2017-10-24
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000e184528d
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=75D43F06514D32384E202020FF112B3B
BOARD_INFO_TYPE=aux

dlech, 2017-10-25:
BrickPi3 cannot automatically detect sensors.You must manually configure it. e.g. `echo "lego-power-storage 0x02" > set_device`

You should be able to use pretty much any supported I2C sensor with BrickPi3.

http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#lego-energy-display

CaptnMo, 2017-10-27:
Thanks again. It works. But that was not easy. Without your help I could not figure out where to put the address 0x02
Works as desired:

`import ev3dev.brickpi as ev3
p = ev3.LegoPort('spi0.1:S1')
p.mode = "nxt-i2c"
p.set_device = "lego-power-storage 0x02"
ed = ev3.Sensor('spi0.1:S1')
ed.value()
ed.mode = "IN-VOLT"
ed.value()
1417
ed.mode = "IN-AMP"
ed.value()
0`

closed 2017-10-27 by CaptnMo

dlech, 2017-10-27:
GitHub tip: use code fence to make stuff easier to read (I edited your previous comment)

https://help.github.com/articles/creating-and-highlighting-code-blocks/