ev3dev/ev3dev #999: BrickPi3 firmware upgrading in ev3dev OS

dgvogol, 2017-10-25:
### What are you trying to do?
Upgrade the BrickPi3 firmware directly in ev3dev OS

- Need the newer openocd debian package from testing, https://packages.debian.org/buster/openocd
- Need the openocd interface configuration files from https://github.com/DexterInd/openocd
- Need a simple scripts to do the firmware upgrade: https://github.com/DexterInd/BrickPi3/raw/master/Firmware/brickpi3samd_flash_firmware.sh

### What did you expect to happen?
Backport the openocd package to stretch
A new debian package including the openocd interface configuration + BrickPi3 firmware upgrading scirpt

### What actually happened?
Just manually verified it on my rpi3+brickpi3, it works.  

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-rpi2-generic-2017-09-30
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000d0d03afb
BOARD_INFO_TYPE=main
```
```
robot@ev3dev:~$ apt-cache show openocd
Package: openocd
Status: install ok installed
Priority: optional
Section: embedded
Installed-Size: 5738
Maintainer: Debian Electronics Packaging Team <pkg-electronics-devel@lists.alioth.debian.org>
Architecture: armhf
Source: openocd (0.10.0-1)
Version: 0.10.0-1+b1
Depends: libc6 (>= 2.15), libftdi1-2 (>= 1.2), libhidapi-hidraw0 (>= 0.8.0~rc1+git20140201.3a66d4e+dfsg), libjim0.77 (>= 0.73), libusb-0.1-4 (>= 2:0.1.12), libusb-1.0-0 (>= 2:1.0.16)
Description: Open on-chip JTAG debug solution for ARM and MIPS systems
Description-md5: 3846dda8ec5ba34e07da3b1d9ee73c0a
Homepage: http://openocd.sourceforge.net/

Package: openocd
Source: openocd (0.9.0-1)
Version: 0.9.0-1+b2
Installed-Size: 4833
Maintainer: Uwe Hermann <uwe@debian.org>
Architecture: armhf
Depends: libc6 (>= 2.15), libhidapi-hidraw0 (>= 0.8.0~rc1+git20140201.3a66d4e+dfsg), libjim0.76 (>= 0.73), libusb-0.1-4 (>= 2:0.1.12), libusb-1.0-0 (>= 2:1.0.9)
Description: Open on-chip JTAG debug solution for ARM and MIPS systems
Description-md5: 3846dda8ec5ba34e07da3b1d9ee73c0a
Homepage: http://openocd.sourceforge.net/
Tag: devel::debugger, hardware::embedded, implemented-in::c, role::program
Section: embedded
Priority: extra
Filename: pool/main/o/openocd/openocd_0.9.0-1+b2_armhf.deb
Size: 2206446
MD5sum: 5a234d99b00ab0ba2fbf41207d1fc5e6
SHA256: 5ac4a1eef0d0268cbae455d15e5ee4b3ad95738a230ea6b44be69f6ebbc91cf3
```

dlech, 2017-10-25:
Are you saying that you successfully upgraded the BrickPi3 firmware using ev3dev already? Can you list the exact steps that you used?

dgvogol, 2017-10-25:
Here are my steps:
- Add debian testing into source list
```
robot@ev3dev:~$ sudo tee /etc/apt/sources.list.d/testing.list << EOF
deb http://ftp.us.debian.org/debian testing main contrib non-free
EOF

[sudo] password for robot:
deb http://ftp.us.debian.org/debian testing main contrib non-free
robot@ev3dev:~$
```
- apt update
```
robot@ev3dev:~$ sudo apt update
Ign:1 http://ftp.us.debian.org/debian stretch InRelease
Get:2 http://ftp.us.debian.org/debian stretch-backports InRelease [91.8 kB]
Get:3 http://ftp.us.debian.org/debian testing InRelease [136 kB]
...
Fetched 39.3 MB in 59s (660 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
704 packages can be upgraded. Run 'apt list --upgradable' to see them.
```
- openocd package information
```
robot@ev3dev:~$ apt-cache show openocd
Package: openocd
Source: openocd (0.10.0-1)
Version: 0.10.0-1+b1
Installed-Size: 5738
Maintainer: Debian Electronics Packaging Team <pkg-electronics-devel@lists.alioth.debian.org>
Architecture: armhf
Depends: libc6 (>= 2.15), libftdi1-2 (>= 1.2), libhidapi-hidraw0 (>= 0.8.0~rc1+git20140201.3a66d4e+dfsg), libjim0.77 (>= 0.73), libusb-0.1-4 (>= 2:0.1.12), libusb-1.0-0 (>= 2:1.0.16)
Description: Open on-chip JTAG debug solution for ARM and MIPS systems
Description-md5: 3846dda8ec5ba34e07da3b1d9ee73c0a
Homepage: http://openocd.sourceforge.net/
Tag: devel::debugger, hardware::embedded, implemented-in::c, role::program
Section: embedded
Priority: optional
Filename: pool/main/o/openocd/openocd_0.10.0-1+b1_armhf.deb
Size: 2345220
MD5sum: b1f0bb258397c3b8f9892570d63f302c
SHA256: 9f3aa0d7a583f4f2d6fd89e7e4260ec5418db688db557b48b363b3cbaf3d6606

Package: openocd
Source: openocd (0.9.0-1)
Version: 0.9.0-1+b2
Installed-Size: 4833
Maintainer: Uwe Hermann <uwe@debian.org>
Architecture: armhf
Depends: libc6 (>= 2.15), libhidapi-hidraw0 (>= 0.8.0~rc1+git20140201.3a66d4e+dfsg), libjim0.76 (>= 0.73), libusb-0.1-4 (>= 2:0.1.12), libusb-1.0-0 (>= 2:1.0.9)
Description: Open on-chip JTAG debug solution for ARM and MIPS systems
Description-md5: 3846dda8ec5ba34e07da3b1d9ee73c0a
Homepage: http://openocd.sourceforge.net/
Tag: devel::debugger, hardware::embedded, implemented-in::c, role::program
Section: embedded
Priority: extra
Filename: pool/main/o/openocd/openocd_0.9.0-1+b2_armhf.deb
Size: 2206446
MD5sum: 5a234d99b00ab0ba2fbf41207d1fc5e6
SHA256: 5ac4a1eef0d0268cbae455d15e5ee4b3ad95738a230ea6b44be69f6ebbc91cf3
```
- Install openocd from testing
```
robot@ev3dev:~$ sudo apt install openocd/testing
Reading package lists... Done
Building dependency tree
Reading state information... Done
Selected version '0.10.0-1+b1' (Debian:testing [armhf]) for 'openocd'
[... 93 lines cut ...]
--  ------------------ ---------- ------ ------------------ ------------
 0* at91samd21j18.cpu  cortex_m   little at91samd21j18.cpu  running
Info : SWD DPIDR 0x0bc11477
target halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x00006af8 psp: 0x20002a80
Info : SWD DPIDR 0x0bc11477
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00006aec psp: 0x20002a80
** Programming Started **
auto erase enabled
Info : SAMD MCU: SAMD21J18A (256KB Flash, 32KB RAM)
wrote 49152 bytes from file brickpi3_firmware.bin in 21.690392s (2.213 KiB/s)
** Programming Finished **
** Verify Started **
verified 43204 bytes in 0.208638s (202.223 KiB/s)
** Verified OK **
Info : SWD DPIDR 0x0bc11477
shutdown command invoked
robot@ev3dev:~$
```
- Remove testing from source list
```
robot@ev3dev:~$ sudo bash -c "rm /etc/apt/sources.list.d/testing.list;apt update"
...
```

dlech, 2017-10-26:
Thanks very much for the detailed info. This should be easier than I thought.

dlech, 2017-10-26:
openocd is packaged/backported: https://github.com/ev3dev/openocd

dlech, 2017-10-26:
firmware is packaged: https://github.com/ev3dev/firmware-brickpi3

dlech, 2017-10-26:
There is a kernel bug that prevents this from working correctly, so I will update again when that fix is released.

dlech, 2017-11-01:
Everything should be in place for this to work starting with the 2017-11-01 ev3dev-stretch snapshot. Please test. https://oss.jfrog.org/artifactory/list/oss-snapshot-local/org/ev3dev/brickstrap/

closed 2017-11-01 by dlech