ev3dev/ev3dev #1000: Python/Brickpi: Set mode on a sensor works in shell but not in code

CaptnMo, 2017-10-27:
### What are you trying to do?
I am adding this code to hello.py from the visual studio code hello world example. Behaviour in my own code is the same  

s2 = ev3.LegoPort('spi0.1:S2')
    s2.mode = "ev3-uart"
    s2.set_device = "lego-ev3-ir"
    ir = ev3.InfraredSensor('spi0.1:S2')
    **ir.mode = 'IR-SEEK'**

### What did you expect to happen?
If I put this code in the shell it runs nicely but if I run it from my python file I get an error. It seems to be the same problem whenever I try to set the mode to something. In the shell it works but not when run from a .py file

### What actually happened?
when running from visual studio code I get:
  File "/home/robot/vscode-hello-python-master/hello.py", line 94, in <module>
    main()
  File "/home/robot/vscode-hello-python-master/hello.py", line 67, in main
    ir.mode = 'IR-SEEK'
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 1716, in mode
    self._mode = self.set_attr_string(self._mode, 'mode', value)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 229, in set_attr_string
    return self._set_attribute(attribute, name, value)
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 211, in _set_attribute
    attribute.write(value.encode())
io.UnsupportedOperation: File not open for writing

### What hardware and software are you using (including version numbers)?
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-rpi2-generic-2017-10-24
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000e184528d
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=75D43F06514D32384E202020FF112B3B
BOARD_INFO_TYPE=aux

CaptnMo, 2017-10-27:
If I add a time.sleep(2) it seems to work, so the difference between shell and code is the time it takes to type the line???? 

s2 = ev3.LegoPort('spi0.1:S2')
s2.mode = "ev3-uart"
s2.set_device = "lego-ev3-ir"
ir = ev3.InfraredSensor('spi0.1:S2')
**time.sleep(2)**
ir.mode = 'IR-SEEK'

dlech, 2017-10-27:
It would make more sense to put the `sleep` before the `InfraredSensor` constructor. It takes some time for the kernel to register the sensor device.

dlech, 2017-10-27:
Surprise! You opened the 1000th ev3dev issue! You win a pair of official ev3dev.org LEGO tiles! (Please send me you mailing address to claim your prize - my email is on my GitHub profile).

CaptnMo, 2017-10-27:
Well, my code works so far so good using the time.sleep statement as a workaround. Thanks again.

CaptnMo, 2017-10-27:
And many thanks for the tiles. I am really looking forward to put them on my robot.

closed 2017-10-27 by CaptnMo