ev3dev/ev3dev #1002: Include gir1.2-ev3devkit in the ev3dev-stretch image

cavenel, 2017-10-29:
### What are you trying to do?
Import and use ev3devKit in Python with:
```
from gi.repository import Ev3devKit
```
(see https://github.com/ev3dev/ev3dev/issues/857)

### What did you expect to happen?
gir1.2-ev3devkit-0.5 to be installed by default in the ev3dev kernel.

### What actually happened?
gir1.2-ev3devkit-0.5 is not installed by default in the last ev3dev kernel:
```
Traceback (most recent call last):
  File "/home/robot/ev3-print3rbot/print3rBot.py", line 22, in <module>
    from gi.repository import Ev3devKit
  File "/usr/lib/python3/dist-packages/gi/importer.py", line 127, in find_module
    'introspection typelib not found' % namespace)
ImportError: cannot import name Ev3devKit, introspection typelib not found
```

Plus installing it from the repositories fails due to missing dependencies:
```
The following packages have unmet dependencies:
 gir1.2-ev3devkit-0.5 : Depends: gir1.2-gudev-1.0 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-10-24
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165343581F
BOARD_INFO_TYPE=main
```

cavenel, 2017-10-29:
Installation of `gir1.2-ev3devkit-0.5` works after downloading the armel package from https://packages.debian.org/stretch/armel/gir1.2-gudev-1.0/download and installing it with `dpkg`:
```
robot@ev3dev:~$ sudo dpkg -i gir1.2-gudev-1.0_230-3_armel.deb 
Selecting previously unselected package gir1.2-gudev-1.0:armel.
(Reading database ... 33276 files and directories currently installed.)
Preparing to unpack gir1.2-gudev-1.0_230-3_armel.deb ...
Unpacking gir1.2-gudev-1.0:armel (230-3) ...
Setting up gir1.2-gudev-1.0:armel (230-3) ...

robot@ev3dev:~$ sudo apt-get install gir1.2-ev3devkit-0.5
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  gir1.2-grx-3.0
The following NEW packages will be installed:
  gir1.2-ev3devkit-0.5 gir1.2-grx-3.0
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 57.8 kB of archives.
After this operation, 310 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://archive.ev3dev.org/debian stretch/main armel gir1.2-grx-3.0 armel 3.0.0 [40.4 kB]
Get:2 http://archive.ev3dev.org/debian stretch/main armel gir1.2-ev3devkit-0.5 armel 0.5.2 [17.4 kB]
Fetched 57.8 kB in 1s (33.7 kB/s)          
Selecting previously unselected package gir1.2-grx-3.0:armel.
(Reading database ... 33280 files and directories currently installed.)
Preparing to unpack .../gir1.2-grx-3.0_3.0.0_armel.deb ...
Unpacking gir1.2-grx-3.0:armel (3.0.0) ...
Selecting previously unselected package gir1.2-ev3devkit-0.5.
Preparing to unpack .../gir1.2-ev3devkit-0.5_0.5.2_armel.deb ...
Unpacking gir1.2-ev3devkit-0.5 (0.5.2) ...
Setting up gir1.2-grx-3.0:armel (3.0.0) ...
Setting up gir1.2-ev3devkit-0.5 (0.5.2) ...
```

Still it would be nice to have it in stretch directly.

dlech, 2017-10-29:
The next stretch snapshot will include this package.

I'm guessing that you had trouble with `gir1.2-gudev-1.0` because you forgot to run `sudo apt update` first. It installs normally for me.

cavenel, 2017-10-29:
Thank you!

Actually I had to update the python ev3dev library to solve a screen problem just before, so I did
```
sudo apt-get update && sudo apt-get upgrade && sudo apt-get dist-upgrade && sudo apt-get autoremove
```
a couple of minutes before trying to install `gir1.2-gudev-1.0`. It managed to update everything (including the python ev3dev library), but still it didn't work for `gir1.2-gudev-1.0`.

dlech, 2017-11-01:
Fix shipped in the 2017-11-01 ev3dev-stretch snapshots

closed 2017-11-01 by dlech