ev3dev/ev3dev #1003: Waiting for completion with wait_while will not return to the programm (using BrickPi3)

CaptnMo, 2017-10-29:
### What are you trying to do?
'''def main():
    
    debug_print(tower.connected)
    tower.run_timed(time_sp=2000, speed_sp=800,stop_action='brake')
    tower.wait_while('running')    
    debug_print('Ready')'''

I try to use wait_while to wait for the end of the execution of the motors' movement. The motor will execute its' movement but the programm will not continue. 

### What hardware and software are you using (including version numbers)?
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-rpi2-generic-2017-10-24
Kernel version:     4.9.50-ev3dev-1.5.1-rpi2
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=A02082
BOARD_INFO_MODEL=Raspberry Pi 3 Model B Rev 1.2
BOARD_INFO_SERIAL_NUM=00000000e184528d
BOARD_INFO_TYPE=main
Board:              board1
BOARD_INFO_FW_VER=1.4.6
BOARD_INFO_HW_REV=3.2.1
BOARD_INFO_MODEL=Dexter Industries BrickPi3
BOARD_INFO_SERIAL_NUM=75D43F06514D32384E202020FF112B3B
BOARD_INFO_TYPE=aux

dlech, 2017-10-29:
cc: @ev3dev/python 

This sounds like a combination of a kernel driver problem and a python library problem.

In the kernel, only EV3 (and EVB) have notifications implemented on the motor `state` attribute.

You don't say which version of the python library you are using, but if it is relying on `poll`, then it won't work as-is on BrickPi (or PiStorms or NXTMMX).

dwalton76, 2018-03-29:
@dlech is this a HW limitation of brickpi/brickpi3 or just a matter of support hasn't been added to the kernel?

dlech, 2018-03-29:
It's a kernel thing.

I also just noticed in the original post that the `stop_action` is `brake`, which only works on EV3/EVB, not any other device. This, on the other hand, is a firmware limitation.

dwalton76, 2018-03-29:
hmm I triple checked, `brake` vs `hold` are working for me.  brickpi3 actually sets the motor state to `holding` (I don't think I've seen it used before)....just something different I noticed.

`It's a kernel thing` ack, I normally avoid the kernel but if this is something fairly painless I can take a stab at it if you point me in the right direction. 

I'll add some code to ev3dev-lang-python so that `Motor.wait()` will work on platforms that do not support poll yet.

dwalton76, 2018-03-29:
I also see the motor state in `overloaded` a lot, that is another one I have never seen used before.

dwalton76, 2018-03-29:
I'll give it a shot....are there some docs on what the workflow is for doing dev on lego-linux-driver?  I see a Makefile but it isn't setup for just `make && make install`.

Also in brickpi3_ports_out.c it says
```The ``state`` attribute will never return the ``stalled`` flag.```

Is that a HW limitation?  The state often returns `overloading` but does so both when the motor is moving and when it is stalled so I can't key off of that to figure out if it is still moving.

dwalton76, 2018-03-29:
I'm guessing these are the docs I need
https://github.com/ev3dev/ev3dev-buildscripts

I remember doing this once a while back :)

dlech, 2018-03-29:
You could look at the requested speed and the actual speed to figure out if the motor is stalled (in the driver itself).

dlech, 2018-07-15:
Should be fixed in kernel release `ev3dev-2.2.0`.

closed 2018-07-15 by dlech

dwalton76, 2018-10-31:
Just noticed this was fixed...this is awesome thank you @dlech !!