ev3dev/ev3dev #1006: latest jessie release for ev3 bricks not booting

ardumont, 2017-11-12:
Hello,

### What are you trying to do?

Following documentation [1], flashing the latest ev3 image [2] for jessie in my sdcard using linux tools [3].

[1] http://www.ev3dev.org/docs/getting-started/
[2] https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2017-09-14/ev3dev-jessie-ev3-generic-2017-09-14.zip
[3] http://www.ev3dev.org/docs/tutorials/writing-sd-card-image-linux-command-line/

### What did you expect to happen?

ev3dev to boot as explained in the documentation.

### What actually happened?

#### tl; dr
ev3dev hangs at boot after mounting the partition(s).

#### details

After downloads (tried multiple times, with firefox or wget), i have a different sha1sum of the zip file.
As there is no digest file proposed for download alongside the file to check against corruption, i don't know where the problem lies.

Objectively, it could be me having trouble downloading the file.
It could also be that the file upstream is corrupted (which would be surprising as i don't see issues about it here). Thus the tryout with the prior release which works (cf. Note point below).

I noticed that when decompressing the archive, i have CRC checksums errors spit (unzip, 7z, etc...).

Example with unzip:
```sh
unzip ev3dev-jessie-ev3-generic-2017-09-14.zip
Archive:  ev3dev-jessie-ev3-generic-2017-09-14.zip
   creating: ev3dev-jessie-ev3-generic-2017-09-14/
  inflating: ev3dev-jessie-ev3-generic-2017-09-14/ev3dev-jessie-ev3-generic-2017-09-14.img   bad CRC 63ad332d  (should be e77c81bc)
   creating: ev3dev-jessie-ev3-generic-2017-09-14/.meta/
  inflating: ev3dev-jessie-ev3-generic-2017-09-14/.meta/manifest.json
  inflating: ev3dev-jessie-ev3-generic-2017-09-14/.meta/image.bmap
  inflating: ev3dev-jessie-ev3-generic-2017-09-14/.meta/logo.svg
```
Note: you need to scroll horizontally to actually see the warning about CRC

Example with 7z:
```sh
7z x ev3dev-jessie-ev3-generic-2017-09-14.zip

7-Zip [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=en_US.UTF-8,Utf16=on,HugeFiles=on,64 bits,4 CPUs Intel(R) Core(TM) i7-5600U CPU @ 2.60GHz (306D4),ASM,AES-NI)

Scanning the drive for archives:
1 file, 222703853 bytes (213 MiB)

Extracting archive: ev3dev-jessie-ev3-generic-2017-09-14.zip
--
Path = ev3dev-jessie-ev3-generic-2017-09-14.zip
Type = zip
Physical Size = 222703853

ERROR: CRC Failed : ev3dev-jessie-ev3-generic-2017-09-14/ev3dev-jessie-ev3-generic-2017-09-14.img

Sub items Errors: 1

Archives with Errors: 1

Sub items Errors: 1
```

Well, I did not notice those bad CRC warnings at first so i continued.

I flashed the micro sd card and had no error raised.

``` sh
$ cat ev3dev-jessie-ev3-generic-2017-09-14.img | sudo dd of=/dev/sdb bs=4M status=progress
[sudo] password for tony:
1887436694 bytes (1.9 GB, 1.8 GiB) copied, 3.11328 s, 606 MB/s
4+14181 records in
4+14181 records out
1887436694 bytes (1.9 GB, 1.8 GiB) copied, 134.427 s, 14.0 MB/s
$ sync
```

Placed the microsd in the ev3 and tried to boot.
Micro sd detected and ev3 tried to boot (led orange flashed multiple times fast).

Once boot starting, i had the splashscreen showing ev3dev and error messages about ext* partitions.
(journal errors, could not mount partition as ext2 due to feature incompatibilities, same with ext3, same with ext4)...

Photo is quite unreadable... but adding it here nonetheless.
[image]

So as it was failing to mount the partition as ext* and finally doing nothing (expectedly ;)
I stopped the ev3, removed the sdcard and tried to fix those errors.
Well, trying to make gparted fix them anyway.
Indeed, gparted noticed errors on the /dev/sdb2 partition (EV3DEV_ROOTFS).

```txt
<i>dumpe2fs 1.43.4 (31-Jan-2017)
Journal superblock magic number invalid!</i>

<i>Unable to read the contents of this file system!
Because of this some operations may be unavailable.
The cause might be a missing software package.
The following list of software packages is required for ext4 file system support:  e2fsprogs v1.41+.</i>
```
Making gparted fixed it (i guess, it answers yes to all fsck questions so i don't think it's quite a good fix).

Booting again, the boot process is able to mount the partition but hangs forever.
Photo: 
[image]

So probably not a good fix :)
As i'm not really sure of the way forward now, i'd rather raise concerns here.

#### For information

Trying the prior release (one marked as depreciated) [4], and following the same instructions mentioned, I did not have any issues. ev3dev starts as entertained by the documentation.

[4] https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2017-06-09/ev3dev-jessie-ev3-generic-2017-06-09.zip

### What hardware and software are you using (including version numbers)?

- computer machine: debian 9 stable (stretch).
- hardware: ev3 (which boots normally with the standard lego firmware)
brick hw: v0.60; brick fw: v1.09h (once booted in the official firmware, settings > brick info).

- 2 micro sdhc: 1 8gib, 1 32gib (tested both with the same result).

I don't know where to do that.
If it's after the ev3dev boot (which i think it is), well, i cannot yet :)

Well, thanks in advance for

1. any help on this :)
2. this which seems quite nice, the documentation is quite awesome already.

Cheers,

dlech, 2017-11-12:
I just downloaded <https://github.com/ev3dev/ev3dev/releases/download/ev3dev-jessie-2017-09-14/ev3dev-jessie-ev3-generic-2017-09-14.zip> and got the md5sum fc4bad5f09fdfdcb38befda3e523402b.

It sounds like you have a bad Internet connection or bad hardware that is causing the file corruption. You could try downloading an ev3dev-jessie snapshot from <https://oss.jfrog.org/artifactory/list/oss-snapshot-local/org/ev3dev/brickstrap/>. It is a different server, so you may have better luck.

dlech, 2017-11-12:
You can find checksums for the snapshot images here: <https://oss.jfrog.org/webapp/#/artifacts/browse/tree/General/oss-snapshot-local/org/ev3dev/brickstrap>.

ardumont, 2017-11-14:
Hello,

Confirming the new downloads work by the way.

I'm closing this.

Thanks for your time

closed 2017-11-14 by ardumont