ev3dev/ev3dev #1007: I2C NXTCam-v5 Python program

ec-m, 2017-11-15:
I would like to communicate with my NXTCam-v5 via I2C to receive the coordinates of multiple objects. I am using ev3dev with Python and I am wondering whether someone has implemented I2C communication between the EV3 and the NXTCam-v5 (or any similar sensor) in Python and could provide some code? That would be great - thank you.

Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2017-11-15:
This is NOT ev3dev-lang-python, but here is something I was playing around with...

```python
#!/usr/bin/env micropython
"""Sample program using EXPLOR3R"""

from uev3dev import *

# init display
display = Display()
fbuf, fbuf_data = display.framebuffer()

# init motors
left_motor = LargeMotor('B')
right_motor = LargeMotor('C')

# fun with I2C

in4 = EV3InputPort('4')
in4.mode = 'other-i2c'
bus4 = SMBus('/dev/i2c-in4')
bus4.set_address(0x01)

firmware = bytes(bus4.read_i2c_block_data(0x00, 8)).decode().strip()
vendor = bytes(bus4.read_i2c_block_data(0x08, 8)).decode().strip()
product = bytes(bus4.read_i2c_block_data(0x10, 8)).decode().strip()
debug_print(firmware, vendor, product)

if vendor != 'mndsnsrs' or product != 'NXTcam5':
    raise ValueError('Wrong sensor')

# functions

def scale(x1, y1, x2, y2):
    """Convert points from NXTCam to screen rectangle"""
    x1 = x1 * display.width // 240
    y1 = y1 * display.height // 160
    x2 = x2 * display.width // 240
    y2 = y2 * display.height // 160
    return x1, y1, x2 - x1, y2 - y1

# Main loop

bus4.write_byte_data(0x41, 0x46)  # set face tracking mode

while True:
    n = bus4.read_byte_data(0x42)
    fbuf.fill(1)
    for i in range(0, n):
        data = bus4.read_i2c_block_data(0x43 + i * 5, 5)
        color = data[0]
        x1 = data[1]
        y1 = data[2]
        x2 = data[3]
        y2 = data[4]
        x, y, w, h = scale(x1, y1, x2, y2)
        fbuf.rect(x, y, w, h, 0)
    display.update(fbuf_data)

    speed = n and ((x1 + x2 // 2) - 128) // 4
    left_motor.run_unregulated(-speed)
    right_motor.run_unregulated(speed)
```

You should be able to copy the I2C commands along with `from smbus import SMBus` and they should work in python3 on ev3dev. Don't expect the motor or display stuff to work though.

I had an algorithm for reading multiple objects, but I don't think I saved it. Basically, in the while loop I would read one byte from 0x42, which tells the number of objects then `for i in range(0, num):` read 5 bytes from 0x43 + num * 5 to read each object. Note: SMBus can only read up to 32 bytes at once, so you can't read all 8 objects in one command.

I2C register definitions are in the User's Manual if you haven't found that already: http://www.mindsensors.com/pdfs/NXTCam5-User-Guide.pdf

ec-m, 2017-11-16:
Thank you for the quick response and the code snippet. I first tried

```python
from smbus import SMBus
bus2 = SMBus('/dev/i2c-in2')
```
but this gave me the error that an integer is required as a parameter for the constructor. Therefore I changed the code to 

```python
bus2 = SMBus(1)
```
I hope this was correct?! However, now when I try reading as follows

```python
firmware = bytes(bus2.read_i2c_block_data(0x00, 8)).decode().strip()
```
or writing using `0x02` as the address as it is documented in the [NXTCam-v5 user guide](http://www.mindsensors.com/pdfs/NXTCam5-User-Guide.pdf) as follows
```python
bus2.write_byte_data(0x02, 0x41, 0x42)  # object tracking enabled
```
the program fails with `OSError: [Errno 121] Remote I/O error`. What am I doing wrong?

I am sorry to be posting questions like this since they are not directly related to ev3dev but I don't know where else to ask (the mindsensors forum for example is quite inactive).

ec-m, 2018-04-06:
Hi @dlech,

thank you for the reply - it helped me a lot and since then I've been playing around with the camera a lot. This is the ev3dev-lang-python code I used to implement the communication (just a single read):

    bus = SMBus(4) # NXTCam-v5 plugged into input port 2, /dev/i2c-in2 -> i2c-4
    address = 0x01
    bus.write_byte_data(address, 0x41, 0x42)  # set object tracking mode
    num = bus.read_byte_data(address, 0x42)  # get number of tracked objects
        if num == 0:
            raise RuntimeError("Zero objects detected.")
        for i in range(0, num):
            data = bus.read_i2c_block_data(address, 0x43 + i * 5, 5)
            # print information on detected objects (rectangles)
            print("color: ", data[0])
            print("x coordinate upper left corner: ", data[1])
            print("y coordinate upper left corner: ", data[2])
            print("x coordinate lower right corner: ", data[3])
            print("x coordinate lower right corner: ", data[4])

One thing I still couldn't figure out though is how to automatically set the mode of the input port to `other-i2c`. I always have to do that manually on the EV3 brick. I already tried a solution similar to the one you have implemented above:

    port = LegoPort(INPUT_2)
    port.mode = 'other-i2c'

However, if I execute this before opening the SMBus for the I2C communication, I always receive the error `PermissionError: [Errno 13] Permission denied` when the line `bus = SMBus(4)` is executed. (The  mode of the input port is correctly changed from `auto` to `other-i2c` though.)

The same thing happens when I try executing a script with `os.system("sh init_port1.sh")` where init_port1.sh contains `echo other-i2c > /sys/class/lego-port/port1/mode`. Notice here that the input port marked as number 2 on the EV3 brick internally is port 1.

Do you (or anybody else) have an idea what the problem might be and how I could achieve what I am trying to do? Thank you for your help and any kind of further assistance!

dlech, 2018-04-08:
On ev3dev-jessie, the I2C adapter device is not created until you set the mode, so it takes a short time for the udev rules to detect the event and set the permissions of the I2C adapter device.

ec-m, 2018-04-09:
Perfect - it worked. I love how some solutions are so simple :) Thank you @dlech

DanieleBenedettelli, 2020-02-12:
> ```python
> from uev3dev import *
> ```

What kind of module is uev3dev? Is there anything like SMBus on pybricks-micropython?

dlech, 2020-02-12:
uev3dev comes from [here](https://github.com/dlech/micropython-ev3dev/tree/ev3dev-stetch/src/uev3dev)

It includes an SMBus compatible [i2c module](https://github.com/dlech/micropython-ev3dev/blob/ev3dev-stetch/src/uev3dev/i2c.py).

DanieleBenedettelli, 2020-02-13:
How do I import uev3dev module in VS code when using the EV3 micropython shebang? The program stops and complains about the module missing

dlech, 2020-02-13:
The `uev3dev` folder needs to be copied to `/home/robot/.micropython/lib` on the EV3.

Or if you just need the I2C stuff, you can copy the `i2c.py` file to your project (in the same folder as your main program) and `import .i2c`.

DanieleBenedettelli, 2020-02-13:
Thanks!
I can now use raw I2C with pybricks-micropython. For example:
`#!/usr/bin/env pybricks-micropython

from pybricks import ev3brick as brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import (Port, Stop, Direction, Button, Color,
                                 SoundFile, ImageFile, Align)
from pybricks.tools import print, wait, StopWatch
from pybricks.robotics import DriveBase
from i2c import SMBus

# Write your program here
brick.sound.beep()
firmware = bytes(bus4.read_i2c_block_data(REG_VERSION, 8)).decode().strip()
vendor = bytes(bus4.read_i2c_block_data(REG_VENDOR_ID, 8)).decode().strip()
product = bytes(bus4.read_i2c_block_data(REG_PRODUCT_ID, 8)).decode().strip()
print(firmware, vendor, product)
`

Audrielcz, 2020-02-19:
Hello, I have a uestion regarding EV3 i2c communication. I am trying to connect Mindsensors remote control to the new EV3 micropython build from LEGO and I am stuck with some issues. First thing which I wanted to do was to use micropython module machine.i2c but apperantely that module does not have i2c class in EV3 build. 

So I went over the internet and found this article and started to dig through your i2c library. Now I am stuck with (which I hope is the last thing) the setup connection.

in the part of your code:
in4 = EV3InputPort('4')
in4.mode = 'other-i2c'

You are using "EV3InputPort" function which probably makes some "in4" class to help out the bus. But in the new micropython, there is nothing similar, or atleast I didnt find it.

For now I am stuck with this code:
bus4 = i2c.SMBus('/dev/i2c-in4') #
bus4.set_address(0x02)
firmware = bytes(bus4.read_i2c_block_data(0x00, 8)).decode().strip()
vendor = bytes(bus4.read_i2c_block_data(0x08, 8)).decode().strip()
product = bytes(bus4.read_i2c_block_data(0x10, 8)).decode().strip()
debug_print(firmware, vendor, product)

which connects to the i2c port, but it hits me with this error message:
Traceback (most recent call last):
  File "/home/robot/test1/main.py", line 22, in <module>
  File "/home/robot/test1/i2c.py", line 200, in read_i2c_block_data
  File "/home/robot/test1/i2c.py", line 133, in _access
  File "fcntl/fcntl.py", line 36, in ioctl
  File "os/os/__init__.py", line 68, in check_error
OSError: 6

Any idea how to fix this please? 
Cheers

Audrielcz, 2020-02-19:
@DanieleBenedettelli for what I read, you were able to start a i2c connection on micropython. Could you share your secret with me please? :)

DanieleBenedettelli, 2020-02-19:
I posted the whole code.

closed 2020-08-17 by dlech

mzajkaniha, 2020-10-07:
i have problem with this code:
`#!/usr/bin/env pybricks-micropython
from pybricks.hubs import EV3Brick
from pybricks.ev3devices import (Motor, TouchSensor, ColorSensor,
                                 InfraredSensor, UltrasonicSensor, GyroSensor)
from pybricks.parameters import Port, Stop, Direction, Button, Color
from pybricks.tools import wait, StopWatch, DataLog
from pybricks.robotics import DriveBase
from pybricks.media.ev3dev import SoundFile, ImageFile

import i2c
ev3 = EV3Brick()

bus4 = i2c.SMBus('/dev/i2c-in2') 
bus4.set_address(0x01)
bus4.write_byte_data(0x41, 0x42)  # set obj tracking mode`

error:
`Traceback (most recent call last):
  File "/home/robot/i2c/main.py", line 16, in <module>
  File "/home/robot/i2c/i2c.py", line 124, in set_address
  File "fcntl/fcntl.py", line 30, in ioctl
  File "os/os/__init__.py", line 68, in check_error
OSError: [Errno 16] EBUSY`

dlech, 2020-10-07:
`EBUSY` happens because the Linux kernel driver is still using the sensor. You can fix this by putting the input port in to `other-i2c` mode.

arthurcoelho442, 2023-09-06:
Hello, I'm trying to take a picture using an NTXcam with micropython, in mindstorn ev3 dev, does anyone know how I can do this?

Audrielcz, 2023-09-06:
> Hello, I'm trying to take a picture using an NTXcam with micropython, in mindstorn ev3 dev, does anyone know how I can do this?

Try using my lib: https://github.com/Audrielcz/Mindsensors_Pybricks

BeWSM, 2023-09-06:
hey @Audrielcz , can I use your code for nxtcam5 to a nxtcam4??

Audrielcz, 2023-09-06:
Hi, I am not sure, did test it only on v5. But it should be similar. My lib need some cleanup but should work.Let me know if you need any helpMatěj Štětka6. 9.