ev3dev/ev3dev #1008: How to configure EOPD sensor in analog mode

LeeJenkins, 2017-11-16:
### What are you trying to do?
I want to use the EOPD sensor from a C++ program. Here's my code:

`
    std::cout << "make instance of eopd" << std::endl;
    ev3dev::sensor eopd4( ev3dev::INPUT_4, { ev3dev::sensor::nxt_analog });
    std::cout << "eopd driver_name: " << eopd4.driver_name() << std::endl;
    std::cout << "eopd modes:" << std::endl;
    ev3dev::mode_set eopdModes = eopd4.modes();
    for( ev3dev::mode_set::iterator it=eopdModes.begin(); it != eopdModes.end(); it++ )
        std::cout << " " << *it;
    std::cout << std::endl;
    eopd4.set_mode("ANALOG-0");
    std::cout << eopd4.value() << std::endl;
`

If I run this code on a freshly booted ev3, the output looks like this:

> make instance of eopd
> Caught system_error with code generic:38 meaning no device connected: Function not implemented

Next I use Brickman and go to Device Browser > Ports > in4 > Set mode, select nxt-analog.

After Setting the mode in Brickman, I can re-run my program and the program works correctly:

> make instance of eopd
> eopd driver_name: nxt-analog
> eopd modes:
>  ANALOG-0 ANALOG-1
> 4970

### What did you expect to happen?
I expect to be able to initialize the EOPD sensor from my C++ program after booting, without resorting to clicking through Brickman configuration screens.

### What hardware and software are you using (including version numbers)?
robot@ev3dev:~$ ev3dev-sysinfo
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
robot@ev3dev:~$

WasabiFan, 2017-11-16:
When instantiating the sensor with the call to `sensor::sensor(address_type address, const std::set<sensor_type> &types)`, the second argument is a set of _constraints_; i.e., you are asking it to find a device connected to port in4 which is using the driver `nxt-analog`. This usually works with most devices because the correct driver is auto-detected, but in this case you aren't using a sensor that will be auto-detected so it doesn't yet have a mode. You should use the `lego_port` class to set the mode of the `in4` port first, then your existing code will detect the sensor properly. See also http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html#other-sensors-and-advanced-usage.

WasabiFan, 2017-11-16:
For future reference, sensors and other devices have a `connected` attribute that tells you whether the instance actually found the device it as looking for (matching the given constraints). You can check that with an assertion after instantiating the class to confirm the device connected properly.

ddemidov, 2017-11-16:
https://github.com/ev3dev/ev3dev-lang-python/pull/402

WasabiFan, 2017-11-16:
@ddemidov This instance is in C++; isn't the `connected` attribute going to be kept in that library?

ddemidov, 2017-11-16:
Ah, sorry then. I obviously mixed this up with python.

LeeJenkins, 2017-11-16:
@WasabiFan I've looked through the page you linked, and also here: http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/ports.html, but I'm still not getting it. I prefixed my code with the following two lines:

    ev3dev::lego_port in4( ev3dev::INPUT_4 );
    in4.set_mode("nxt-analog");

After rebooting and running my program I still see the generic:38 error. What am I missing?

LeeJenkins, 2017-11-16:
One other thing: I cannot check if the sensor is `connected` because the error happens in the constructor of `ev3dev::sensor`.

WasabiFan, 2017-11-16:
Oh, whoops... I think the method you want is `set_set_device`, not the `set_mode`. (and yes, that does _really_ appear to be the name of the method. That naming choice would be a question for @ddemidov).

https://github.com/ddemidov/ev3dev-lang-cpp/blob/c0c49c93387432a68d09ba0ba96e489423703441/ev3dev.h#L1796-L1805:

```c++
  // Set Device: write-only
  // For modes that support it, writing the name of a driver will cause a new
  // device to be registered for that driver and attached to this port. For
  // example, since NXT/Analog sensors cannot be auto-detected, you must use
  // this attribute to load the correct driver. Returns -EOPNOTSUPP if setting a
  // device is not supported.
  auto set_set_device(std::string v) -> decltype(*this) {
    set_attr_string("set_device", v);
    return *this;
  }
```

ddemidov, 2017-11-16:
This is probably the only attribute in e3dev sysfs where name starts with `set_`. So the real question is probably: 'Why is it named that way?' :smile:

WasabiFan, 2017-11-16:
Side-note: @LeeJenkins You might need to add a small sleep/delay after setting the port mode to give the driver a chance to initialize the new sensor node. Sadly there isn't a cleaner way to do that as far as I'm aware.

ddemidov, 2017-11-16:
Another note: the master branch in `ev3dev-lang-cpp` has been updated for Debian stretch, but it seems @LeeJenkins still uses jessie. Input port names were among the things affected by the move from jessie to stretch.

There is jessie-compatible version at [jessie](https://github.com/ddemidov/ev3dev-lang-cpp/tree/jessie) branch.

LeeJenkins, 2017-11-17:
@WasabiFan I tried calling `set_set_device()`, but it threw an exception with the message `Caught system_error with code system:95 meaning Operation not supported`. 

Then I tried putting a `sleep` after the call to `set_mode`

    std::cout << "make instance of lego_port in4" << std::endl;
    ev3dev::lego_port in4( ev3dev::INPUT_4 );
    std::cout << "call in4.set_mode(\"nxt-analog\")" << std::endl;
    in4.set_mode("nxt-analog");
    usleep(100*1000);
    std::cout << "make instance of eopd" << std::endl;
    ev3dev::sensor eopd4( ev3dev::INPUT_4, { ev3dev::sensor::nxt_analog });
    std::cout << "eopd4.value() : " << eopd4.value() << std::endl;

Now it works like a boss.

I appreciate your help!

closed 2017-11-17 by LeeJenkins

WasabiFan, 2017-11-17:
Glad to hear that! I don't fully understand the `set_device`/`mode` situation ([docs](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/ev3.html#input-ports)), but I've never actually used the port interface nor have I ever been involved in discussions around it so that isn't particularly surprising.