ev3dev/ev3dev #1009: Go into single user mode to reset forgotten password?

stormingrobots, 2017-11-18:
### What are you trying to do?
lost password

### What did you expect to happen?
cannot login 

### What actually happened?
cannot login

### What hardware and software are you using (including version numbers)?
revision 006 4.4.87-22-ev3dev-ev3

dlech, 2017-11-19:
I don't know of a way to boot to single user mode.

You could try putting the SD card in another Linux computer and edit `/etc/passwd` and `/etc/shadow` manually to clear the password like [this](https://askubuntu.com/a/84697/376284).

Or, if you don't care about losing files, you can just re-flash the SD card.

markyi370, 2018-03-24:
If you happen to have a similar ARM device (like Beaglebone) you can put the SD card in it and chroot, then change the password from there.

This is the method I use when I'm too lazy to figure out what password it's supposed to be.

closed 2020-08-17 by dlech