ev3dev/ev3dev #1010: Latency reading EV3 sensors

massimiliano-mantione, 2017-11-30:
### What are you trying to do?

Reading sensor values in a tight, endless loop.

### What did you expect to happen?

Receiving the sensor values with minimal latency.

### What actually happened?

The sensor values are apparently received "later".

### What hardware and software are you using (including version numbers)?

```
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3
```

The test program has been compiled with `go version go1.8.3`.
The build command to cross compile is `env GOARM=5 GOARCH=arm GOOS=linux gb build .......`.

A description of the test environment is [here](https://github.com/massimiliano-mantione/go-bots/blob/master/test-latency/README.md).

I know the testing setup can appear complex, but I wanted to be scientific in my experiments.

The ultimate goal is to build a competitive (i.e. *fast*) line following robot, and to be fast it needs to react quickly to changes in sensor values.

At 2m/s 10ms mean 2cm, which can make the difference between missing or taking a 90 degrees turn. What I am observing is an average latency of more than 50ms.

dlech, 2017-11-30:
There is some discussion on this subject already at #324.

dlech, 2018-03-22:
Also see https://github.com/ev3dev/ev3dev-lang-python/issues/451 which suggests using the `bin_data` attribute.

QuirkyCort, 2018-03-23:
I suspect that in your test, it's not the color sensor that is reading the LED late. It may be that there's a short delay before the LED turns on. In some of my own tests, I found that there are significant system load for a short duration whenever I turn the LED on. Perhaps the kernel needs to do something to activate the LED? That would fit with your observation that decreasing the nice level of your program makes things worse (...less time for the kernel to do whatever it is doing to turn on the LED).

You may want to modify your test to use the output ports instead. That avoids the LED issue and would be more relevant to your ultimate goal of a fast line follower. Using my own custom Python bindings (https://github.com/QuirkyCort/ev3dev-lang-python-fast), 2 color sensors (RGB mode), and 2 motors (...in regulated mode), I was able to achieve <5ms average.

massimiliano-mantione, 2018-03-23:
@QuirkyCort I also have my own binding (for Go, not Python).
The main things they do differently from the standard ones is to keep the sensor value file open and to use `bin_data` instead of the textual value.

You are right in saying that leds can have a different latency than motors, and in principle I should have tested motors. Perhaps a reasonable setup could have been detecting something with an IR sensor, then activating a medium motor that would in turn push a touch sensor, and measure the latency there. But then the problem is that there is inherent physical latency in this setup (the motor movement), which is hard to predict and difficult to separate from the one I want to measure.

Anyway I think that my test highlights two issues:

* the unpredictability of the kernel scheduler
* the fact that actions seem "asynchronous" (the API gives a command but the effect will happen "later", when something else is scheduled, be it a kernel thread or I don't know what)

What I mean is that I would expect writes to device files to have effect "immediately".

In my latest competition I ended up using [ev3rt](http://ev3rt-git.github.io), and I could do a very complex "read, compute, actuate" loop about 18k times per second.

Even there the scheduler was not perfect but the longest loop iteration took 2ms, and it happened about once every three seconds. Literally every other iteration was about 1/18000 of a second. I suspect that those periodic delays were related to the bluetooth stack but at those levels I did not care, it was fast enough but most of all predictable enough.

Using ev3dev I could never get that kind of predictability: in my tests several loop iterations end up taking tens of milliseconds.

About ev3dev, I am seriously thinking to try to compile a kernel with the rt-preempt patch set.
I would also activate [FCSE](https://static.lwn.net/images/conf/rtlws11/papers/proc/p01.pdf).
Then I would run the rt-preempt latency tests and see the results I'd get.

@dlech I remember you wrote somewhere that the rt-preempt patch was problematic on armv5, or something similar... do you think rt-preempt + FCSE would be viable on the EV3?

dlech, 2018-03-23:
I haven't tried rt-preempt on the EV3, but on BeagleBone + FatcatLab's EVB, which uses the same motor and sensor drivers as the EV3, with an RT kernel, we would start missing interrupts on the motor's quadrature encoder, so the motor position and speed calculations would be off. And even if everything did work perfectly with an RT kernel, I think you would still be looking at delays on the order of 5ms.

The approach I would like to try (but haven't got around to yet) is to use one of the PRUs in the processor for hard realtime stuff and let the main processor run Linux and take care of the non-realtime stuff.

QuirkyCort, 2018-03-24:
> But then the problem is that there is inherent physical latency in this setup (the motor movement), which is hard to predict and difficult to separate from the one I want to measure.

What I meant was to use the motor output ports to control an LED. If you connect a motor to the EV3 through a breakout cable, you can easily tap the PWM output to power an LED (...with the appropriate resistors). The motor is just there to handle the auto detection and isn't actually used.

dlech, 2018-07-15:
> The approach I would like to try (but haven't got around to yet) is to use one of the PRUs in the processor for hard realtime stuff and let the main processor run Linux and take care of the non-realtime stuff.

FYI, we now have the tachometer implemented in the PRU, so if anyone wants to try to build a realtime kernel, go for it.

massimiliano-mantione, 2018-08-02:
> FYI, we now have the tachometer implemented in the PRU, so if anyone wants to try to build a realtime kernel, go for it.

This sounds great, I'll give it a try.

Could you please point me to the instructions on how to rebuild the ev3dev ev3 kernel?

I pulled the ev3dev/debian-jessie-cross docker image and cloned the ev3dev/ev3-kernel git repo, and I can start a shell in the container , expose the kernel source as a volume, and run the cross compiler (I run everything on a Linux host).

However I could not find a wiki page that explains how to configure and build the proper kernel.
All the instructions I could find are from the pre-docker era and are marked as obsolete.

What I expect is some build incantation that gives me an image file I can put in the boot partition of the sd card instead of the default ev3dev one :-)
Then I will start patching the kernel, rebuilding, and see where I get.

dlech, 2018-08-02:
https://github.com/ev3dev/ev3dev-buildscripts