ev3dev/ev3dev #1011: How to get the MicroInfinity CruizCore XG1300L gyroscope working

LeeJenkins, 2017-12-01:
_I'm copying this issue from [ev3dev-lang-cpp](https://github.com/ddemidov/ev3dev-lang-cpp/issues/34) because it seems to be an ev3dev issue and not specifically a c++ issue._

I'm trying to use an XG1300L gyroscope attached to my EV3, but I'm not able to get it working in my C++ program. I'm looking for help getting it to connect. I assume I've just missed a simple step, but I cannot figure out where I've gone wrong.

I'm following the example code in the [ev3dev-mapping](https://github.com/bmegli/ev3dev-mapping) code by [bmegli](https://github.com/bmegli). Specifically, I looked at how he used the gyro in the [dead-reconing module](https://github.com/bmegli/ev3dev-mapping-modules/blob/master/ev3dead-reconning/main.cpp). It doesn't look like he's doing anything beyond declaring the sensor.

    32    const char *GYRO_PORT="i2c-legoev35:i2c1";
    ...
    85    ev3dev::i2c_sensor gyro(GYRO_PORT, {"mi-xg1300l"});

Later the code checks `gyro->connected()` before sending a command. Easy, right? Well, maybe not. It looks like bmegli may have [modified the i2c_sensor class](https://github.com/ev3dev/ev3dev-lang/issues/167) to get it working.

I certainly haven't been able to get it to work. Here's my code:

    void gyroTest() {
        std::cout << "make instance of migyro" << std::endl;
        ev3dev::i2c_sensor mnfinity_gyro( "i2c-legoev35:i2c1", {"mi-xg1300l"} );
        if( !mnfinity_gyro.connected() ) {
            std::cout << "migyro.connected(): FALSE" << std::endl;
            return;
        }
        std::cout << "migyro.connected(): TRUE" << std::endl;
        return;
    }

When I run the program, it always prints `FALSE`. This doesn't look like rocket surgery, so I must be passing an incorrect argument value. But I cannot find any other example code and I don't know if I need to get bmegli's patch. Someone who has used this sensor will surely be able to point me in the right direction.

As a sanity check, I booted the EV3 with the LEGO firmware and then ran a sample program from the Microinfinity web site that uses the LEGO Mindstorms block-programming environment. The program was able to display readings from the sensor, so I know the hardware is good and attached properly.

Here's my ev3dev sysinfo:

    robot@ev3dev:~$ ev3dev-sysinfo -m
    
    **System info (from `ev3dev-sysinfo`)**
    ```
    Image file:         ev3dev-jessie-ev3-generic-2017-09-14
    Kernel version:     4.4.87-22-ev3dev-ev3
    Board:              LEGO MINDSTORMS EV3 Programmable Brick
    Revision:           0006
    Brickman:           0.8.1
    ev3devKit:          0.4.3
    ```
    
    robot@ev3dev:~$

There is a short discussion for this issue on [ev3dev-lang-cpp](https://github.com/ddemidov/ev3dev-lang-cpp/issues/34). I've tried:

- calling lego_port::set_mode("nxt-i2c) before instantiating the sensor, which did not change the outcome
- calling lego_port::set_set_device("mi-xg1300l"), which threw an exception "Operation not supported"
- adding a two-second sleep after the set_mode() and another before calling connected(); neither helped

LeeJenkins, 2017-12-01:
... tagging @bmegli , who appears to have done a lot of work with the XG1300L previously.

dlech, 2017-12-01:
It looks like `set_device` is only currently supported for NXT/Analog sensors, not I2C (this statement applies to EV3 only, not other platforms).

As a workaround, you can do the following before running your program:

    sudo sh -c 'echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-5/new_device'

LeeJenkins, 2017-12-01:
Ah! This is a big step forward. I ran the command to write the driver name and I2C address to new_device. (In my case I'm using port 1 so it's i2c-3.) After that, I can see the gyro in the Brickman Device Browser and I can watch the angle value changing on screen. Very cool.

However, I still cannot access the sensor in my program. I've changed the code a little bit:

    void gyroTest() {
        std::cout << "make instance of mnfinity_gyro" << std::endl;
        ev3dev::sensor mnfinity_gyro( ev3dev::INPUT_1, {"mi-xg1300l"} );
        if( !mnfinity_gyro.connected() ) {
            std::cout << "mnfinity_gyro.connected(): FALSE" << std::endl;
            return;
        }
        std::cout << "mnfinity_gyro.connected(): TRUE" << std::endl;
        return;
    }

I'm using `ev3dev::sensor` instead of `ev3dev::i2c_sensor` as in my previous post.

One other thing I tried: based on the [Appendix C docs](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/i2c.html), the XG1300L should be considered a "Other/I2C" sensor because its I2C address is 0x01. So I tried calling `lego_port::set_mode("other-i2c")` before creating the sensor object, but that didn't make any difference. Then I made a shell script to set the mode and then write the driver name to new_device. My program still didn't work.

In doing the above test, I noticed that when I query the port information in my program, it didn't match what I expect. I ran these shell commands (in a sudo'd bash shell):

    echo other-i2c > /sys/class/lego-port/port0/mode
    echo mi-xg1300l 0x01 > /sys/bus/i2c/devices/i2c-3/new_device

Then in my C++ program I created an instance `ev3dev::lego_port in1( ev3dev::INPUT_1 );` and dumped the port information (never mind trying to connect for the moment). Here's what it reported:

    in1.address(): in1
    in1.connected(): 1
    in1.driver_name(): legoev3-input-port
    in1.status(): no-sensor
    in1.mode(): other-i2c
    in1.device_index(): 0

I expected the `driver_name` to be "mi-xg1300l", not "legoev3-input-port". It looks like `lego_port::set_set_device()` is the call to set the driver name, right? But that's the one you said isn't supported for I2C sensors, yeah? Is there a technical reason that it isn't supported? Or is it just that no one has had time to do it? I do embedded linux FW development for a living, so maybe it's something I could look into?

dlech, 2017-12-01:
Ah, yes. I forgot about setting the `other-i2c` mode. The automatic sensor type detection won't stay in I2C mode if there is no data, so setting the mode to `other-i2c` (or `nxt-i2c` if you need voltage on pin 1 - but most I2C sensors don't need this) is the safe thing to do.

> I expected the driver_name to be "mi-xg1300l", not "legoev3-input-port".

Your object is a `lego-port` device, not a `lego-sensor` device, so it would be expected that the driver name is the port driver and not the sensor driver. The `driver_name` is basically telling you which kind of port this is.

> It looks like lego_port::set_set_device() is the call to set the driver name, right?

No, it just registers a `lego-sensor` device with the name given... which will be matched by udev to a driver with the same name.

> But that's the one you said isn't supported for I2C sensors, yeah?

Not on EV3 anyway. It has been implemented, for example, on BrickPi (because BrickPi doesn't have automatic detection of sensors).

> Is there a technical reason that it isn't supported?

Not that I am aware of.

> Or is it just that no one has had time to do it?

Or there just hasn't been a need since this is the only "supported" sensor that can't be automatically detected.

> I do embedded linux FW development for a living, so maybe it's something I could look into?

Awesome. You can find the relevant code here: https://github.com/ev3dev/lego-linux-drivers/blob/v1.6.0/ev3/ev3_ports_in.c#L1005

And the BrickPi3 implantation to use as a reference is here: https://github.com/ev3dev/lego-linux-drivers/blob/v1.6.0/brickpi3/brickpi3_ports_in.c#L341

A couple of other things... ev3dev jessie is "done" so I won't be taking patches for that (unless there is a *really* compelling reason), so changes should be made on the ev3dev-stretch branch. And you can find some info on building ev3dev kernels here: https://github.com/ev3dev/ev3dev-buildscripts.

dlech, 2017-12-01:
It would be even better if we could find a way to automatically detect the sensor though (nothing became of this in #210).

bmegli, 2017-12-03:
@LeeJenkins 

I am sorry for the delay. Note that I may not be up-to-date if a lot of things changed.

There are 3 things here

# Port Mode

CruizCore is not `other-i2c`, it's `nxt-i2c`, see the [docs](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html#supported-sensors)

Forcing `other-i2c` mode may be reason why some things are not working.

The port mode (`nx2-i2c`) should be correctly detected when plugging the sensor as far as I remember.

You don't have to do anything here.

# Driver

While the port mode is correctly detected, the device driver is not, so this is where you have to load it manually. You have already figured that I think.

Try: 
```bash
sudo sh -c 'echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-X/new_device'
```
Where X corresponds to your port, probably X=port_number+2

In ev3dev-mapping I loaded the driver from the [script](https://github.com/bmegli/ev3dev-mapping-modules/blob/master/scripts/ev3init.sh#L20) this way, before calling the program.

`echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-X/new_device` was what I put in the documentation in the past. 

Now it's `echo mi-xg1300l 0x01 > /sys/class/lego-port/port<N>/set_device` so something was improved/changed and if it's not working there may be some bug or problem in documentation.

# C++ bindings

Now you should be able to create C++ bindings i2c_sensor:

```C++
ev3dev::i2c_sensor gyro(GYRO_PORT, {"mi-xg1300l"});
```

Where GYRO_PORT should be something along the line `i2c-legoev35:i2c1`, you can check that under console for the lego-sensor.

In the past I have added this 2 parameter constructor in my fork but this was also added by @ddemidov [here](https://github.com/ddemidov/ev3dev-lang-cpp/commit/0f13224bc0619cb100c228bd6e3ea965307e52ed), and should work.

-------------------------------------------

I would first try it from the console:
1. plug the sensor
2. `echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-X/new_device`
3. check if it is working (cat values)
4. check the port for the device (cat, I mean the `i2c-legoev35:i2c1`)

On the C++ bindings side:

Also I think you only have to use the `i2c_sensor` class only if you need to take advantage of specific features like setting polling frequency and you could use generic `sensor` class otherwise. I am not sure of that, it has been some time since I used it.

bmegli, 2017-12-03:
Ok, I looked at C++ bindings code master:

From [C++ bindings master](https://github.com/ev3dev/ev3dev-lang/blob/master/cpp/ev3dev.cpp#L579)
```C++
i2c_sensor::i2c_sensor(port_type port_, address_type address_)
{
  connect({{ "port_name", { port_ }},
           { "driver_name",      { nxt_i2c_sensor }},
           { "address",   { address_ }}});
}
```

And compared it to what I have in my fork and what C++ bindings had in the past which is:

From [my fork](https://github.com/bmegli/ev3dev-lang-cpp/blob/master/ev3dev.cpp#L614) and also [past c++ bindings](https://github.com/ddemidov/ev3dev-lang-cpp/commit/0f13224bc0619cb100c228bd6e3ea965307e52ed)
```C++
i2c_sensor::i2c_sensor(address_type address, const std::set<sensor_type> &type):
  sensor(address,  type)
{
	
}
```

I think that new implementation no longer does the same. 

The new bindings implementation will try to force `nxt_i2c_sensor` as driver_name.
The driver_name reported by the device (mi-xg1300l I suppose) will not match and bindings device will not be created/connected. 

@ddemidov, can you have a look at this?

bmegli, 2017-12-03:
TLDR version:

You need both:
1. `sudo sh -c 'echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-X/new_device'`
2.  `ev3dev::i2c_sensor gyro(correct_address, {"mi-xg1300l"});`

Find out the `correct_address` from the console from `/sys/class/lego-sensor/sensorX/address`

LeeJenkins, 2017-12-05:
@bmegli Thanks for the information! Now I am seeing sensor.connected() return true. I really appreciate you taking the time to provide an answer.

closed 2017-12-05 by LeeJenkins

LeeJenkins, 2017-12-06:
Two quick notes:

1. For the sake of future devs who may come here searching for the same info, the registration command that @bmegli typed needs an "s": "device" should be "devices", like this:

    sudo sh -c 'echo "mi-xg1300l 0x01" > /sys/bus/i2c/device/i2c-X/new_device'

2 The documentation in [Appendix A: Sensors](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#microinfinity-digital-gyroscope-and-accelerometer) is incorrect. Footnote [42] says this:

    Register I2C device:
        echo mi-xg1300l 0x01 > /sys/class/lego-port/port<N>/set_device

It might be nice to have the documentation updated. There was a link on the page to edit the docs in GitHub, but when I clicked it didn't look like anything I know how to edit.

dlech, 2017-12-06:
The "incorrect" command is actually correct for BrickPi, BrickPi3 and PiStorms. It could also be made to work on EV3/EVB as discussed above.