ev3dev/ev3dev #1014: brickd uses 60%+ of CPU while idling after downloading from VSCode

WasabiFan, 2017-12-21:
From immediately after downloading a project via VSCode until the brick is restarted, brickd eats 60-70% of the CPU, even if Code is closed afterward. It also consumes ~8% of the RAM.

```
  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  300 root      20   0   17212   4400   4080 R 69.0  7.7   1:24.73 brickd
```

ndward, 2017-12-30:
I love the VS Code EV3 python workflow but have one huge problem: scripts take an incredibly long time to launch: between 20s and more than two minutes. Here's an example of a script that takes more than two minutes before showing text on the LCD:

```
#!/usr/bin/env python3
from ev3dev.ev3 import *
from PIL import Image, ImageDraw, ImageFont
import ev3dev.fonts as fonts
from time import sleep

lcd = Screen()

# Connect ultrasonic sensor to any sensor port
# and check it is connected.
us = UltrasonicSensor() 
assert us.connected, "Connect a US sensor to any sensor port"
# Put the US sensor into distance mode.
us.mode = 'US-DIST-CM'

while True:    # forever
    lcd.clear()
    lcd.draw.text((45, 40), str(us.value()/10)+' cm', font=fonts.load('helvB24')) # convert to cm
    lcd.update()
    sleep(0.1) # so the display doesn't change too frequently
```

The scripts that take forever to launch are the ones that import PIL, fonts etc, which I assume to be relatively large modules. The slowness is the same whether I launch the script from VS Code or from Brickman. I'm using Win10 and thought my PC might be the cause of the problem but trying on another Win10 machine I got the same result. So I suppose either my EV3 is defective or else the issue raised her by @WasabiFan is the cause, but then many other people would be experiencing this problem, which doesn't seem to be the case. My scripts on the PC are in a folder that contains only scripts - there are none of the special files and folders that were included in the vscode-hello-python-master project since my impression is these files are not needed. Am I wrong?
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2017-12-13
Kernel version:     4.9.58-ev3dev-1.6.0-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340377E
BOARD_INFO_TYPE=main

dlech, 2017-12-30:
@ndward, how much of the two minutes is downloading files and how much of the time is actually running the file? You can watch the status bar at the bottom of the window in VS Code to see when files are downloading, then you should see output in the "Output" pane when the actual program starts when the downloading is done.

ndward, 2017-12-30:
The downloading is very fast e.g. 50 scripts in 6 seconds. When I right-click the above-mentioned script in the browser and choose run, I immediately see 'Started' in the output panel, but text appears on the ev3 lcd only 2 minutes later. If I start the same script from Brickman, it also takes more than two minutes before text appears on the LCD.
While I have your attention, @dlech , may I suggest that it would be great to have a 'download and run' option when right-clicking a script in the folder panel (as opposed to the ev3 browser panel). It's annoying to have to keep switching between the folder panel (to open files in the editor) and the ev3 browser (to run specific scripts, assuming one does not want to keep modifying launch.json to target different scripts). Also, there is a risk that the user who has saved modifications to a file will then run the file from the browser, forgetting that she must first download all the files to the brick. Also, being able to download and run a specific file would mean that it would no longer be necessary to download and run ALL the script files in the project folder, as currently seem to be the case.
Congrats for the VS Code extension which makes working with python on the EV3 brilliantly simple, except for the slowness problem I'm having.

dlech, 2017-12-30:
Thanks for the info. It sounds like it might indeed be the same issue here.

And, as always, please open a new issue for any new ideas or problem, otherwise, they will get lost and forgotten.

ndward, 2017-12-31:
@WasabiFan wrote 'From immediately after downloading a project via VSCode until the brick is restarted, brickd eats 60-70% of the CPU'. I tried restarting the brick and re-ran the above-mentioned script which took more than two minutes to start when launched from VS Code or Brickman. After restarting the brick, the same script, launched from Brickman, started displaying text after about one minute, which is better than before but still very slow. Then I connected the brick to VS Code and ran the same script _from Brickman_: one minute to start. Then I ran the same script _from VS Code_: one minute to start. But this is the same script that had taken more than two minutes to start before I restarted the brick, which suggests to me that there is a problem that accumulates with the time the brick has been running. @WasabiFan , I'd like to generate the same data (%CPU etc) that you did - can you please tell me what I need to do?

WasabiFan, 2017-12-31:
I booted the EV3, connected from VSCode, clicked the "download" button, and then ran `top` via SSH (which shows the CPU usage of every process, <kbd>Ctrl</kbd>+<kdb>C</kbd> to close `top`).

ndward, 2017-12-31:
This is what I got while the ev3 was idle (of course, the values vary a bit with time):
```
 PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
   73 root     -51   0       0      0      0 S 14.8  0.0   8:52.46 irq/245-ti-ads7
   61 root      20   0       0      0      0 S 13.9  0.0  11:24.39 kworker/0:3
  743 robot     20   0    6428   2796   2336 R  7.1  4.9   0:24.22 top
  331 avahi     20   0    6360   2168   1996 S  1.5  3.8   0:47.28 avahi-daemon
  476 root      20   0   51660   5568   4164 S  1.5  9.7   2:07.25 brickman
  627 robot     20   0   11660   2636   2324 S  0.9  4.6   0:08.46 sshd
```
This while downloading:
```
 PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
   61 root      20   0       0      0      0 S 17.5  0.0  16:41.99 kworker/0:3
   73 root     -51   0       0      0      0 S 16.6  0.0  11:47.77 irq/245-ti-ads7
  476 root      20   0   51660   5428   4288 S 16.3  9.5   2:37.11 brickman
  627 robot     20   0   11660   2632   2304 R 12.1  4.6   0:15.44 sshd
  855 robot     20   0    6432   2812   2384 R  7.7  4.9   0:04.54 top
  629 robot     20   0    2296   1324   1208 S  3.8  2.3   0:01.40 sftp-server
  331 avahi     20   0    6360   2160   1992 S  1.8  3.8   1:07.87 avahi-daemon
```
Adding up the %CPU values (including the small ones not shown above) I get 78.5%

This while the above-mentioned script was starting from VS Code (right-click+run in the ev3 browser):
```
  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  836 robot     20   0   12728   9172   5108 R 56.7 16.0   0:06.74 python3
   61 root      20   0       0      0      0 S 15.4  0.0  15:21.33 kworker/0:3
   73 root     -51   0       0      0      0 S 15.1  0.0  11:01.18 irq/245-ti-ads7
  824 robot     20   0    6432   2848   2416 R  6.5  5.0   0:02.25 top
  331 avahi     20   0    6360   2164   1992 S  1.8  3.8   1:02.44 avahi-daemon
  476 root      20   0   51660   5268   4112 S  1.6  9.2   2:26.20 brickman
```
Adding up the %CPU values (including the small ones not shown above) I get 100% - is this a problem? Does that mean the CPU is 'saturated' in some way? Could this be the cause of the slow starts?

This once the script had fully started (text displayed on LCD):
```
 PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  794 robot     20   0   21244  16224   7712 R 28.6 28.3   1:48.59 python3
   61 root      20   0       0      0      0 S 27.7  0.0  13:19.80 kworker/0:3
   73 root     -51   0       0      0      0 S 12.9  0.0  10:08.67 irq/245-ti-ads7
  800 robot     20   0    6432   2748   2316 R  6.0  4.8   0:01.92 top
```
Adding up the %CPU values (including the small ones not shown above) I get 83.6%

WasabiFan, 2017-12-31:
The situation described in this issue isn't what you are experiencing; `brickd` doesn't show up in the lists you posted. My tests were run on an image from a few months earlier, which may be the underlying difference.

ndward, 2018-03-29:
I'm sorry to note that the March 16 release of the EV3 Stretch OS had not fixed this problem of slowness, but it was not to be. My 14 line test script (see https://github.com/ev3dev/ev3dev/issues/1022) still takes up to 2m48s to start on my EV3 i.e. to begin displaying values. I at last managed to persuade someone, @QuirkyCort in Singapore, to kindly try running the script and he tells me that using the 16 March version of the Stretch OS it took about 35s to start, why I still find to be unreasonably long but which is much faster than on my system. Note that my very slow times are the same whether I launch the script from VS Code or from Brickman. So I am tempted to conclude that my EV3 has a hardware problem and that I need to buy a new EV3. **However I have been able to determine that a large part of the slowness comes from the process brickd becoming 'hyperactive' and taking over much of the CPU.**

Just after rebooting the EV3 I am able to get much faster times to start my test script, around 60s.
I spend several hours trying to identify what causes the launch time to slow down to 2m30s at some point and concluded that **the problem occurs when the connection to the brick is lost and then re-established.**

Here is a reproducible procedure on my Win64 system to cause the launching of the above script to pass from about 60s to about 2m60s:

1. Choose 'disconnect' in the EV3 browser within VS Code
2. Choose 'reconnect' in the EV3 browser

**Once I have done that the brickd process on the EV3 seems to go hyperactive and running 'top' via SSH indicates that brickd occupies about 58% of the CPU even while the EV3 is idle.** 

This is 'top' snapshot after rebooting the EV3, with the EV3 idle:
[image]

Here is a top snapshot taken while the EV3 is idle but after brickd has become hyperactive due to the disconnect/reconnect:
[image]
Can I assume that '%CPU(s) 100.0 sy' means that the processor is saturated? Is that a problem? Once brickd has become hyperactive the only way I know of getting it out of that state is to reboot the EV3. I don't seem to be able to kill the brickd process with `kill -9 [PID]` or `killall -9 brickd`, for example. It seems to me there is something very wrong with brickd, or the way it is handled.

But interestingly **if the disconnect is done by unplugging and then reconnecting the USB cable between the PC and the EV3 then I can use 'reconnect' within the VS Code EV3 browser WITHOUT causing brickd to become hyperactive.** So it would appear that it is not the 'reconnect' that triggers the brickd problem in my case, but the 'disconnect' done in VS Code?

My observations confirm what @wasabifan reported in this issue in the sense that brickd ends up going hyperactive sometimes, but I identify the trigger as reconnecting to the EV3 whereas @wasabifan apparently gets a similar problem just from downloading a project to the EV3. I find that if I choose 'Send project to device' in the EV3 browser window it does NOT cause brickd to go hyperactive. Also, if I use F5 to download and run a project it does NOT cause brickd to go hyperactive.

I don't think the brickd problem is the only thing causing scripts to take an abnormally long time to open, but it still accounts for most of the delay in the case of my test script, since it extends the start time from 60s to about 2m30s. I stress that once brickd has gone hyperactive this very delayed start occurs whether the script is run from VS Code or from Brickman. It may be that there are other actions that can cause brickd to become hyperactive. I don't know whether disconnections due to lost WiFi or Bluetooth connections also trigger the problem. I understand that (at least in March 2017) the function of brickd was only to shut down the brick should the battery voltage fall too low – it's annoying that such a minor process can go wrong in such a damaging way. I hope the problem with brickd will be fixed with the next release.

dlech, 2018-04-15:
This should be fixed in `brickd_1.1.2`. I haven't made a snapshot image that includes this yet, so `sudo apt update && sudo apt upgrade` for now.

P.S. Thanks @ndward for digging in and finding a simple, reliable way to reproduce the problem.

closed 2018-04-15 by dlech