ev3dev/ev3dev #1015: Have brickrun warn user when running a script with Windows-style line endings

WasabiFan, 2017-12-21:
### What are you trying to do?

Run a script after forgetting to switch to Unix-style line endings.

### What did you expect to happen?

To be warned of the mistake.

### What actually happened?

```
/bin/bash^M: bad interpreter: No such file or directory
```

or

```
/usr/bin/env: 'python3\r': No such file or directory
```

### What hardware and software are you using (including version numbers)?

```
========== ev3dev-sysinfo ==========
Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340720B
BOARD_INFO_TYPE=main
```

tcwan, 2018-01-10:
Won't it be easier just to run the script through a EOL sanitizer before attempting to execute it?

dlech, 2018-01-11:
Yes, but I think a warning would still be good so that the user knows they are doing something wrong and their script would start slightly faster if we didn't have to convert it every time it is downloaded again with the wrong line endings. It would also be good for the VS Code extension to do this before we get too.