ev3dev/ev3dev #1016: GDBus.Error: net.connman.Error.OperationAborted when connecting to Wi-Fi

fferri, 2017-12-21:
I'm trying to connect to my Wi-Fi network, using a TP-LINK TL-WN722N high gain dongle.

The Wi-Fi functionality seems to work fine. As soon as I connect the dongle I can power on the Wi-Fi adapter, I can see a list of networks, and selecting my network I am prompted for the password. 

However after some seconds, the error message
```
Error:

GDBus.Error: net.connman.Error.OperationAborted: Operation aborted
```
and also:
```
Error:

connect-failed
```
is shown on the screen.

It seems it is not a problem of invalid key. I tried to purposefully enter an invalid key, and I get the same `GDBus.Error: net.connman.Error.OperationAborted` error followed by:
```
Error:

invalid-key
```

EDIT: after several attempts, I succeded connecting to Wi-Fi

**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-09-30
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534FF0EA
BOARD_INFO_TYPE=main
```

dlech, 2018-01-01:
Could you try connecting with `connmanctl`? 

http://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/

AssyaB, 2018-07-31:
I also have a similar problem : 
Sometimes Raspberry Pi connects to wifi normally, but most of the times it doesn't work. When I plug a screen and use the interface to connect I get the following error message: `GDBus.Error: net.connman.Error.OperationAborted: Operation aborted`.
So I took an Ethernet cable and connected directly to my computer via PuTTy and tried to connect to the wifi via `connmanctl `. This is what happened:
```
login as: robot
Using keyboard-interactive authentication.
Password:
Linux ev3dev 4.14.54-ev3dev-2.2.0-rpi2 #1 SMP Sun Jul 15 15:06:58 CDT 2018 armv7                                                                             l
             _____     _
   _____   _|___ /  __| | _____   __
  / _ \ \ / / |_ \ / _` |/ _ \ \ / /
|  __/\ V / ___) | (_| |  __/\ V /
  \___| \_/ |____/ \__,_|\___| \_/

Debian stretch on LEGO MINDSTORMS EV3!
Last login: Tue Jul 31 13:17:20 2018 from 10.245.2.30
robot@ev3dev:~$ connmanctl
Error getting VPN connections: The name net.connman.vpn was not provided by any                                                                              connmanctl> enable wifi
Error wifi: Already enabled
connmanctl> scan wifi
Scan completed for wifi
connmanctl> services
*AR Wired                ethernet_b827eb2f6a87_cable
*A  Capgemini            wifi_b827eb7a3fd2_43617067656d696e69_managed_psk
    DIRECT-D3661C1F      wifi_b827eb7a3fd2_4449524543542d4433363631433146_managed_psk
                         wifi_b827eb7a3fd2_hidden_managed_wep
    Company - Wireless WiFi wifi_b827eb7a3fd2_436f6d70616e79202d20576972656c6573732057694669_managed_psk
    dlink                wifi_b827eb7a3fd2_646c696e6b_managed_none
    skylightscts0        wifi_b827eb7a3fd2_736b796c696768747363747330_managed_psk
    skylightshub         wifi_b827eb7a3fd2_736b796c6967687473687562_managed_psk
    boxmyjob             wifi_b827eb7a3fd2_626f786d796a6f62_managed_wep
    skylightscts10       wifi_b827eb7a3fd2_736b796c69676874736374733130_managed_psk
    SKYLIGHTS - Wireless WiFi wifi_b827eb7a3fd2_534b594c4947485453202d20576972656c6573732057694669_managed_psk
    SIGFOX               wifi_b827eb7a3fd2_534947464f58_managed_psk
    SIGFOX_GUEST         wifi_b827eb7a3fd2_534947464f585f4755455354_managed_psk
    dlink-46F3           wifi_b827eb7a3fd2_646c696e6b2d34364633_managed_psk
    CAPTURS              wifi_b827eb7a3fd2_43415054555253_managed_psk
    IOT-VALLEY           wifi_b827eb7a3fd2_494f542d56414c4c4559_managed_psk
    Volvic               wifi_b827eb7a3fd2_566f6c766963_managed_psk
                         wifi_b827eb7a3fd2_hidden_managed_none
    DWR-921-1613         wifi_b827eb7a3fd2_4457522d3932312d31363133_managed_psk
    AI-THINKER_19B041    wifi_b827eb7a3fd2_41492d5448494e4b45525f313942303431_managed_none
    dlink-46F3-5GHz      wifi_b827eb7a3fd2_646c696e6b2d343646332d3547487a_managed_psk
    Wifi-Connit-5GHz     wifi_b827eb7a3fd2_576966692d436f6e6e69742d3547487a_managed_psk
connmanctl> agent on
Agent registered
connmanctl> connect wifi_b827eb7a3fd2_43617067656d696e69_managed_psk
Agent ReportError wifi_b827eb7a3fd2_43617067656d696e69_managed_psk
  connect-failed
Agent request cancelled by ConnMan
Error /net/connman/service/wifi_b827eb7a3fd2_43617067656d696e69_managed_psk: Operation aborted
connmanctl>

```

The sysinfo returns:
```

[‎31/‎07/‎2018 16:13]  HAUPT, Raphael:  
robot@ev3dev:~$
robot@ev3dev:~$ **System info (from `ev3dev-sysinfo`)**
-bash: syntax error near unexpected token `('
robot@ev3dev:~$ ```
> Image file:         ev3dev-stretch-rpi2-generic-2018-07-15
> Kernel version:     4.14.54-ev3dev-2.2.0-rpi2
> Brickman:           0.10.0
> BogoMIPS:           38.40
> Board:              board0
> BOARD_INFO_HW_REV=A020D3
> BOARD_INFO_MODEL=Raspberry Pi 3 Model B Plus Rev 1.3
> BOARD_INFO_SERIAL_NUM=00000000d02f6a87
> BOARD_INFO_TYPE=main
> Board:              board1
> BOARD_INFO_FW_VER=1.4.3
> BOARD_INFO_HW_REV=3.2.1
> BOARD_INFO_MODEL=Dexter Industries BrickPi3
> BOARD_INFO_SERIAL_NUM=472728DE515035524E4A2020FF070D2D
> BOARD_INFO_TYPE=aux
> 
 

```
Any ideas where the problem could be ?

dlech, 2018-07-31:
Are there any errors in the kernel log (output of `dmesg` command)?

AssyaB, 2018-08-01:
I didn't find an error in the kernel log.
You can look at the result of the command `dmesg` :
```
robot@ev3dev:~$ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.54-ev3dev-2.2.0-rpi2 (david@freyr) (gcc version 6.4.1 20171012 (Linaro GCC 6.4-2017.11)) #1 SMP Sun Jul 15 15:06:58 CDT 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000] free_area_init_node: node 0, pgdat 80c85700, node_mem_map ba3a1000
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 17 pages/cpu @ba349000 s38604 r8192 d22836 u69632
[    0.000000] pcpu-alloc: s38604 r8192 d22836 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait consoleblank=0
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940224K/970752K available (7168K kernel code, 577K rwdata, 2232K rodata, 1024K init, 710K bss, 22336K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x80008000 - 0x80800000   (8160 kB)
                     .init : 0x80b00000 - 0x80c00000   (1024 kB)
                     .data : 0x80c00000 - 0x80c9071c   ( 578 kB)
                      .bss : 0x80c98294 - 0x80d49d64   ( 711 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26405 entries in 78 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000269] Console: colour dummy device 80x30
[    0.000788] console [tty1] enabled
[    0.000822] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000861] pid_max: default: 32768 minimum: 301
[    0.001173] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001204] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002149] Disabling memory control group subsystem
[    0.002240] CPU: Testing write buffer coherency: ok
[    0.002647] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003032] Setting up static identity map for 0x100000 - 0x10003c
[    0.003160] Hierarchical SRCU implementation.
[    0.003821] smp: Bringing up secondary CPUs ...
[    0.004519] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.005262] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.005986] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.006088] smp: Brought up 1 node, 4 CPUs
[    0.006158] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.006179] CPU: All CPU(s) started in HYP mode.
[    0.006196] CPU: Virtualization extensions available.
[    0.007027] devtmpfs: initialized
[    0.016836] random: get_random_u32 called from bucket_table_alloc+0x100/0x254 with crng_init=0
[    0.017446] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.017684] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.017727] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.018312] pinctrl core: initialized pinctrl subsystem
[    0.019051] NET: Registered protocol family 16
[... 217 lines cut ...]
[    4.229024] Bluetooth: HCI device and connection manager initialized
[    4.233792] Bluetooth: HCI socket layer initialized
[    4.237058] Bluetooth: L2CAP socket layer initialized
[    4.240403] Bluetooth: SCO socket layer initialized
[    4.256779] Bluetooth: HCI UART driver ver 2.3
[    4.260226] Bluetooth: HCI UART protocol H4 registered
[    4.263470] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    4.266889] Bluetooth: HCI UART protocol Broadcom registered
[    4.695935] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[... cut ...]
[    6.278413] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[    7.215087] IPv6: ADDRCONF(NETDEV_UP): enxb827eb2f6a87: link is not ready
[    8.311449] Under-voltage detected! (0x00050005)
[    9.966891] IPv6: ADDRCONF(NETDEV_CHANGE): enxb827eb2f6a87: link becomes ready
[   12.471432] Voltage normalised (0x00000000)
[   27.139836] systemd-journald[109]: File /var/log/journal/d33dd43b67924683bccf4b7105b79717/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
```

Rombersoft, 2022-04-05:
Unfortunately this problem is not resolved up to this time yet...

jajir, 2022-05-12:
I have same problem with EV3. Unable to connect to local Wi-Fi with image "ev3dev-stretch-ev3-generic-2020-04-10.img"

jajir, 2022-05-13:
Problem could be related to microSD card. I change card to new one and it works fine.

taylor-cz, 2022-07-24:
I have exactly the same issue with wifi connection. Ran out of ideas, so I'd like to try to use another SD card too. May I ask which one worked?

jajir, 2022-07-25:
SD card replacement doesn't help. I found that reinstalling of image really helped. I used card Kingston CANVAS Select plus 32GB SDCS2/32GB 31696-001.A00LF.

taylor-cz, 2022-07-26:
I see, that's a bad news. I reflashed recent image from https://github.com/ev3dev/ev3dev/releases/download/ev3dev-stretch-2020-04-10/ev3dev-stretch-ev3-generic-2020-04-10.zip to SD card this weekend before starting experimenting with wifi connection again. 
May I ask which USB wifi adapter are you using and possibly what wifi AP parameters?
I'm using Edimax Wireless Nano USB 2.0 Adapter. Using an AP on 2.4GHz with WPA2/PSK (AES) security.

Since typing the passphrase on the brick is cumbersome, I used this procedure https://www.ev3dev.org/docs/tutorials/setting-up-wifi-using-the-command-line/ for all tries except first one.

taylor-cz, 2022-07-30:
Please anybody can share working configuration regarding parameters in the previous post?
This is really pain when there are too much variables what can be wrong.

muellerj, 2023-01-26:
I'm having the same problem with `ev3dev-stretch-ev3-generic-2020-04-10.img`. Is there an old image known to be working?

muellerj, 2023-01-26:
Update: I managed to get this working by enabling `Allow legacy 802.11b rates` on my router. The problem was not connected to the SD card or the release.

SunixLiu, 2025-01-11:
For me, ev3dev-stretch-ev3-generic-2020-04-10.img doesn't work, but this one: ev3dev-stretch-ev3-generic-2019-10-23.img works, seems driver issue?