ev3dev/ev3dev #1017: Driver Compile Settings for T1U USB Dongle

antonvh, 2017-12-22:
### What are you trying to do?
I'm trying to use a 5GHz USB stick for Lego World 2018, because the 2.4 band is usually full.

### What did you expect to happen?
Compilation succeeded.

### What actually happened?
Downloaded drivers from here: 
http://static.tp-link.com/res/down/soft/Archer_T1U_V1_150909.zip

```
robot@ev3dev-2:~/Archer_T1U_V1_150909/Driver$ sudo make
make -C UTIL/ osutil
make[1]: Entering directory '/home/robot/Archer_T1U_V1_150909/Driver/UTIL'
cp -f os/linux/Makefile.6.util /home/robot/Archer_T1U_V1_150909/Driver/UTIL/os/linux/Makefile
make -C /lib/modules/4.4.47-19-ev3dev-ev3/ SUBDIRS=/home/robot/Archer_T1U_V1_150909/Driver/UTIL/os/linux modules
make[2]: Entering directory '/lib/modules/4.4.47-19-ev3dev-ev3'
make[2]: *** No rule to make target 'modules'.  Stop.
make[2]: Leaving directory '/lib/modules/4.4.47-19-ev3dev-ev3'
Makefile:527: recipe for target 'osutil' failed
make[1]: *** [osutil] Error 2
make[1]: Leaving directory '/home/robot/Archer_T1U_V1_150909/Driver/UTIL'
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 2
```

In the compilation instructions it says to make sure these settings are right:
```
#PLATFORM: Target platform
PLATFORM = PC

ifeq ($(PLATFORM),PC) 
# Linux 2.6 
LINUX_SRC = /lib/modules/$(shell uname -r)/build 
# Linux 2.4 Change to your local setting 
#LINUX_SRC = /usr/src/linux-2.4 
LINUX_SRC_MODULE = /lib/modules/$(shell uname -r)/kernel/drivers/net/wireless/ CROSS_COMPILE = 
endif
```

What should LINUX_SRC =  and LINUX_SRC_MODULE =  be on ev3dev?

antonvh, 2017-12-22:
I'm trudging on.
Editing 3 makefiles to say:
LINUX_SRC = /usr/src/linux-headers-$(shell uname -r)/

but now error: mach/irqs.h: No such file or directory

antonvh, 2017-12-22:
I returned the dongle. I thoroughly dislike compiling old drivers.

closed 2017-12-22 by antonvh

dlech, 2017-12-22:
Yeah, if this driver has Linux 2.4 as an option, I think it is a lost cause.

There seem to be very few 5GHz options for Linux. [this page](https://www.elinux.org/RPi_USB_Wi-Fi_Adapters) mentions RT5572 USB ID 148f:5572.

Maybe some ideas [here](https://www.raspberrypi.org/forums/viewtopic.php?f=46&t=31019)

antonvh, 2017-12-22:
I ordered an AC600. Hope it works better. It's a little larger, though. http://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/global/wireless_adapters_ac600_dual-band/ew-7811utc