ev3dev/ev3dev #1018: EV3 not booting into EV3DEV

ACEslava, 2017-12-27:
**Problem:**
Trying to boot into ev3dev-jessie-ev3-generic-2017-09-14 on an 8GB SDHC. EV3 simply boots into normal firmware.
**Possible cause:**
- Missing Root partition.
**Attempted Fix:**
- Tried ev3-ev3dev-jessie-2015-12-30, did not work.
- Tried different image writer, did not work.
- Tried repartitioning SD, did not work.
- Tried different SD, did not work.

dlech, 2017-12-27:
Does your EV3 recognize the SD card when using the official firmware? I'm wondering if it has a bad SD card reader. The only way to debug this problem is to use a serial adapter on input port 1. 

http://botbench.com/blog/2013/08/15/ev3-creating-console-cable/

dlech, 2017-12-27:
Also, check the firmware version of the EV3.

ACEslava, 2017-12-27:
Firmware is 1.09 as of yesterday, that was one of my attempted fixes. 

     Before I destroy an EV3 cable, I want to see if I can get the ev3dev root partition on the sd, the boot partition is the only one. Another partition is created while writing to the SD but it is blank and my computer (Win10) says it's unusable unless I format the partition.

dlech, 2017-12-27:
The second partition is not actually blank. It is just a Linux file system that windows does not recognize. You should not make any changes at all to the SD card after you flash it.

It the official firmware does not recognize the SD card, then I think your EV3 is defective and you should contact LEGO customer support.