ev3dev/ev3dev #1019: Project won't build due to javadoc error in Sysfx.java

phowson, 2017-12-28:
### What are you trying to do?
Run mvn install for ev3dev

### What did you expect to happen?
The project to correctly build

### What actually happened?
The project does not build due to a minor javadoc error on Sysfx, requires a one-line change

`
[ERROR] D:\devel\lego\ev3dev-lang-java\src\main\java\org\ev3dev\io\Sysfs.java:67: error: @param name not found
[ERROR] * @param enabled Enabled
[ERROR] ^
`
### What hardware and software are you using (including version numbers)?

dlech, 2018-01-01:
Which ev3dev-lang-java are you using?

jabrena, 2018-01-01:
I think that he is using this library:
https://github.com/mob41/ev3dev-lang-java
https://github.com/mob41/ev3dev-lang-java/tree/develop/src/main/java/org/ev3dev/io

mob41, 2018-01-03:
Noted. I will look through the JavaDoc problems these days. Thanks!

closed 2018-04-20 by dlech