ev3dev/ev3dev #1020: VS Code suggestions

ndward, 2017-12-30:
First, congratulations to @dlech and everyone else involved for making the VS Code extension - it makes working with python on the EV3 brilliantly simple. I find the VS Code workflow to be excellent for beginners – far easier than previous workflow options. I can imagine that the VS Code workflow could easily be used in a classroom context, which interests me greatly as a computer science teacher. Now that it is no longer necessary to run commands like sudo chvt 6 (to give the script control over the LCD) or chmod +x filename (to make a file executable) it should no longer be necessary for beginners to use the SSH terminal at all. The only major problems that I have with VS code currently (on Windows 10 PC) are the extreme slowness with which scripts launch (20 seconds to 2 minutes or more) – see issue #1014 - and the instability of wireless connections (I lose WiFi connections after a few minutes, and am unable to establish Bluetooth connections).

SUGGESTIONS:
1.	Currently, if I open a different project folder, I lose the connection to the brick. It would be good if the connection could be maintained, or automatically re-established.
2.	When I connect by WiFi, I usually lose the connection after a few minutes or tens of minutes. This may be a fault of my WiFi, but it would be good if the connection could be re-established automatically.
3.	Users unfamiliar with VS Code will expect to be able to open, edit and run (and in our context, download and run) a loose file, as opposed to a folder. Opening and editing a loose file is indeed straightforward, and running in the terminal is possible (though it seems you have to right-click the editor and choose 'run in terminal', which is not obvious). But it seems that in VS Code there is no easy way to download a loose file to the ev3. Hopefully users will quickly realise that in VS Code it is more or less essential to open an entire project folder rather than individual files.
4.	For new users, it's not obvious how to reconnect or connect to a different device (right-click on or next to the red button in the ev3 browser window). Can it be made clearer?
5.	When connecting, I often see multiple connection choices, even when only one connection is available. Currently, with only a WiFi connection available, I am given a choice between 'ev3dev-4 Ethernet' (a name I never gave to any connection) and 'ev3dev'. Sometimes I have to choose between the options by trial and error, which is unsatisfactory.
6.	To connect to the _same_ device with a different connection type (e.g. switch from USB to WiFi connection) one has to choose 'connect to a different device', which isn't obvious or logical.
7.	About 10% of the python scripts that I have used previously without problems would not run in VS Code because they had CRLF line endings. Ideally VS Code should automatically convert line endings to LF at some point, either on the PC on when downloaded. Or they could be flagged in some way, or a warning message could be given. Currently, if I try to run a file with CRLF line endings I get the unhelpful error message '/usr/bin/env: 'python3\r': No such file or directory' – can the error message be made more helpful? VS Code indicates the line ending type in the bottom-right hand corner of the window and clicking on that indication allows for CRLF to be changed to LF, which is very neat. 
8.	**It would be great to have a 'download and run' option when right-clicking a script in the folder panel (as opposed to the ev3 browser panel).** It's annoying to have to keep switching between the folder panel (to open files in the editor) and the ev3 browser (to run specific scripts, assuming one does not want to keep modifying launch.json to target different scripts). Also, there is a risk that the user who has saved modifications to a file will then run the unmodified file from the ev3 browser, forgetting that she must first download all the files to the brick. Also, being able to download and run a specific file would mean that it would no longer be necessary to download and run ALL the script files in the project folder, as currently seems to be the case.
9. When I connect to the EV3 the ev3 browser does NOT automatically show me the contents of home/robot, even though the triangle is pointing down which would normally indicate that the contexts are being shown. It would be good if the contents of home/robot could be shown automatically.
10. When I disconnect my wired (USB) connection via Brickman, and no other connection is present, so that Brickman displays 'No connections', the connection to VS Code is in fact still there - I can still run scripts from VS Code etc. Odd?
11. There is no obvious way within VS Code to stop scripts that are running on the EV3. Pressing the back button on the EV3 may not always be convenient e.g. when the EV3 robot is at the other side of the room. I have found that within VS Code I can force a script to stop running on the ev3 by attempting to run another script, but that is obviously unsatisfactory.
12. To start the python interpreter in the terminal I have to type 'python' ('python3' will not work) whereas in the SSH terminal I have to type 'python3'. This is unintuitive.
13. Files are ordered differently in the folder panel and the ev3 browser panel, which is confusing. For example, the folder panel puts 1-9.py before 1-10.py while the ev3 browser (and the Brickman file browser) puts 1-10.py before 1-9.py. Ideally, I think the ev3 browser and the Brickman file browser should order files like VS Code does in the folder panel, especially if the VS Code workflow is going to be the focus of ev3dev development.
14. Assuming that users will put many scripts in each project folder, I think it's likely that the F5 'download and run' shortcut will not be of much use since it would be necessary to retarget the launch.json file each time you want to run a different script with F5.

The following are suggestions and comments for ev3dev rather than VS Code:
1.	The WiFi password entry font is too big – one can't distinguish E and F.
2.	In the Brickman file browser the left and right buttons seem to be 'wasted', duplicating the functionality of the up button. To make these buttons more useful I suggest that the left button could be used to move to the parent folder (unless already in home/robot) and the right button could be used to move to the first file in the list. Also, I would expect that if I hold down the up or down buttons then I should automatically scroll slowly through the files, which currently does not happen. Currently, I sometimes have to press the up or down buttons dozens of times to reach the fie I want, if the folder contains dozens of files, which is a drag.
3.	Bigger fonts! The largest font available currently seems to be size 24 – I would find it useful to have larger fonts available, maybe as big as 100, at least in a couple of font styles.
4.	I request support (if not already supported) for Mindsensors 8 Channel Servo Controller for NXT or EV3 (http://www.mindsensors.com/ev3-and-nxt/25-8-channel-servo-controller-for-nxt-or-ev3)

I'm sure @dlech won't like me putting all these suggestions in one issue, but I could not bring myself to open 17 new issues. Sorry, David!

Image file:         ev3dev-stretch-ev3-generic-2017-12-13
Kernel version:     4.9.58-ev3dev-1.6.0-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340377E
BOARD_INFO_TYPE=main

dlech, 2017-12-31:
Thank you for the complement. :smile: 

> I'm sure @dlech won't like me putting all these suggestions in one issue, but I could not bring myself to open 17 new issues. Sorry, David!

:cry:  I will read it all, but it is too much for one issue and will never be resolved.

A number of the items you have listed already have similar issues open, so should be addressed eventually.

And here is my one answer for one issue:

> I request support (if not already supported) for Mindsensors 8 Channel Servo Controller for NXT or EV3

* http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/sensor_data.html#mindsensors-com-8-channel-servo-controller
* http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-stretch/motors.html#servo-motor-class
* http://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/motors.html#servo-motor

dlech, 2018-01-01:
Answer of the day:

> Bigger fonts! The largest font available currently seems to be size 24 – I would find it useful to have larger fonts available, maybe as big as 100, at least in a couple of font styles.

I assume this is an ev3dev-lang-python suggestion?

dlech, 2018-01-03:
> In the Brickman file browser the left and right buttons seem to be 'wasted',

This will be addressed in #426 since that will require fundamental changes to navigation anyway.

dlech, 2018-01-04:
> The WiFi password entry font is too big – one can't distinguish E and F.

There is already an issue for font size at #462

dlech, 2018-01-08:
> 14. Assuming that users will put many scripts in each project folder, 

I usually only write one main entry point per folder, so let's not base anything on assumptions. 

> I think it's likely that the F5 'download and run' shortcut will not be of much use since it would be necessary to retarget the launch.json file each time you want to run a different script with F5.

There is a user interface specifically for switching between launch configurations, so IMHO I don't see this as a problem. It is using VS Code the way it is intended to be used.

[image]

dlech, 2018-01-08:
> 13. Files are ordered differently in the folder panel and the ev3 browser panel,

This is a VS Code "bug": https://github.com/Microsoft/vscode/issues/27759

dlech, 2018-01-11:
> 12. To start the python interpreter in the terminal I have to type 'python' ('python3' will not work) whereas in the SSH terminal I have to type 'python3'. This is unintuitive.

I imagine this has to do with how you installed Python on your computer. It has been a long time since I installed Python on Windows, so I don't remember if this is an option in the installer or not. On Linux and macOS (via homebrew), `python2` and `python3` both work as expected.

dlech, 2018-01-11:
https://www.python.org/dev/peps/pep-0397/

basically says that Window is just going to be different.

dlech, 2018-03-22:
> 11. There is no obvious way within VS Code to stop scripts that are running on the EV3.

The red stop button is not obvious enough?

[image]

dlech, 2018-03-22:
> 10. When I disconnect my wired (USB) connection via Brickman, and no other connection is present, so that Brickman displays 'No connections', the connection to VS Code is in fact still there - I can still run scripts from VS Code etc. Odd?

How exactly are you starting these scripts?

dlech, 2018-03-22:
> 9. When I connect to the EV3 the ev3 browser does NOT automatically show me the contents of home/robot, even though the triangle is pointing down which would normally indicate that the contexts are being shown. It would be good if the contents of home/robot could be shown automatically.

Agreed. This is the intended behavior (I thought this was working already).

dlech, 2018-03-22:
> 8. It would be great to have a 'download and run' option when right-clicking a script in the folder panel (as opposed to the ev3 browser panel).

Actually, you should be using the debug feature of VS Code. See https://github.com/ev3dev/ev3dev/issues/1020#issuecomment-375475260 for a screenshot or press <kbd>F5</kbd> if you don't want to switch to the debug sidebar.

dlech, 2018-03-22:
> 7. About 10% of the python scripts that I have used previously without problems would not run in VS Code because they had CRLF line endings. Ideally VS Code should automatically convert line endings to LF at some point, either on the PC on when downloaded.

Sounds like a good idea.

> Or they could be flagged in some way, or a warning message could be given. Currently, if I try to run a file with CRLF line endings I get the unhelpful error message '/usr/bin/env: 'python3\r': No such file or directory' – can the error message be made more helpful?

Nope. This is an operating system error message. Not one we have control over.

dlech, 2018-03-22:
> 6. To connect to the same device with a different connection type (e.g. switch from USB to WiFi connection) one has to choose 'connect to a different device', which isn't obvious or logical.

This is a tough one because we have limited information to the OS due to the nature of extension.

dlech, 2018-03-22:
> 5. When connecting, I often see multiple connection choices, even when only one connection is available. Currently, with only a WiFi connection available, I am given a choice between 'ev3dev-4 [...]

Interesting. The name you never gave the connection is a name that was automatically assigned by the operating system. I'm not sure why you would be seeing connections that are not valid though.

dlech, 2018-03-22:
> For new users, it's not obvious how to reconnect or connect to a different device (right-click on or next to the red button in the ev3 browser window). Can it be made clearer?

Perhaps. I would like to make reconnection automatic, if possible.

dlech, 2018-03-22:
> 3. Users unfamiliar with VS Code will expect to be able to open, edit and run (and in our context, download and run) a loose file, as opposed to a folder. Opening and editing a loose file is indeed [...]

Yes, this is a bit of a shortcoming with VS Code. Since it is more of a text editor and less of an IDE, there is some manual configuration required.

My current line of thinking is that our best bet is to make some kind of template that has all of the configuration setup already, then make a way to apply that template to a single file that is opened as you suggest.

dlech, 2018-03-22:
> 2. When I connect by WiFi, I usually lose the connection after a few minutes or tens of minutes. This may be a fault of my WiFi, but it would be good if the connection could be re-established automatically.

https://github.com/ev3dev/vscode-ev3dev-browser/issues/55

https://github.com/ev3dev/vscode-ev3dev-browser/issues/44

dlech, 2018-03-22:
> 1. Currently, if I open a different project folder, I lose the connection to the brick. It would be good if the connection could be maintained, or automatically re-established.

The extension is reloaded when you open a new folder, so nothing we can do about that. Automatic reconnection might be a possibility though.

ndward, 2018-03-31:
Thanks very much, @dlech, for taking the time to respond to those suggestions.

Regarding **point 11**, ' _There is no obvious way within VS Code to stop scripts that are running on the EV3._', I never see the red stop button when launching EV3 scripts from VS Code, even if I use the Debug panel. Am I doing something wrong? I'm guessing you are using a different debug configuration to me.

Regarding **point 10** ' _When I disconnect my wired (USB) connection via Brickman, and no other connection is present, so that Brickman displays 'No connections', the connection to VS Code is in fact still there - I can still run scripts from VS Code etc. Odd?_': I have just rebooted my EV3 and (re)connected to it from VS Code. The EV3 displays 'State: Disconnected' but there is clearly a connection because the VS Code EV3 browser says so and because I can launch scripts on the EV3 in all the usual ways such as F5 or right-click>run in the EV3 browser.

Regarding **point 8** ' _It would be great to have a 'download and run' option when right-clicking a script in the folder panel (as opposed to the ev3 browser panel)._' The reason for that suggestion was that for people who tend to switch often between many different (short) scripts it is tedious to have to keep modifying launch.json to be able to use F5. It's possible to launch the scripts from the EV3 browser but that means constantly moving between the folder panel (to open scripts) and the EV3 browser (to run scripts on the EV3). I may have found a good solution. I modified the line in launch.json
`"program": "/home/robot/${workspaceRootFolderName}/myscript.py"`   to
`"program": "/home/robot/${workspaceRootFolderName}/${relativeFile}"`
With that modification, pressing F5 simply downloads and runs whatever script is active in VS Code – no need to modify launch.json anymore.

Probably most of my suggestions are misguided or trivial – I think the most important ones are point 8, where I propose a modification so that pressing F5 can download and run the active script without having to modify launch.json, and point 11, where the stop button is not visible to me when I run EV3 scripts. But even these suggestions are trivial compared to the enormous 'slow launch' problem discussed in issues https://github.com/ev3dev/ev3dev/issues/1014 and https://github.com/ev3dev/ev3dev/issues/1022 , where the brickd process sometimes goes hyperactive, sometimes adding minutes to the time taken for scripts to launch. I would be great if you could fix that!

A large reason for the delay in running scripts is that, if I understand correctly, the python interpreter has to be launched every time a python script is run. Can't the interpreter be kept running continuously, ready to interpret different scripts that are passed to it? I haven't used RPyC for a while but when I did use it I had the impression that it worked like that, for scripts would run almost instantly, as I recall. Should it be possible to use RPyC with VS Code?

Finally, in the most recent version of VS Code I am 'seeing red' a lot – warnings that my scripts have 'problems' because, for example, 'Unable to import ev3dev.ev3' which is not really a problem but which indicates that the ev3dev module is not installed on the PC. I understand it should be possible to do that, which would get rid of the distracting warnings and would allow Intellisense to work as it should, but I don't know how to install the module on the PC. Do you?

dlech, 2018-04-02:
> Regarding point 11, ' There is no obvious way within VS Code to stop scripts that are running on the EV3.', I never see the red stop button when launching EV3 scripts from VS Code, even if I use the [...]

Something like this?

https://github.com/ev3dev/vscode-hello-python/blob/master/.vscode/launch.json

It has been a while since I have used this, so maybe something has broken.

> The EV3 displays 'State: Disconnected' but there is clearly a connection because the VS Code EV3 browser says so and because I can launch scripts on the EV3 in all the usual ways such as F5 or right-click>run in the EV3 browser.

There must be some sort of connection if you can start a program on your EV3 from VS Code... unless you have a magical EV3.

> "program": "/home/robot/${workspaceRootFolderName}/${relativeFile}"

Yes, this is a good solution if you have many scripts that you want to run.

> Also, in the most recent version of VS Code I am 'seeing red' a lot – warnings that my scripts have 'problems' because, for example, 'Unable to import ev3dev.ev3' which is not really a problem but which indicates that the ev3dev module is not installed on the PC. I understand it should be possible to do that, which would get rid of the distracting warnings and would allow Intellisense to work as it should, but I don't know how to install the module on the PC. Do you?

This is coming from the VS Code python extension. You should be able to install ev3dev-lang-python on your PC using `pip` or `easy_install`. I'm not really familiar with python on windows, so I can't be more specific than that. It would be a good question to ask at https://github.com/ev3dev/ev3dev-lang-python/issues.

> Can't the interpreter be kept running continuously, ready to interpret different scripts that are passed to it?

That is an interesting idea. Yes, I am sure it is possible. Not a trivial task though.

> Should it be possible to use RPyC with VS Code?

I don't see why not.

> I'm not sure which is the best method – the second one is obviously much neater. (Please advise.)

The first. With the second option, once a module is imported, calling import again does nothing, so if you updated your code, it wouldn't be re-imported. Also, you will run into problems with global variables retaining the value from the last time you ran your program, which can cause strange and hard to track down bugs.

> WHAT IS GOING ON?

It is the thing I mentioned about imports only happening once. The first time you import the `ev3dev` module, for example, it imports a bunch of other modules behind the scenes, which takes a long time. The next time you import the same module, there is nothing to do since the module is already imported, so it takes almost no time.

ndward, 2018-04-05:
> There must be some sort of connection if you can start a program on your EV3 from VS Code... unless you have a magical EV3.

In point 10 I used Brickman to 'disconnect' but of course the USB cable is still physically in place. What I am saying is that although Brickman then reports that its state is 'no connections' (and wired says 'disconnected') this is not the case since I can still download and run scripts through the USB cable connection.

An easier demonstration is to establish a USB cable connection between VS Code and the brick with the Brickman 'Wireless and Networks ' status displayed as 'Connected' and 'All Network Connections>Wired' also says 'State: Connected'. If one chooses 'Disconnect' then Brickman reports 'No connections' and 'Disconnected' but the VS Code EV3 Browser is green and it is still possible to launch scripts, run 'top' etc.

About the stop button, I do see that button if I run python scripts locally, but I don't see when running scripts on the EV3. My configuration file is essentially the same as the one you have suggested.

As I understand it, with the standard VS Code EV3 python workflow every time one runs a script on the EV3 from VS Code (or even from Brickman), the python interpreter has to be launched (that only take about 3 seconds) and then any necessary modules such as ev3dev.ev3 have to be loaded, which can take a long time on my system (with Stretch, of course). For example, it takes about 50s to for this simple script to start (it's the fonts that are mainly responsible for the delay - there are more than 400 files in that folder):
```
from ev3dev.ev3 import *
from PIL import Image, ImageDraw, ImageFont
import ev3dev.fonts as fonts
```
It seems very inefficient to have to load the interpreter and the ev3dev.ev3 module every time one wants to run a python script on the EV3, which could mean loading the interpreter and the ev3dev.ev3 module dozens of times per hour. It would be much more efficient to have the python interpreter running continuously and the ev3dev.ev3 module imported and always ready to use (I suppose almost every ev3 python script uses ev3dev.ev3, with the exception of your 'hello' script). I can do this with SSH and the file above starts in 1s instead of 50s (after being run once), but the process is relatively clumsy and would not be good for beginners. Could it become a standard way of working? I understand that in competitions it is very important for scripts to start quickly. Are there major problems with this approach? Maybe there is a problem with global variables?

**Can you confirm that Ctrl-C is the correct way to stop a script that has been started in the live python interpreter?** I've seen messages that indicate that this may not be correct with Python 3.6 (I'm using 3.5.3). 

My test script contains import ev3dev.fonts as fonts but the script only needs one font (helvB24) so it seems unreasonable to import all the other fonts. **How do I import and use just one font?** I tried from ev3dev.fonts import helvB24 without success.

The shebang #!/usr/bin/env python3 points to python3 but the same folder contains also python3m and micropython. What is the difference between python3 and python3m? Is micropython ready to use? When I put micropython in the shebang I get ImportError: no module named 'ev3dev'.

Since I didn't get much feedback about whether other users were having the same problems as myself and therefore could not rule out a possible defect in my brick, I just bought another EV3 kit and can confirm that it behaves in exactly the same way, with scripts taking up to two minutes to start (if the brickd problem kicks in).

By the way, I'm thinking if I could kill brickd then that would stop the brickd going hyper and hogging the CPU – **how do I kill brickd and is it a good idea**?

As always, thanks for your help.

ndward, 2018-04-06:
I just found an explanation of 'top' at https://www.booleanworld.com/guide-linux-top-command/ which makes me think that 'load average' is a better indicator of whether the system is overworked, compared to %CPU 100 sy. The page says that:

 _'The load average section represents the average “load” over one, five and fifteen minutes. “Load” is a measure of the amount of computational work a system performs. On Linux, the load is the number of processes in the R and S states at any given moment. The “load average” value gives you a relative measure of how long you must wait for things to get done. Let us consider a few examples to understand this concept. On a single core system, a load average of 0.4 means the system is doing only 40% of work it can do. A load average of 1 means that the system is exactly at capacity — the system will be overloaded by adding even a little bit of additional work. A system with a load average of 2.12 means that it is overloaded by 112% more work than it can handle.'_

In the screenshot of 'top'  above, the load average values are 2.71, 2.65, 2.5. That must mean that the system is overloaded by 171% more work than it can handle, even though this capture was done when brickd was not hyperactive. This seems to me to be further proof that the system is overwhelmed and explains the very slow start times that I experience. In fact I see in the first screenshot on https://github.com/ev3dev/ev3dev/issues/1014 that even when the EV3 has been rebooted and is idle the load average is 2.68 so **even when idle the system is overloaded by 168% more work than it can handle. Am I reading this right**? I'm assuming the EV3 is a single core system - is that correct?

dlech, 2018-04-08:
> In point 10 I used Brickman to 'disconnect' but of course the USB cable is still physically in place.

This just means that the IPv4 connection is disconnected. The VS Code extension uses a local link IPv6 connection, so as long as the USB cable is plugged in, you are "connected" as far as the VS code extension is concerned.

> For example, it takes about 50s to for this simple script to start (it's the fonts that are mainly responsible for the delay - there are more than 400 files in that folder):

I would suggest reporting this (and any other python-related issues) to https://github.com/ev3dev/ev3dev-lang-python/issues. More python-interested eyes will see it there.

> Could it become a standard way of working? I understand that in competitions it is very important for scripts to start quickly. Are there major problems with this approach? Maybe there is a problem with global variables?

I don't know enough about this off-hand to give an answer. We would just have to try it and see what happens.

> Can you confirm that Ctrl-C is the correct way to stop a script that has been started in the live python interpreter?

<kbd>Ctrl</kbd>+<kbd>C</kbd> simply triggers a `KeyboardInterrupt` exception. How it is handled is up to your script. For example, if you are using threads in your script, it might just stop one thread, but no others. So, it should not be considered a universal way to "stop a script". I don't know of any universal alternative, so AFAIK, you need to carefully craft your script so that a a `KeyboardInterrupt` exception will stop everything if you plan on starting and stopping scripts from the interactive python prompt.

> What is the difference between python3 and python3m?

https://lmddgtfy.net/?q=What%20is%20the%20difference%20between%20python3%20and%20python3m%3F

> Is micropython ready to use?

https://github.com/ev3dev/ev3dev/issues/937#issuecomment-332264115

> how do I kill brickd and is it a good idea?

```
sudo systemctl stop brickd # stop currently running instance
sudo systemctl disable brickd # prevent it from starting at boot
```

If you disable it, your brick will not shut down automatically on low battery (could damage rechargeable batteries) and some features of the VS Code extension may not work.

dlech, 2018-04-08:
> even when idle the system is overloaded by 168% more work than it can handle. Am I reading this right?

Yes, I believe so. It would be better to discuss this topic under #1022.

dlech, 2018-07-20:
I think most important ideas here have been addressed, so closing. We can always open a new issue for specific action items.

closed 2018-07-20 by dlech