ev3dev/ev3dev #1021: Compiling RTL8821AU drivers for Ev3 brick (Edimax AC600)

antonvh, 2017-12-31:
### What I want
A 5Ghz USB Wifi stick in my EV3. T2U had old drivers. Trying Edimax AC600.
It doesn't show in conman, like my trusty 2.4Ghz edimax stick, so compiling drivers.
It shows up in lsusb as: 
`Bus 001 Device 003: ID 7392:a812 Edimax Technology Co., Ltd `

### What did you expect to happen?
Compilation succeeded.

### What actually happened?
Compiation terminated/failed

```
./arch/arm/include/asm/irq.h:7:23: fatal error: mach/irqs.h: No such file or directory
 #include <mach/irqs.h>
```
### Full console printout
The driver zip file expanded into a crazy long file path. Not so nice for legibility.
```
robot@ev3dev-2:~/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.2016
0506_BTCOEX20150921-58$ make
/bin/sh: 1: bc: not found
make ARCH=armv5tejl CROSS_COMPILE= -C /lib/modules/4.4.47-19-ev3dev-ev3/build M=/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58  modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.47-19-ev3dev-ev3'
Makefile:617: arch/armv5tejl/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv5tejl/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.47-19-ev3dev-ev3'
Makefile:1625: recipe for target 'modules' failed
make: *** [modules] Error 2
robot@ev3dev-2:~/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.2016
0506_BTCOEX20150921-58$ cd /usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch/
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch$ ls
Kconfig  arc  arm64  blackfin  cris  h8300    ia64  m68k   microblaze  mn10300  openrisc  powerpc  score  sparc  um         x86
alpha    arm  avr32  c6x       frv   hexagon  m32r  metag  mips        nios2    parisc    s390     sh     tile   unicore32  xtensa
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch$ cd arm
arm/   arm64/ 
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch$ cd arm
arm/   arm64/ 
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch$ cd arm
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch/arm$ ls
Kconfig        kernel        mach-clps711x   mach-footbridge  mach-ixp4xx    mach-mv78xx0  mach-picoxcell  mach-s5pv210   mach-u300       mm            plat-versatile
Kconfig-nommu  kvm           mach-cns3xxx    mach-gemini      mach-keystone  mach-mvebu    mach-prima2     mach-sa1100    mach-uniphier   net           probes
Kconfig.debug  lib           mach-davinci    mach-highbank    mach-ks8695    mach-mxs      mach-pxa        mach-shmobile  mach-ux500      nwfpe         tools
Makefile       mach-alpine   mach-digicolor  mach-hisi        mach-lpc18xx   mach-netx     mach-qcom       mach-socfpga   mach-versatile  oprofile      vdso
boot           mach-asm9260  mach-dove       mach-imx         mach-lpc32xx   mach-nomadik  mach-realview   mach-spear     mach-vexpress   plat-iop      vfp
common         mach-at91     mach-ebsa110    mach-integrator  mach-mediatek  mach-nspire   mach-rockchip   mach-sti       mach-vt8500     plat-omap     xen
crypto         mach-axxia    mach-efm32      mach-iop13xx     mach-meson     mach-omap1    mach-rpc        mach-stm32     mach-w90x900    plat-orion
firmware       mach-bcm      mach-ep93xx     mach-iop32x      mach-mmp       mach-omap2    mach-s3c24xx    mach-sunxi     mach-zx         plat-pxa
include        mach-berlin   mach-exynos     mach-iop33x      mach-moxart    mach-orion5x  mach-s3c64xx    mach-tegra     mach-zynq       plat-samsung
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch/arm$ make clean
make: *** No rule to make target 'clean'.  Stop.
robot@ev3dev-2:/usr/src/linux-headers-4.4.47-19-ev3dev-ev3/arch/arm$ cd /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58
robot@ev3dev-2:~/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.2016
0506_BTCOEX20150921-58$ ls
Kconfig  Makefile  clean  core  hal  ifcfg-wlan0  include  os_dep  platform  runwpa  wlan0dhcp
robot@ev3dev-2:~/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.2016
0506_BTCOEX20150921-58$ nano Makefile 
robot@ev3dev-2:~/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.2016
0506_BTCOEX20150921-58$ make
/bin/sh: 1: bc: not found
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.4.47-19-ev3dev-ev3/build M=/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58  modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.47-19-ev3dev-ev3'
/bin/sh: 1: bc: not found
  CC [M]  /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.o
In file included from ./arch/arm/include/asm/hardirq.h:6:0,
                 from include/linux/hardirq.h:8,
                 from include/linux/interrupt.h:12,
                 from include/net/flowcache.h:4,
                 from include/net/netns/xfrm.h:9,
                 from include/net/net_namespace.h:28,
                 from include/linux/netdevice.h:43,
                 from /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/include/osdep_service_linux.h:35,
                 from /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/include/osdep_service.h:41,
                 from /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/include/drv_types.h:32,
                 from /home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.c:22:
./arch/arm/include/asm/irq.h:7:23: fatal error: mach/irqs.h: No such file or directory
 #include <mach/irqs.h>
                       ^
compilation terminated.

```
and then
```
scripts/Makefile.build:258: recipe for target '/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.o' failed
make[2]: *** [/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.o] Error 1
Makefile:1402: recipe for target '_module_/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58' failed
make[1]: *** [_module_/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.4.47-19-ev3dev-ev3'
Makefile:1626: recipe for target 'modules' failed
make: *** [modules] Error 2
```

### What hardware and software are you using (including version numbers)?
Image file:         ev3dev-jessie-ev3-generic-2017-02-11
Kernel version:     4.4.47-19-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

antonvh, 2017-12-31:
It's got something to with CONFIG_SPARSE_IRQ but I can't figure out how to fix it.

dlech, 2017-12-31:
It's not clear what exactly you have tried. I gather that you are trying to compile on the EV3 itself using the drivers downloaded from Realtek. It is not clear if you have installed the header files for the Linux kernel (e.g. `sudo apt install linux-headers-ev3dev-ev3`).  This will have the missing `mach/irqs.h` file. You might need to add `arch/arm/mach-davinci/include` to your include paths. Also, you may have better luck with https://github.com/abperiasamy/rtl8812AU_8821AU_linux.

I'm not crazy about the idea of including in the kernel for just one person. But it is possible. See https://github.com/ev3dev/rtl8188eu for an example.

antonvh, 2018-01-01:
I put this in my .bashrc is that correct?
```
# Some stuff to compile drivers
export CPATH=arch/arm/mach-davinci/include
```

There is no /include in that folder? Is that possible? Should I look somewhere else?
```
/usr/src/linux-headers-4.4.87-22-ev3dev-ev3/arch/arm/mach-davinci$ ls
Kconfig  Makefile
```

Furthermore inside `/usr/src/linux-headers-4.4.87-22-ev3dev-ev3/arch/arm/include/mach` there is no irqs.h. It seems to be asked by `irq.h` in that same directory in the statement:
```
#ifndef CONFIG_SPARSE_IRQ
#include <mach/irqs.h>
#else
#define NR_IRQS NR_IRQS_LEGACY
#endif
```

I didn't know where to set CONFIG_SPARSE_IRQ so I forced `#define NR_IRQS NR_IRQS_LEGACY`

Then the compilation crashed a bit further on a bracket missing in a script. :/

Should I give up and go with 2.4Ghz wifi?

antonvh, 2018-01-01:
I've come a bit further. Now I'm running into this:
```
scripts/basic/fixdep: 1: scripts/basic/fixdep: Syntax error: "(" unexpected
scripts/Makefile.build:258: recipe for target '/home/robot/EW-7811UTC_Linux_Driver_4.3.19.5/RTL8821AU_Linux_v4.3.19.5_17672_BTCOEX20150921-58.20160506_support_kernel4.4/driver/rtl8821AU_linux_v4.3.19.5_17672.20160506_BTCOEX20150921-58/core/rtw_cmd.o' failed
```

and it seems to be about this file, which is part of the ev3dev headers. What to do now?
/usr/src/linux-headers-4.4.87-22-ev3dev-ev3/scripts/basic/fixdep

It is a binary, can't see much inside.

antonvh, 2018-01-01:
Getting even closer! This helps a lot:
`robot@ev3dev-2:/usr/src/linux-headers-4.4.87-22-ev3dev-ev3$ sudo make modules_prepare`
It crashes halfway, but builds enough to get the driver to compile.

dlech, 2018-01-01:
> I put this in my .bashrc is that correct?

I wouldn't put in in `.bashrc`. Just define it when you are compiling.

> I didn't know where to set CONFIG_SPARSE_IRQ

The EV3 kernel doesn't use sparse IRQ, so defining it will break things.

---

I'm starting to think that the best thing to do would be to cross-compile the kernel on your computer and then point the wifi driver to that kernel source tree to compile the driver.

antonvh, 2018-01-01:
I was afraid you'd say that. :(
Now I have to make a virtual machine and do the whole thing. And my hd is full.

dlech, 2018-01-01:
If you have a Raspberry Pi, you could do it there (without a cross compiler) and it would be faster than the EV3.

antonvh, 2018-01-01:
I have a Pi2 in a brickpi. I will try. First put an ev3dev stretch image on the pi2 I suppose?

dlech, 2018-01-01:
You don't have to. Just install `git` and `build-essential`.

antonvh, 2018-01-01:
Argh. No luck. Downloaded the latest rpi2-ev3dev-stretch image from the nightly builds and now the rpi flashes 7 times : kernel.img not found. Downloading the latest release now.

WasabiFan, 2018-01-01:
> Now I have to make a virtual machine and do the whole thing. And my hd is full.

Any chance your PC is running Windows (10)?
https://github.com/ev3dev/ev3dev-buildscripts#system-requirements and https://github.com/ev3dev/ev3dev/issues/969

antonvh, 2018-01-01:
yeah. it is. a good time to try microsofts shiny new linux toy. A new hd to contain debian images should arrive tomorrow.

WasabiFan, 2018-01-01:
If you try out those instructions, pull requests with fixes/clarifications would be appreciated 😀 I think there were some issues noted in #969 that we never integrated back into the real guide.

antonvh, 2018-01-02:
> If you have a Raspberry Pi, you could do it there (without a cross compiler) and it would be faster than the EV3.

The Pi is giving me the exact same errors as the Ev3 brick. But faster. I will try a generic stretch image for the pi, not an ev3dev.

dlech, 2018-01-02:
Give this a try. You will need to upgrade your kernel. BTW, I used the github repo I linked and it compiled without error (well, one warning that was treated as an error).

[rtl8812au-modules-4.4.87-22-ev3dev-ev3_1+1_armel.deb.zip](https://github.com/ev3dev/ev3dev/files/1598368/rtl8812au-modules-4.4.87-22-ev3dev-ev3_1.1_armel.deb.zip)

antonvh, 2018-01-02:
but you cross compiled on the pc? Not on a Pi or Ev3 I suppose?

It doesn't work. Alas.

Did you compile the rtl8821au too in that same file? Or is rtl8812 only?

I did `sudo dpkg -i` to install it

I'm at kernel 4.9 - latest stretch image. 
```
robot@ev3dev:~$ uname -r
4.9.58-ev3dev-1.6.0-ev3
```

dlech, 2018-01-02:
*sigh*. In your original post, you said you were using jessie.

dlech, 2018-01-02:
Here is a driver compiled for the 4.9.58-ev3dev-1.6.0-ev3 kernel.

[rtl8812au-modules-4.9.58-ev3dev-1.6.0-ev3_1+1_armel.deb.zip](https://github.com/ev3dev/ev3dev/files/1598610/rtl8812au-modules-4.9.58-ev3dev-1.6.0-ev3_1.1_armel.deb.zip)

antonvh, 2018-01-02:
OMGOMGOMG It works!!!! Brilliant.

(I'm really very sorry about that kernel version.)

Now I'm going to finish the installation for LEGO World CPH. And order 8 more of those sticks.

Then I will make a tutorial on how to install linux on win 10 and compile kernels, like @WasabiFan asked.

closed 2018-01-02 by antonvh

antonvh, 2018-01-02:
OK Will do. I need to try that docker thing to get my image.

antonvh, 2018-01-07:
I can also confirm the driver works on the Jessie release. With one caveat: it breaks bluetooth. Maybe because of a brown-out.

antonvh, 2018-01-17:
I would like to extend the tutorial on cross compiling. (http://www.ev3dev.org/docs/tutorials/using-docker-to-cross-compile/) because I can't get it to work for kernel modules.
@dlech did you use the above system to compile the driver you posted? Or did you cross compile on a Debian or Ubuntu system? Or did you compile on one of the ev3dev docker images for brickstrapping?
Ultimately I'd like to document a complete cross-platform docker-based workflow. WSL has too many limitations. It doesn't even run docker properly.

dlech, 2018-01-17:
For this particular case, I used <https://github.com/ev3dev/ev3dev-kpkg> (cross-compiling on an Ubuntu system). It is what I use for official kernel releases and it builds kernel module packages at the same time that it builds the kernel package. I just had to add a debian directory with the packaging information to the wifi repository.

I've never used Docker for kernel stuff, only userspace applications. But if you want to write up something about using Docker for cross-compiling in general, please submit it to <https://github.com/ev3dev/edu-docs>. I'm trying to move away from using the main website for docs so that we can have versioned docs (e.g. jessie vs. stretch).

dlech, 2021-03-09:
This driver is already included in ev3dev-stretch.