ev3dev/ev3dev #1022: Extreme slowness under Stretch

ndward, 2017-12-31:
I mentioned my problem in issue #1014 but @WasabiFan concludes that my problem cannot be related to brickd so therefore I am starting a new thread. Repeating and updating my message on the other thread:
I love the VS Code EV3 python workflow but have one huge problem: scripts take an incredibly long time to launch: between 20s and more than two minutes. Here's an example of a script that takes **more than two minutes** before showing text on the LCD, even though it has only 14 lines:

```
#!/usr/bin/env python3
from ev3dev.ev3 import *
from PIL import Image, ImageDraw, ImageFont
import ev3dev.fonts as fonts
from time import sleep

lcd = Screen()

# Connect ultrasonic sensor to any sensor port
# and check it is connected.
us = UltrasonicSensor() 
assert us.connected, "Connect a US sensor to any sensor port"
# Put the US sensor into distance mode.
us.mode = 'US-DIST-CM'

while True:    # forever
    lcd.clear()
    lcd.draw.text((45, 40), str(us.value()/10)+' cm', font=fonts.load('helvB24')) # convert to cm
    lcd.update()
    sleep(0.1) # so the display doesn't change too frequently
```

After restarting the brick the start time for this script went from two minutes to one minute, but that's still very long. The scripts that take forever to launch are the ones that import PIL, fonts etc, which I assume to be relatively large modules. The slowness is the same whether I launch the script from VS Code or from Brickman. I'm using Win10 and thought my PC might be the cause of the problem but trying on another Win10 machine I got the same result. I also got the same result after flashing another SD card with the December 13 Stretch image, so it cannot be a problem with the card. Using the alternate card, I tested the script without using VS Code at all and still had the one minute start time (until text appeared on the LCD). So I supposed either my EV3 is defective or else there  is a problem with the Stretch setup.

I then tried a card with a **Jessie** image and ran the same script – it took 28s to start. Not using the large font (i.e. not importing PIL or fonts) the script started in 15s. Using the large font but not importing PIL, which does not seem to be needed, the script started in 20s. All these start times with Jessie seem reasonable to me, much much faster than with Stretch, so I'm inclined to think that my brick's hardware is okay. By the way, I see the slowness even when simply scrolling through the Brickman file browser - the browser is very slow to react compared to the Jessie version. It's worth mentioning that, as detailed in issue #1014, the CPU seems to be working at 100% while the script is starting.

I'm currently at a loss to explain this slowness -  I seem to have eliminated every possible cause except a problem with the Stretch setup, but in that case many other people would be reporting this problem of slowness, which does not seem to be the case.

PS Regarding the line `import ev3dev.fonts as fonts`, is it really necessary to import all the fonts? There are so many. Importing all the fonts seems to take several seconds. Shouldn't it be possible to import just the font(s) needed by the script? What would that line look like?

========== ev3dev-sysinfo ==========
Image file: ev3dev-stretch-ev3-generic-2017-12-13
Kernel version: 4.9.58-ev3dev-1.6.0-ev3
Brickman: 0.10.0
ev3devKit: 0.5.2
Board: board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=00165340377E
BOARD_INFO_TYPE=main

antonvh, 2018-01-06:
I have the same slowness. Even without using VS Code. A simple `git pull` takes over two minutes.

There is a process that takes quite a lot of processing time. I have no idea what it is:
   73 root     -51   0       0      0      0 S 16.6  0.0  10:40.26 irq/245-ti-ads7

I can't kill it.

dlech, 2018-01-06:
That is the analog digital converter driver in the kernel. There are some major changes in it between ev3dev-jessie and ev3dev-stretch, so I guess it needs some work.

antonvh, 2018-01-06:
What can I do to (help) solve this? Where do I start looking?

ndward, 2018-01-07:
I'm reassured that I'm not the only one experiencing this problem but that's still only two of us. Quite a number of people must have tried the Stretch image and no one else has reported this problem so presumably most people are experiencing normal, rapid responses?

dlech, 2018-01-07:
> What can I do to (help) solve this? Where do I start looking?

In the ev3dev 4.9 kernel, we switched to the mainline kernels iio susbsystem for the ADC. I think it does a bunch of memcpys to multiplex the data to the various drivers (input ports, output ports, power supply) that wasn't done before. You could search the IIO mailing list to see if there have been any improvements to this between 4.9 and now. 

https://marc.info/?l=linux-iio

Some keywords to look for: "callback buffer", `iio_channel_get_all_cb`, "demux"

Or, maybe there is a way we can use DMA to help offload some from the processor.

Related files:
* https://github.com/ev3dev/ev3-kernel/blob/ev3dev-stretch/drivers/iio/industrialio-buffer.c
* https://github.com/ev3dev/ev3-kernel/tree/ev3dev-stretch/drivers/iio/buffer
* https://github.com/ev3dev/lego-linux-drivers/tree/ev3dev-stretch/linux/iio

ndward, 2018-03-29:
I'm sorry to note that the March 16 release of the EV3 Stretch OS had not fixed this problem of slowness. The 14 line script at the top of this page still takes up to 2m48s to start on my EV3 i.e. to begin displaying values. I at last managed to persuade someone, @QuirkyCort in Singapore, to kindly try running the script and he tells me that using the 16 March version of the Stretch OS it took about 35s to start, why I still find to be unreasonably long but which is much faster than on my system. Note that my very slow times are the same whether I launch the script from VS Code or from Brickman. So I am tempted to conclude that my EV3 has a hardware problem and that I need to buy a new EV3. However I have been able to determine that **a large part of the slowness comes from the process brickd becoming 'hyperactive' and taking over much of the CPU**. That means that this issue overlaps with https://github.com/ev3dev/ev3dev/issues/1014 so I continue this discussion on that page.

dlech, 2018-04-08:
Relevant comment: https://github.com/ev3dev/ev3dev/issues/1020#issuecomment-379334086

ndward, 2018-04-09:
This is a snapshot of 'top' after made while the EV3 sits idle after rebooting. Note the absence of a hyperactive brickd process:
[image]
I don't understand how the load average can be 2.68 (meaning that the system is overloaded by 168% more work than it can handle) when the CPU% figures in the lower section only add up to about 39.4%. Without 'top' running, the figure would be just 32.3%. How can the CPU be overloaded if it is less than 40% busy?

ddemidov, 2018-04-09:
See https://prutser.wordpress.com/2012/05/05/understanding-linux-load-average-part-2/

dlech, 2018-04-09:
Yes, EV3 is single core.

dlech, 2018-04-15:
I released a new kernel for ev3dev-stretch `4.14.34-ev3dev-1.6.0-ev3`. Please try this and see if you are still getting high load.

What I am seeing in `top` is comparable to ev3dev-jessie.

```
top - 04:46:11 up 1 day, 28 min,  2 users,  load average: 0.08, 0.07, 0.08
Tasks:  65 total,   1 running,  43 sleeping,   0 stopped,   0 zombie
%Cpu(s):  5.2 us,  5.6 sy,  0.0 ni, 89.2 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem :    57164 total,     2144 free,    21880 used,    33140 buff/cache
KiB Swap:    98300 total,    91388 free,     6912 used.    31348 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND      
   69 root     -51   0       0      0      0 S 13.0  0.0 187:37.18 irq/245-ti-+ 
 4328 robot     20   0    8480   3060   2620 R  5.0  5.4   5:33.49 top          
 4739 root      20   0       0      0      0 I  4.3  0.0   1:05.93 kworker/0:2  
  438 root      20   0   70152   5012   3764 S  1.2  8.8  22:42.56 brickman     
   52 root      20   0       0      0      0 S  0.6  0.0   7:17.38 spi0         
  369 mosquit+  20   0    6588   1888   1828 S  0.3  3.3   4:41.03 mosquitto    
    1 root      20   0   28072   4704   3928 S  0.0  8.2   0:20.33 systemd      
    2 root      20   0       0      0      0 S  0.0  0.0   0:00.07 kthreadd     
    4 root       0 -20       0      0      0 I  0.0  0.0   0:00.00 kworker/0:0H 
    6 root       0 -20       0      0      0 I  0.0  0.0   0:00.00 mm_percpu_wq 
    7 root      20   0       0      0      0 S  0.0  0.0   0:04.20 ksoftirqd/0  
    8 root      20   0       0      0      0 I  0.0  0.0   0:39.04 rcu_preempt  
    9 root      20   0       0      0      0 I  0.0  0.0   0:00.01 rcu_sched    
   10 root      20   0       0      0      0 I  0.0  0.0   0:00.00 rcu_bh       
   11 root      20   0       0      0      0 S  0.0  0.0   0:00.03 kdevtmpfs    
   12 root       0 -20       0      0      0 I  0.0  0.0   0:00.00 netns        
   14 root      20   0       0      0      0 S  0.0  0.0   0:00.00 oom_reaper
```

ndward, 2018-04-15:
After installing a card with the new image I found that the brickpi problem was present as before. After then running `sudo apt update` and `sudo apt upgrade` **I was no longer able to put brickd into hyperactive mode. I think this is a big step forwards – well done, @dlech!

BUT I think a new problem has arisen: update() for the LCD seems to have stopped working.** For example, this script:
```
#!/usr/bin/env python3
from ev3dev.ev3 import *
from time import sleep

lcd = Screen()
lcd.draw.text((48,13),'Hello, world.')
lcd.update()
sleep(6)
```

returns this error:

```
Traceback (most recent call last):
  File "/home/robot/mine/printtext.py", line 7, in <module>
    lcd.update()
  File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 3325, in update
    raise Exception("Not supported")
Exception: Not supported
```

I don't know if this is related to brickd.

The starting times seem to be the same as they were previously before brickd hyperactivity kicked in – up to a minute or more in some cases.

I don't get load average figures as low as yours but they are much lower than prior to the new image – as I write this the EV3 is idling with load averages reported by 'top' as 0.12, 0.36, 0.48.

In order to run `sudo apt update` and `sudo apt upgrade` I needed the EV3 to be online, a state which I have not been able to achieve for many weeks. There was something wrong with Internet Connection Sharing on my Win64 machine – trying to turn it on caused bad failures, forcing me to restart the PC. Pages like this one https://serverfault.com/questions/884251/internet-connection-sharing-after-windows-10-fall-creators-update-hangs-on-star indicate that others are having similar problems following a Windows update late last year. One person suggested disabling and resetting the Windows Defender firewall and that seemed to help – no more crashes. I had other problems too with Internet Connection Sharing – I wonder whether other Windows-based users of ev3dev have had similar problems.

**Another thing I've noticed is that although my USB connection to the EV3 has always been very stable I have lost it several times this evening – I just saw it fail a few seconds ago for no apparent reason (when I was doing nothing), for example. The connection via USB doesn't seem to last more than 10 or 20 minutes.** I haven't tried wireless connections with the new image.

**To summarise, the brickd problem seems to be history but I believe two new problems have appeared: the failure of update() for the LCD and an instability even in USB connections.**

Two quick questions: if I have multiple EV3s can I move ev3dev cards between them or is each card tied to a specific EV3?

Instead of doing` import ev3dev.fonts as fonts` I often want to import just one font (helvB24, for example). How do I do that?

dlech, 2018-04-15:
The updates include a major kernel upgrade from 4.9 to 4.14, so I suspect there will be multiple new issues related to this. Please open new issues for any problems that you find.

> BUT I think a new problem has arisen: update() for the LCD seems to have stopped working.

The 4.14.x kernel uses a new display driver. ev3dev-lang-python will need to be updated accordingly.

> The connection via USB doesn't seem to last more than 10 or 20 minutes.

This is also probably related to the kernel upgrade that is now using a DMA controller that was not previously used. We might have to turn that off or something.

> Instead of doing import ev3dev.fonts as fonts I often want to import just one font (helvB24, for example). How do I do that?

not sure. @ev3dev/python?

ddemidov, 2018-04-16:
> Instead of doing import ev3dev.fonts as fonts I often want to import just one font (helvB24, for example). How do I do that?

[Importing `ev3def.fonts`](https://github.com/ev3dev/ev3dev-lang-python/blob/jessie/ev3dev/fonts/__init__.py) does not actually import any fonts. A font is only loaded when you call [ev3dev.fonts.load()](https://github.com/ev3dev/ev3dev-lang-python/blob/jessie/ev3dev/fonts/__init__.py#L16) with an appropriate font name.

ndward, 2018-04-16:
It takes 54s for this script to run under stretch 
```
#!/usr/bin/env python3
import ev3dev.fonts as fonts
```
and 9 seconds for this to run

`#!/usr/bin/env python3`

If `import ev3dev.fonts as fonts` isn't importing fonts then what is it doing for those extra 45s?

ndward, 2018-04-16:
I tried importing all that and it took 52s so it looks like you're right. Specifically (I tried them one by one):
```
import pkg_resources #49s
import os.path # 9s
from PIL import ImageFont # 12s
```
Of course these times don't add up to 52s since there's other stuff going on (download, start interpreter). If we deduct 9s from each of those for the download+interpreter we get 40+0+3=43 to compare with the 52-9 = 43 when they are imported together - the numbers make sense. So it's the import pkg_resources that takes nearly all the time (about 40s). I don't know what that is. Is there any way it could be lightened?

ddemidov, 2018-04-16:
`pkg_resources` is used to get file names for the font files packaged with ev3dev: http://setuptools.readthedocs.io/en/latest/pkg_resources.html.
We only use [`resource_listdir`](https://github.com/ev3dev/ev3dev-lang-python/blob/jessie/ev3dev/fonts/__init__.py#L10) and [`resource_filename`](https://github.com/ev3dev/ev3dev-lang-python/blob/jessie/ev3dev/fonts/__init__.py#L23-L24), so instead of `import pkg_resources` we could `from pkg_resources import package_listdir, package_filename`. I doubt it will change the import time, but you could measure if it makes any difference.

ddemidov, 2018-04-16:
Also see https://github.com/pypa/setuptools/issues/926

ddemidov, 2018-04-16:
Moving the discussion to https://github.com/ev3dev/ev3dev-lang-python/issues/456

dlech, 2018-04-22:
Closing this issue since it seems the high CPU usage ("load average:" in `top` > 1) is fixed. And there are other issues open for the other problems that were discussed.

closed 2018-04-22 by dlech