ev3dev/ev3dev #1023: Schedule to replace Jessie as the main Debian distro

jabrena, 2017-12-31:
Hi,

I would like to know when `EV3Dev` is going to recommend `Stretch` as the main Debian version.

At the moment, the default distro is `Jessie`:
http://www.ev3dev.org/downloads/

But exist some changes from `Jessie` to `Stretch` and I would like to develop a `Stretch`  branch for the project. 

Merry Christmas and happy new year

Juan Antonio

dlech, 2017-12-31:
I guess the answer is never unless someone comes up with a plan. I haven't been particularly motivated to work on it much lately.

jabrena, 2018-01-02:
Happy new year @dlech,

If you observe, in last month some utilities was added to Stretch so the migration to this distro is necessary. Besides, I am not sure if exist an issue with the list of changes at `/sys/class` level to review in the library. In my case, I would like to offer support for `Jessie` and `Stretch`

Cheers

Juan Antonio

antonvh, 2018-01-06:
What utilities do you mean?

jabrena, 2018-01-10:
Hi @antonvh,

the changes that I have in mind are:

At OS level: Strech will improve in some technical aspect jessie distro.

At EV3Dev level, I have the following points to review:

+ The usage of brickrun
+ The new library for Graphics: https://github.com/ev3dev/grx
+ The utility for battery monitor

I suppose that exist more points but this is a starting list of issues to review.

I think that we should create a plan for every teams and apply in a gradual way.

Juan Antonio

antonvh, 2018-01-10:
Ah ok. There is still the issue #1022 to resolve. Until someone has time to fix it, stretch is not realy usable. I reverted to jessie for my LWCPH project. Doesn't grx run on jessie too?

jabrena, 2018-01-10:
> Doesn't grx run on jessie too?

It is a good question.

@dlech is it possible to use on Jessie?

In my case, I would like to use GRX to get control of the Display for all devices (EV3 & PiStorms)

Juan Antonio

dlech, 2018-01-11:
There is an older version of GRX on Jessie, but it is not API compatible with the Stretch version.

closed 2020-08-17 by dlech