ev3dev/ev3dev #1024: Daisy chain 8 EV3s using USB networking

markcrosbie, 2018-01-02:
### What are you trying to do?
For a large project I need to network 8 EV3s together. I am trying to use USB networking, with each EV3 connecting to the next one in the chain, and a 'master' EV3 connected to the head of the chain  with a wifi connection so I can connect via my laptop. Ideally I'd like static IP addresses on each EV3. They don't need internet access. 

Pictorially:
laptop <- wifi -> EV1 (master) <- usb -> EV2 <- usb -> EV3 etc.

A USB cable runs from the top port on EV1 to the side port on EV2, then from the top port of EV2 to the side port on EV3 and so on. 

I'm having trouble configuring the EV3s. I've read as much as I can on tethering, but can't figure out if all 8 EV3s need to have it enabled, or only the 'head' EV3. Each EV3 has multiple Wired interfaces appear in the 'All Network Interfaces' menu. There's a Wired entry with a USB symbol beside it, and another Wired entry with a small 'block' symbol beside it (best way I can describe it). 

What would be the best way to create this network?

### What actually happened?
I can't get EV2 onwards to assign IP addresses to EV3 etc.

### What hardware and software are you using (including version numbers)?
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-stretch-ev3-generic-2017-10-24
Kernel version:     4.9.49-ev3dev-1.5.1-ev3
Brickman:           0.10.0
ev3devKit:          0.5.2
Board:              board0
BOARD_INFO_HW_REV=7
BOARD_INFO_MODEL=LEGO MINDSTORMS EV3
BOARD_INFO_ROM_REV=6
BOARD_INFO_SERIAL_NUM=0016534EC713
BOARD_INFO_TYPE=main
```

dlech, 2018-01-02:
Hmm... tethering would be nice, but you don't have control over the IP address of the `tether` interface, so that could be a problem if you want to connect to devices down the chain. If you only want to go up the chain though, this could work. For example, if each EV3 acts as a client only and connects to a server farther up the chain, then this will work. If you want peer-to-peer connections, then this won't work.

If we can't get tethering to work, then you will have to manually configure a network bridge on each EV3. I'll have to play with it a little bit and see if I can figure something out.

markcrosbie, 2018-01-02:
Thanks for the quick reply @dlech ! Ideally I'd like each EV3 to have full network connectivity to all other EV3s, but as a fall-back it's sufficient for the master to have connectivity down-chain to all the slaves. My plan is to run a RPyC service on each EV3 down-chain and have the master EV3 connect to it poll for status, send commands etc. Each EV3 is also running code locally to do various tasks. 

I was wondering if an 8-port USB hub would be easier - to connect 7 EV3s to the master?

Failing that I'll go the 8x wifi adapter route. It's just a bit pricey.

Rosstb, 2018-01-05:
> Hmm... tethering would be nice, but you don't have control over the IP address of the tether interface, so that could be a problem if you want to connect to devices down the chain. If you only want [...]

@dlech Is it possible to control a "slave" EV3 with a "master" EV3? As in, the "master" runs all or most of the code, and the "slave" is basically a multiplexer that adds more motor and sensor ports? I am not sure if this is any different from what @markcrosbie  was talking about, but it would be nice to know for sure.

dlech, 2018-01-05:
Sure, you can do this using RPyC. You can make the "slave" do as little or as much processing as you want.

markcrosbie, 2018-01-05:
I’ve just coded this up in RPyC - very simple services interface with a registry for ‘slave’ discovery makes for quick master-slave coding.

Brisbane-QLD, 2018-11-24:
> I’ve just coded this up in RPyC - very simple services interface with a registry for ‘slave’ discovery makes for quick master-slave coding.
> […](#)

I’m currently migrating from EV3-G to ev3dev Python due to daisy-chain bugs with unsatisfactory workarounds.  My background is Linux software development, not system administration or networking. Currently, I only need to daisy-chain 4 bricks but I have future projects which will require many more EV3s.
My test environment has 2 bricks, both with ev3dev cards and RPyC installed; RPyC is not currently auto-launched on start-up/boot. I have a new 4 port USB hub, and I do NOT need to execute programs from my Windows 10 PC.
If you could please provide some instructions for configuring the environment and a simple remote motor execution example then I should be able to extrapolate everything from that information.

dwalton76, 2018-11-24:
Some discussion about daisy chaining and python happening in:
- https://github.com/ev3dev/ev3dev-lang-python/issues/385
- https://github.com/ev3dev/ev3dev-lang-python/issues/570

Brisbane-QLD, 2018-11-24:
> Some discussion about daisy chaining and python happening in:
> 
> * [ev3dev/ev3dev-lang-python#385](https://github.com/ev3dev/ev3dev-lang-python/issues/385)
> * [ev3dev/ev3dev-lang-python#570](https://github.com/ev3dev/ev3dev-lang-python/issues/570)

Thanks but not very helpful. I was seeking some examples.

I'll re-read the first link which is mainly a discussion which points here and to the second link.
The second link has nothing other than pointing here and to the first link.

dwalton76, 2018-11-24:
> Thanks but not very helpful. I was seeking some examples.

The python library does not support daisy chaining yet so there are not going to be examples to be found.  There is some interest in daisy chaining from yourself and over on [eurobricks](https://www.eurobricks.com/forum/index.php?/forums/topic/168586-ev3dev-venturing-into-the-world-of-ev3dev-and-python/) so maybe it gets enough interest for someone to pick up the two issues linked above and implement them.

markcrosbie, 2018-11-25:
@Brisbane-QLD in the end I used wifi dongles in each EV3 and ran RPyC over wifi rather than try to get USB daisy chaining to work. One EV3 was the 'master' as it held the service registry, and the rest ran slaves. It worked!

Brisbane-QLD, 2018-12-03:
@markcrosbie I have managed to create a working USB solution. The main issue was IP address visibility so migrating to WiFi shouldn't be difficult.

jabrena, 2018-12-07:
Hi,

In my case, I using a Brick as master with tethering enabled and using a USB Hub to connect multiple bricks, 2 in this case:

Pictorially:
```
laptop <- ethernet -> EV1 (master) <- usb -> EV2 
                                                            <- usb -> EV3
```

Both slaves bricks, take a IP, but when I try to reach the second brick from master, it is not able to reach that brick.

```
robot@ev3dev:~$ lsusb
Bus 001 Device 046: ID 0bda:8152 Realtek Semiconductor Corp. 
Bus 001 Device 053: ID 0694:0005 Lego Group 
Bus 001 Device 050: ID 0694:0005 Lego Group 
Bus 001 Device 027: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
robot@ev3dev:~$ 
``` 

In the following output, it is possible to see how the second brick is not reacheable:

```
robot@ev3dev:~$ lsusb
Bus 001 Device 046: ID 0bda:8152 Realtek Semiconductor Corp. 
Bus 001 Device 053: ID 0694:0005 Lego Group 
Bus 001 Device 050: ID 0694:0005 Lego Group 
Bus 001 Device 027: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
robot@ev3dev:~$ ping 169.254.45.75
PING 169.254.45.75 (169.254.45.75) 56(84) bytes of data.
From 169.254.182.205 icmp_seq=1 Destination Host Unreachable
From 169.254.182.205 icmp_seq=2 Destination Host Unreachable
From 169.254.182.205 icmp_seq=3 Destination Host Unreachable
^C
--- 169.254.45.75 ping statistics ---
6 packets transmitted, 0 received, +3 errors, 100% packet loss, time 5231ms
pipe 4
robot@ev3dev:~$ ping 169.254.110.178
PING 169.254.110.178 (169.254.110.178) 56(84) bytes of data.
64 bytes from 169.254.110.178: icmp_seq=1 ttl=64 time=3.19 ms
64 bytes from 169.254.110.178: icmp_seq=2 ttl=64 time=1.63 ms
64 bytes from 169.254.110.178: icmp_seq=3 ttl=64 time=2.21 ms
^C
--- 169.254.110.178 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 1.632/2.348/3.195/0.645 ms
```

Why, it is not reachable?

Juan Antonio

Brisbane-QLD, 2018-12-08:
@jabrena I had the same problem with a USB hub. When I had USB working it was because the USBs were on different networks. I migrated to Wifi because I thought it might be useful in the future, and it was much easier. I just turn on the bricks in any order and they auto-boot rpyc. I need the notebook for the Wifi router so I'm running the programs wirelessly from the notebook.

<pre><code>laptop <- ethernet -> Wifi router <- Wifi -> Brick001
                                  <- Wifi -> Brick002
                                  <- Wifi -> Brick003</code></pre>

A compatible EdiMax USB Wifi adapter is only about US$8, and the EdiMax router I'm using is about US$20 so it's not very expensive.

jabrena, 2018-12-08:
Good morning @Brisbane-QLD, 

many thanks for the feedback. for 2 Bricks, `tethering` goes well and it is very comfortable but for 3 bricks or more, yesterday I didn't understand why it is not working. Indeed, I tried in another way without any success:

```
laptop <- ethernet -> EV1 (Master) <- usb -> EV2 (Slave of EV1, Master of EV3) 
                                                                             <- usb -> EV3
```