ev3dev/ev3dev #1025: Trouble Connecting to the Internet via USB and Bluetooth

1Peaches, 2018-01-03:
### What are you trying to do?
Ideally, I would be able to connect to the internet by using bluetooth through a Mac, but I also tried using a USB. 
I
### What did you expect to happen?
I expected to see the Status: Online but instead see the Status: Connected symbol

### What actually happened?
I instead saw the Status: Connected symbol. When using bluetooth, I changed the default settings to OSX, but wasn't able to connect to the ev3 at. For what its worth, the laptop and ev3 say I am connected via bluetooth. The USB is working fine until I try and install pip. I read the other issue regarding connecting to the internet, but was still unable to resolve the issue.

### What hardware and software are you using (including version numbers)?
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2018-01-03:
> I changed the default settings to OSX,

Don't change it. Put it back to automatic.

It sounds like you missed the Internet Sharing step.

[image]

1Peaches, 2018-01-03:
I switched it back and added the internet sharing. Thanks!! I'm now able to connect via ssh but when I type in wget google.com I get: 
Resolving google.com (google.com)... failed: Name or service not known.
wget: unable to resolve host address ‘google.com’ and I'm not able to do sudo apt-get update.

dlech, 2018-01-03:
You have to disconnect and reconnect after you enable Internet Sharing. If you are using Bluetooth, remove and re-pair the EV3 while Internet Sharing via Bluetooth is enabled.

1Peaches, 2018-01-03:
I removed and re-paired the EV3, but it still says connecting and won't connect to the internet.

1Peaches, 2018-01-03:
I couldn't figure out how to use the bluetooth, but I stole a wifi dongle from my brother's raspberry pi, and I'm now online. Thanks!! I'm glad I didn't mess up the default settings.

closed 2018-01-03 by 1Peaches