ev3dev/ev3dev #1026: Sound.speak does not work with double quotes

amandaoneal, 2018-01-11:
### What are you trying to do?

Call Sound.speak on a String with double quotes in it, such as:

ev3.Sound.speak('Hello " world').wait()

(Using rpyc to communicate between computer & EV3)

### What did you expect to happen?

It would speak "Hello world" in some form

### What actually happened?

ValueError: No closing quotation

Stack:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 305, in _dispatch_request
    res = self._HANDLERS[handler](self, *args)
  File "/usr/lib/python3/dist-packages/rpyc/core/protocol.py", line 535, in _handle_call
    return self._local_objects[oid](*args, **dict(kwargs))
  **File "/usr/lib/python3/dist-packages/ev3dev/core.py", line 3446, in speak
    espeak = Popen(shlex.split(cmd_line), stdout=PIPE)**
  File "/usr/lib/python3.4/shlex.py", line 276, in split
    return list(lex)
  File "/usr/lib/python3.4/shlex.py", line 266, in __next__
    token = self.get_token()
  File "/usr/lib/python3.4/shlex.py", line 93, in get_token
    raw = self.read_token()
  File "/usr/lib/python3.4/shlex.py", line 169, in read_token
    raise ValueError("No closing quotation")
ValueError: No closing quotation

### What hardware and software are you using (including version numbers)?

Hardware: EV3

Software:
**System info (from `ev3dev-sysinfo`)**
```
Image file:         ev3dev-jessie-ev3-generic-2017-09-14
Kernel version:     4.4.87-22-ev3dev-ev3
Board:              LEGO MINDSTORMS EV3 Programmable Brick
Revision:           0006
Brickman:           0.8.1
ev3devKit:          0.4.3

dlech, 2018-01-11:
cc: @ev3dev/python

WasabiFan, 2018-01-11:
:laughing: All the sound code is pretty brittle. Can you open an issue at https://github.com/ev3dev/ev3dev-lang-python to track this with the other Python library issues? Or, if you're feeling up for it, we welcome pull requests! It should be an easy fix.

amandaoneal, 2018-01-11:
OK, made an attempt: https://github.com/amandaoneal/ev3dev-lang-python/pull/1

WasabiFan, 2018-01-11:
It looks like it would probably work, at least for the case of the quotes. However, I'm thinking we could probably cover all our bases by using `shlex.quote`: https://docs.python.org/3.4/library/shlex.html#shlex.quote

Also, when opening the Pull Request, make sure you choose the option for opening it against the official ev3dev-lang-python repo rather than your fork of it (so the code goes to the official copy).

amandaoneal, 2018-01-12:
Whoops, newbie mistake :) I don't do much on GitHub. The master branch isn't using shlex anymore at all, so I assumed there was a reason to move away from it. I noticed that in the master version, the real problem is that they're formatting '/usr/bin/espeak --stdout {0} "{1}"', so the double quotes are actually the main problem, since {1} is inside "".

Attempt 2: https://github.com/ev3dev/ev3dev-lang-python/pull/443

closed 2018-01-12 by dlech