ev3dev/ev3dev #1027: Jessie Docker Cross Compiler Images missing ARM-enabled GDB

tcwan, 2018-01-18:
### What are you trying to do?
Debug compiled arm code remotely

### What did you expect to happen?
ARM-enabled gdb to be available

### What actually happened?
```
$ which gdb
/usr/bin/gdb
compiler@4bc36bc595b4:/src/ev3dev-c/eg/hello$ ls
Makefile  hello  hello.c  hello.o
compiler@4bc36bc595b4:/src/ev3dev-c/eg/hello$ gdb hello 
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello...done.
(gdb) target remote 192.168.2.2:3333
Remote debugging using 192.168.2.2:3333
warning: while parsing target description (at line 1): Target description specified unknown architecture "arm"
warning: Could not load XML target description; ignoring
warning: `/lib/ld-linux.so.3': Shared library architecture unknown is not compatible with target architecture i386.
warning: `/lib/ld-linux.so.3': Shared library architecture unknown is not compatible with target architecture i386.
```

### What hardware and software are you using (including version numbers)?
ev3dev-jessie-ev3-generic Docker image (ID: e519b937c466)

tcwan, 2018-01-18:
arm-linux-gnueabi-gdb is missing from ev3dev/debian-stretch-cross as well.

dlech, 2018-01-18:
I don't think debian ships a cross gdb with their cross compilers, so getting gdb in ev3dev/debian-{jessie,stretch}-cross is probably not an option.

Feel free to do your own research but I couldn't find one when I went looking.

tcwan, 2018-01-18:
Are you using one from your Host (PC) platform?
I don't know if the non linux-gnueabi GDB packages from MacPorts work with the executable format (Linux ELF ?)

There are:

arm-elf-gdb                     
arm-none-eabi-gdb 

packages available for MacPorts.

tcwan, 2018-01-18:
I'm not familiar with Debian, but I found this 
[GDB Multiarch](https://packages.debian.org/stretch/gdb-multiarch)

dlech, 2018-01-18:
Nice find. If it works, I'll add it to the docker images.

tcwan, 2018-01-18:
Seems to work, though I'm not very good with command line gdb (I rely too much on Eclipse).

E.g., how can I configure the debug symbols for the program (see backtrace output)? I though it is included in the file already when I compiled with `-g`.

```
compiler@8ed2ccadef26:/src/ev3dev-c/eg/hello$ gdb-multiarch hello
GNU gdb (Debian 7.12-6) 7.12.0.20161007-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from hello...done.
(gdb) target remote 192.168.2.2:3333
Remote debugging using 192.168.2.2:3333
Reading /lib/ld-linux.so.3 from remote target...
warning: File transfers from remote targets can be slow. Use "set sysroot" to access files locally instead.
Reading /lib/ld-linux.so.3 from remote target...
Reading symbols from target:/lib/ld-linux.so.3...Reading /lib/62b2761d7a64fc51fd22235cff9e0ce5fccaa3.debug from remote target...
Reading /lib/.debug/62b2761d7a64fc51fd22235cff9e0ce5fccaa3.debug from remote target...
(no debugging symbols found)...done.
0xb6fcea60 in ?? () from target:/lib/ld-linux.so.3
(gdb) info reg
r0             0x0	0
r1             0x0	0
r2             0x0	0
r3             0x0	0
r4             0x0	0
r5             0x0	0
r6             0x0	0
r7             0x0	0
r8             0x0	0
r9             0x0	0
r10            0x0	0
r11            0x0	0
r12            0x0	0
sp             0xbefff740	0xbefff740
lr             0x0	0
pc             0xb6fcea60	0xb6fcea60
cpsr           0x10	16
(gdb) bt
#0  0xb6fcea60 in ?? () from target:/lib/ld-linux.so.3
#1  0x00000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) stepi
0xb6fcea64 in ?? () from target:/lib/ld-linux.so.3
(gdb) info reg
r0             0x0	0
r1             0x0	0
r2             0x0	0
r3             0x0	0
r4             0x0	0
r5             0x0	0
r6             0x0	0
r7             0x0	0
r8             0x0	0
r9             0x0	0
r10            0x30504	197892
r11            0x0	0
r12            0x0	0
sp             0xbefff740	0xbefff740
lr             0x0	0
pc             0xb6fcea64	0xb6fcea64
cpsr           0x10	16
(gdb) c
Continuing.
Reading /lib/arm-linux-gnueabi/libc.so.6 from remote target...
Reading /lib/arm-linux-gnueabi/a1c44276e59f35ee612225f9a0579014fa8777.debug from remote target...
Reading /lib/arm-linux-gnueabi/.debug/a1c44276e59f35ee612225f9a0579014fa8777.debug from remote target...
[Inferior 1 (process 754) exited normally]
(gdb) 
```

tcwan, 2018-01-18:
On the Mac, `arm-elf-gdb` is too old and does not recognize the Linux ELF format.
In addition, I needed to install GDB as `sudo port install gdb +multiarch`. 

Nonetheless, gdb needs some configuration parameters in order to recognize the correct elf format.

```
$ ggdb hello
GNU gdb (GDB) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-apple-darwin17.3.0".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
"/Users/tcmac/gitrepo/EV3/src/ev3dev-c/eg/hello/hello": not in executable format: File format is ambiguous.
Matching formats: elf32-littlearm elf32-littlearm-symbian elf32-littlearm-vxworks.
Use "set gnutarget format-name" to specify the format.
(gdb) set gnutarget elf32-littlearm
```

dlech, 2018-01-18:
> how can I configure the debug symbols for the program (see backtrace output)?

You probably need to install debug symbols on the EV3 for libc6 and any other library you have dynamically linked.

    sudo apt install libc6-dbg

Or maybe install the cross debug symbols on the host system

    sudo apt install libc6-dbg-armel-cross

tcwan, 2018-01-22:
> Or maybe install the cross debug symbols on the host system
```
>sudo apt install libc6-dbg-armel-cross
```

libc6-dbg-armel-cross seems to be a Ubuntu thing. I can't find it from the docker image apt listing. There is libc6-dbg but that is for the host platform I believe.
```
compiler@8ed2ccadef26:/src$ apt list libc6-dbg
Listing... Done
libc6-dbg/stable,stable 2.24-11+deb9u1 amd64
```

dlech, 2018-01-22:
> libc6-dbg-armel-cross seems to be a Ubuntu thing

Strange, I thought I found it on the debian site. Oh, well.

I've never had an application where I could use the ev3dev/debian-$DISTRO-cross images, so I don't know how debugging packages work there, but for the ev3dev/debian-$DISTRO-$ARCH-images, I install the debug packages on the EV3 and gdbserver on the EV3 will upload the debug symbols to gdb (running via qemu emulation) in the docker image.

tcwan, 2018-01-25:
After experimenting with various settings, it appears that the `libc6-dbg` symbols for the ev3dev-stretch image are not stored in a place that remote gdb expects. 

The `libc6-dbg` package installs symbols to /usr/lib/debug, while gdb expects them from `/lib/xxx`. In addition, the hashes for the libraries does not match up (libc6-dbg  splits the symbol files into subdirectories with the first 2 hex digits as folder names), so I suspect that there is a version mismatch between the host (docker image) cross-compiler libraries and the actual libraries on the target (EV3)

```
(gdb) target remote 192.168.2.2:3333
Remote debugging using 192.168.2.2:3333
Reading /lib/ld-linux.so.3 from remote target...
warning: File transfers from remote targets can be slow. Use "set sysroot" to access files locally instead.
Reading /lib/ld-linux.so.3 from remote target...
Reading symbols from target:/lib/ld-linux.so.3...Reading /lib/62b2761d7a64fc51fd22235cff9e0ce5fccaa3.debug from remote target...
Reading /lib/.debug/62b2761d7a64fc51fd22235cff9e0ce5fccaa3.debug from remote target...
(no debugging symbols found)...done.
0xb6fcea60 in ?? () from target:/lib/ld-linux.so.3
(gdb) cont
Continuing.
Reading /lib/arm-linux-gnueabi/libm.so.6 from remote target...
Reading /lib/arm-linux-gnueabi/libc.so.6 from remote target...
Reading /lib/arm-linux-gnueabi/0d16732063e56d1e7725ffef558a37952bc2f1.debug from remote target...
Reading /lib/arm-linux-gnueabi/.debug/0d16732063e56d1e7725ffef558a37952bc2f1.debug from remote target...
Reading /lib/arm-linux-gnueabi/a1c44276e59f35ee612225f9a0579014fa8777.debug from remote target...
Reading /lib/arm-linux-gnueabi/.debug/a1c44276e59f35ee612225f9a0579014fa8777.debug from remote target...

```

On-EV3 gdb debugging works with proper symbols display for libc6, but it is not a viable solution since it is extremely slow.

tcwan, 2018-01-25:
I've managed to get Eclipse Remote GDB Debugging working on the Host!

[Eclipse Remote GDB Screenshots](https://imgur.com/a/4HjlB)

Everything is mostly transparent after some configuration nightmare, but basically I can step and trace through custom libraries (ev3dev-c) built on the Host (using Docker cross-compilation), since it is found in the same project. 

Stepping through shared system libraries does not work due to the missing debug symbols, but I guess it is not as critical at this juncture.

tcwan, 2018-01-25:
@dlech:

I think you can close this issue once you've decided whether to include gdb-multiarch in the `debian-*-cross` docker images. I'm not using it in my Eclipse configuration but others may find it useful.

dlech, 2018-03-22:
I've added the `gdb-multiarch` package. Should be pushing out a new image soon...

closed 2018-03-22 by dlech