ev3dev/ev3dev #1029: Any experience integrating Docker cross-building support into IDE

tcwan, 2018-01-19:
### What are you trying to do?
Build (and Debug) projects written in IDE running on Host PC inside Docker Cross Compiler Image

### What did you expect to happen?

Existing docker cross-compiler images for C/C++ projects are primarily command line driven.

I'm trying to find out if anyone has experience integrating the C/C++ cross-build process into an IDE running on the Host OS (PC). 

*My target IDE is Eclipse, but I'm willing to consider alternative IDEs.*

I found an Eclipse Plugin (Docker Tools), but I don't have enough experience with Docker, Eclipse or the [Docker Tools](https://wiki.eclipse.org/Linux_Tools_Project/Docker_Tooling/User_Guide) plugin to know if it'll work for this environment.

### What actually happened?

Building C/C++ applications has to be done via command line currently.

### What hardware and software are you using (including version numbers)?

ev3dev/debian-stretch-cross (Image ID 1ac15970696e)

dlech, 2018-01-19:
I use Visual Studio Code for just about everything these days. [Here is an example](https://github.com/ev3dev/grx/blob/ev3dev-stretch/.vscode/tasks.json#L60) where I added tasks to start and stop the docker container as well as one to build using docker.

tcwan, 2018-01-24:
Thanks for the info. I checked out VSC and Atom, but I find that it is still not a complete solution for a Write-Compile-Debug environment. The plugins for VSC is still rather immature due to its recent history. Hopefully it'll be better in the future and I can revisit the choice.

Currently I'm trying to adapt [Docker for Eclipse CDT](https://www.eclipse.org/community/eclipse_newsletter/2017/april/article1.php) for my purposes. It can automatically map the necessary paths into the container and perform the build.
There is still some rough edges (especially in terms of the correct configuration of the toolchain paths inside the Docker container) but it looks the most promising.

I'm still investigating the Debugging setup.

closed 2018-01-26 by tcwan

tcwan, 2018-08-03:
To add an answer to this issue (since this seems to be one of the top hits when googling Docker and IDE Integration), look at the eclipse IDE guide [here](https://github.com/ev3dev/edu-docs).