ev3dev/ev3dev #1030: Webcam problems(detecting but cannot take picture)

TechnoTew, 2018-01-21:
### What are you trying to do?
Connecting Logitech V-UAM27 to ev3dev

### What did you expect to happen?
Being able to take a pic

### What actually happened?
Can detect it through lsusb but no picture even after using the photobooth.py provided by rhempel(Thanks BTW)

### What hardware and software are you using (including version numbers)?

**System info (from `ev3dev-sysinfo`)**     
```                                         
Image file:         ev3dev-jessie-ev3-generi
c-2017-06-09                                
Kernel version:     4.4.87-22-ev3dev-ev3    
Board:              LEGO MINDSTORMS EV3 Prog
rammable Brick                              
Revision:           0006                    
Brickman:           0.8.1                   
ev3devKit:          0.4.3                   
```

TechnoTew, 2018-01-22:
The following is the result 
Bus 001 Device 002: ID 046d:08a0 Logitech, Inc. QuickCam IM
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

dlech, 2018-01-22:
What is the output of `lsmod | grep gspca_zc3xx`? We may not have the driver enabled in the kernel config.

TechnoTew, 2018-01-23:
There was no output after keying in the command

dlech, 2018-04-22:
Fix released in `4.14.35-ev3dev-2.0.0-ev3` kernel.

closed 2018-04-22 by dlech

TechnoTew, 2018-05-06:
I'm using jessie @dlech

dlech, 2018-05-06:
If you want to use this webcam, then you will need to use ev3dev-stretch.

TechnoTew, 2018-05-06:
What is the difference between stretch and jessie?

It should be fine if I'm still able to host a online server to connect to ev3 like seanbot

Also where is helper.py, i can't seem to find it

dlech, 2018-05-06:
stretch is newer and under active development. jessie is no longer being updated. I don't know anything about helper.py.

TechnoTew, 2018-05-06:
Oh so stretch is now being developed, I'll move over to stretch when I have the time.

Thanks @dlech

TechnoTew, 2018-05-06:
Using stretch now, fswebcam still doesn't work

Output:"--- Opening /dev/video0...
stat: No such file or directory"

TechnoTew, 2018-05-06:
@dlech Why would fswebcam downgrade my quickcam resolution

I'm using V-UAM27, QuickCam IM, the stated max resolution is '640x480' but I'm only able to get 320x240.

But good job getting it to work!

dlech, 2018-05-06:
The USB port on the EV3 is USB 1.1 only, so any feature of a webcam that requires USB 2.0 won't work. Some webcams won't work at all.

TechnoTew, 2018-05-07:
How would I check, can't seem to find socifications for the usb

dlech, 2018-05-07:
You could check the webcam vendors website. However, my experience is that they tend to publish little to no information on the technical details.