ev3dev/ev3dev #1031: RPyC and Daisy Chaining

Rosstb, 2018-01-21:
I have spent several hours working on trying to get my EV3's to daisy chain using RPyC. I have been trying to use this tutorial: https://sites.google.com/site/ev3python/learn_ev3_python/rpyc But, I can't figure out how to use it to connect two EV3's instead of an EV3 and a PC. I have Googled it, but I can't find any resources on daisy chaining EV3's with RPyC. Would somebody be able to write a short tutorial explaining how to do this properly? Or, if somebody has code I could look at, that might help too.

Thanks!

dlech, 2018-01-21:
cc @ev3dev/python

ddemidov, 2018-01-21:
In the example you mentioned (also, see [this one](http://ev3dev-lang.readthedocs.io/projects/python-ev3dev/en/stable/rpyc.html) in the python library docs) there is a controller (PC), and a controllee (the EV3). The controllee has to run the RPYC server with `./rpyc_server.sh` command, and the controller has to connect to that server with `rpyc.classic.connect(<controllee-ip-addr>)`.

I guess that daisy chain means there are several controllees and controllers, so you need to run an RPYC server on each of the controllees, and each controller should connect to its controllee with `rpyc.classic.connect(<controllee-ip-addr>)`.

Rosstb, 2018-01-21:
robot@ev3dev:~/rpyc$ python3 rpyc.py
Traceback (most recent call last):
  File "rpyc.py", line 1, in <module>
    import rpyc
  File "/home/robot/rpyc/rpyc.py", line 2, in <module>
    conn = rpyc.classic.connect('169.254.175.20')

This is an error I get when I run the code from the tutorial in the python library docs. I took out the conn = rpyc.classic.connect('ev3dev')  of the EV3 line and replaced it with conn = rpyc.classic.connect('169.254.XXX.XX'). I run the code using SSH from my PC to the Master EV3, and it should run a motor in the A port on the Slave EV3. Any idea what the error is about?

ddemidov, 2018-01-22:
Is that the complete error? Did you install rpyc on all of the EV3s?

Rosstb, 2018-01-22:
Yes, that is the complete error, and yes, RPyC is installed on both of the EV3's.

ddemidov, 2018-01-22:
It looks like you named your file `rpyc.py`. This may be the problem: this overlaps with the rpyc module, so that when you are doing `import rpyc`, python interpreter tries to import your script instead.

Rosstb, 2018-03-31:
I really appreciate you guys helping me with this! Unfortunately, I didn't have time to get back to this until now. Anyway, I thought about my setup, and realized I was running it wrong. The Master EV3 was also running the RPYC server. According to my understanding, the Slave is supposed to be running that. I am still getting errors however.

import rpyc
conn = rpyc.classic.connect('169.254.175.20') # host name or IP address of the EV3
ev3 = conn.modules['ev3dev.ev3']      # import ev3dev.ev3 remotely
m = ev3.LargeMotor('outA')
m.run_timed(time_sp=1000, speed_sp=600)

This is the code that I run on the Master EV3 after the Slave starts running the RPYC Server. 

Traceback (most recent call last):
  File "RPYCTest.py", line 2, in <module>
    conn = rpyc.classic.connect('192.168.137.3') # host name or IP address of th                                                                                                             e EV3
  File "/usr/lib/python3/dist-packages/rpyc/utils/classic.py", line 68, in conne                                                                                                             ct
    return factory.connect(host, port, SlaveService, ipv6 = ipv6, keepalive = ke                                                                                                             epalive)
  File "/usr/lib/python3/dist-packages/rpyc/utils/factory.py", line 89, in conne                                                                                                             ct
    s = SocketStream.connect(host, port, ipv6 = ipv6, keepalive = keepalive)
  File "/usr/lib/python3/dist-packages/rpyc/core/stream.py", line 132, in connec                                                                                                             t
    return cls(cls._connect(host, port, **kwargs))
  File "/usr/lib/python3/dist-packages/rpyc/core/stream.py", line 102, in _conne                                                                                                             ct
    s.connect(sockaddr)
ConnectionRefusedError: [Errno 111] Connection refused

Also, is the slave EV3 supposed to have the gadget check box checked or not?

dlech, 2018-03-31:
How are you connecting the EV3s? USB, Bluetooth, Wi-Fi?

Rosstb, 2018-03-31:
The two EV3's are connected with USB.

dlech, 2018-03-31:
You only need to check the gadget box if you want to share the Internet connection of one EV3 to the other.

Rosstb, 2018-04-02:
Thanks @dlech! @ddemidov, any ideas on the error?

ddemidov, 2018-04-02:
Can you ping slave from master on this ip?

Rosstb, 2018-04-02:
How would I go about doing that? I apologize for not knowing much about this subject, or how to go about stuff like this.

ddemidov, 2018-04-02:
You can do this by going to ssh console on the master, and running 
```
ping 192.168.137.3
```
from there.

Rosstb, 2018-04-09:
Yes, I was able to do that @ddemidov . The output from the ssh console is:

64 bytes from 192.168.137.3: icmp_seq=1 ttl=64 time=0.613 ms
64 bytes from 192.168.137.3: icmp_seq=2 ttl=64 time=0.425 ms
64 bytes from 192.168.137.3: icmp_seq=3 ttl=64 time=0.422 ms
64 bytes from 192.168.137.3: icmp_seq=4 ttl=64 time=0.440 ms
64 bytes from 192.168.137.3: icmp_seq=5 ttl=64 time=0.424 ms
64 bytes from 192.168.137.3: icmp_seq=6 ttl=64 time=0.427 ms
64 bytes from 192.168.137.3: icmp_seq=7 ttl=64 time=0.612 ms
64 bytes from 192.168.137.3: icmp_seq=8 ttl=64 time=0.541 ms
64 bytes from 192.168.137.3: icmp_seq=9 ttl=64 time=0.421 ms

I used ctrl C to stop this at icmp_seq=97.

Rosstb, 2018-04-09:
@ddemidov Yes, I was able to do that. The output in the ssh console is:

robot@ev3dev:~$ ping 169.254.90.201
PING 169.254.90.201 (169.254.90.201) 56(84) bytes of data.
64 bytes from 169.254.90.201: icmp_seq=1 ttl=64 time=1.96 ms
64 bytes from 169.254.90.201: icmp_seq=2 ttl=64 time=2.30 ms
64 bytes from 169.254.90.201: icmp_seq=3 ttl=64 time=1.92 ms
64 bytes from 169.254.90.201: icmp_seq=4 ttl=64 time=1.85 ms
64 bytes from 169.254.90.201: icmp_seq=5 ttl=64 time=2.12 ms
64 bytes from 169.254.90.201: icmp_seq=6 ttl=64 time=2.86 ms
64 bytes from 169.254.90.201: icmp_seq=7 ttl=64 time=1.85 ms
64 bytes from 169.254.90.201: icmp_seq=8 ttl=64 time=1.95 ms
64 bytes from 169.254.90.201: icmp_seq=9 ttl=64 time=1.62 ms
64 bytes from 169.254.90.201: icmp_seq=10 ttl=64 time=1.60 ms
64 bytes from 169.254.90.201: icmp_seq=11 ttl=64 time=1.69 ms
64 bytes from 169.254.90.201: icmp_seq=12 ttl=64 time=1.95 ms
64 bytes from 169.254.90.201: icmp_seq=13 ttl=64 time=2.11 ms
64 bytes from 169.254.90.201: icmp_seq=14 ttl=64 time=1.60 ms
64 bytes from 169.254.90.201: icmp_seq=15 ttl=64 time=1.75 ms
64 bytes from 169.254.90.201: icmp_seq=16 ttl=64 time=2.28 ms
64 bytes from 169.254.90.201: icmp_seq=17 ttl=64 time=1.41 ms
64 bytes from 169.254.90.201: icmp_seq=18 ttl=64 time=1.36 ms
64 bytes from 169.254.90.201: icmp_seq=19 ttl=64 time=1.97 ms
^C

ddemidov, 2018-04-09:
You have IPs in `169.254.*.*` range here and in https://github.com/ev3dev/ev3dev/issues/1031#issuecomment-377701632, but in the error message there is `conn = rpyc.classic.connect('192.168.137.3')`. Could that be a problem?
Other than that, I am out of ideas (except may be check that `rpyc_server.sh` is still running on the controllee, and does not spit any errors).

Brisbane-QLD, 2018-12-04:
@Rosstb Do you still need help with this? The USB solution is very tricky because the master needs to be able to ping all the slaves, and based on my attempts USB hub port IP visibility is very challenging compared to a traditional router/switch network. Myself and another person have daisy chaining working using EV3s with Wifi adapters, and there's an inexpensive compatible EdiMax Wifi adapter available. https://github.com/ev3dev/ev3dev-lang-python/issues/570

zhangbinxy, 2019-09-18:
I guess when you run rpyc_classic.py, you should define the hosname, add '--host YOUR SERVER ADDRESS' to it.
ie.
rpyc_classic.py --host  169.254.90.201